lvgl/lvgl #7188: Persian Language Character Handling Bug

Amir1387aht2, 2024-10-30:
### LVGL version

v9.2.1

### What happened?

First of all, thank you for supporting languages like arabic and persian

Now what's the problem:
Today i tried to use persian language for keyboard and inputfield, when i typed "سلام" text on keybaord, the character glyph was wrong, detaily, on "سلا" word everything was fine, but when i added "م", it broked, i mean the "لا" word should connect to "م" and "س", but it wont
[image]
[image]

### How to reproduce?

```
lv_obj_t * kb = lv_keyboard_create(lv_screen_active());

    /*Create a text area. The keyboard will write here*/
    lv_obj_t * ta1;
    ta1 = lv_textarea_create(lv_screen_active());
    lv_obj_align(ta1, LV_ALIGN_TOP_LEFT, 30, 30);
    lv_obj_add_event_cb(ta1, ta_event_cb, LV_EVENT_ALL, kb);
    lv_textarea_set_placeholder_text(ta1, "Placholder");
    lv_obj_set_size(ta1, 180, 50);

    lv_keyboard_set_textarea(kb, ta1);
    lv_obj_set_width(kb, 210);

    lv_obj_set_style_text_font(kb, &lv_font_dejavu_16_persian_hebrew, 0);
    lv_obj_set_style_text_font(ta1, &lv_font_dejavu_16_persian_hebrew, 0);
    
   ```

kisvegabor, 2024-10-30:
Hey,

Have you enabled `LV_USE_ARABIC_PERSIAN_CHARS`? 

If it doesn't help, I think I know what the problem could be:
- We insert the text to the text area [here](https://github.com/lvgl/lvgl/blob/master/src/widgets/textarea/lv_textarea.c#L298)
- This function also updates the text of the label [here](https://github.com/lvgl/lvgl/blob/master/src/widgets/label/lv_label.c#L682)
- In `lv_label_set_text` Arabic characters are processed [here](https://github.com/lvgl/lvgl/blob/master/src/widgets/label/lv_label.c#L151)

`lv_text_ap_proc` is called on each insertions to convert the characters to the correct form. It also means that new characters are inserted to an already processed text, and I think `lv_text_ap_proc` can't work on already processed texts. 

If really this is the problem, I think we should write a function like `lv_text_ap_normalize` which reverts all characters to their base form, so that we can call `lv_text_ap_proc` without any issues.

As I'm not speaking Arabic, it's hard to debug this issue for me.  Could take a deeper look on it?

Amir1387aht2, 2024-10-30:
As a persian is my primary language, i can help with that, i take a deep look into it, but please tell me how lv_text_ap_proc work

kisvegabor, 2024-10-30:
Great! 

> but please tell me how lv_text_ap_proc work, i mean, when it's called

What you set a new text with `lv_label_set_text()`, `lv_text_ap_proc` is called to convert all the characters to the correct Arabic form (isolated, end, etc). After that we store the converted text.

Amir1387aht2, 2024-10-30:
> Great! 
> 
> > but please tell me how lv_text_ap_proc work, i mean, when it's called
> 
> What you set a new text with `lv_label_set_text()`, `lv_text_ap_proc` is called to convert all the characters to the correct Arabic form (isolated, end, etc). After that we store the converted text.

Everytime we click a button on keyboard it's called right? So lv_text_ap_proc should be edited?

Amir1387aht2, 2024-10-30:
Ahhhh now I understand the problem, i don't know this way works or not but how about save the inputed text 2 times, one is processed and other is original text, when needs to process text again, we work with original text and put result in other strings

Amir1387aht2, 2024-10-30:
Other way is in ``` lv_text_ap_proc ``` check for all possible glyps and then re process text

Amir1387aht2, 2024-10-31:
I understand, I've work on lv_text_ap_normalize function

Amir1387aht2, 2024-11-01:
@kisvegabor There is a serious bug in keyboard that i don't understand,
It seems "لا" word have issue, the problem is when i type "ل" and "ا" to make "لا", it seems function you wrote to calculate length of character, it calculate "لا" just 1, if i want to say it simply, when you type "لا" and then press delete key on keyboard, it's remove both "ل" and "ا" that's incorrect

kisvegabor, 2024-11-06:
Can you track where it goes wrong? It returns 2, so at least the basics are ok:
```c
  int a = lv_text_get_encoded_length("لا");
  printf("%d\n", a);  //2
```

Amir1387aht2, 2024-11-06:
@kisvegabor I found a cool thing, 

> ```
>  int a = lv_text_get_encoded_length("لا");
>   printf("%d\n", a);  //2
> ```

This code work diffrent for textinput, i used same function (``` lv_text_get_encoded_length ```) to display text input text length and it printed "لا" work 1

[image]

Amir1387aht2, 2024-11-09:
Working on it but hard to understand what going on

kisvegabor, 2024-11-13:
Thank you for working on it! :pray:

closed 2024-12-09 by lvgl-bot