lvgl/lvgl #7195: Documentation Dark Mode Has One Really Bad Color

vwheeler63, 2024-10-30:
### LVGL version

v9.3

### What happened?

cc:  @kisvegabor 

In Dark Mode in the documentation, the color that is used to highlight *recognized function names* is so dark that it is nearly invisible:  it nearly blends into the black background.  See example here:  https://docs.lvgl.io/master/details/base-widget/scroll.html#scroll-manually  which looks like this on my screen:

[image]

Furthermore, in LIGHT mode, it is so dark it is nearly black against the light background so I almost cannot tell it is highlighted:

[image]

I investigated the generated cascading sequence in the documentation web pages and found that it only exists in `./out_html/_static/pygments.css` (provided by Sphinx via the `pygments` Python package for code-block highlighting) as

```css
.highlight .nf { color: #06287e } /* Name.Function */
```
and does not otherwise occur in the remaining (cascaded) 6 style sheets included after that.  So it can be successfully overridden in `./docs/_static/css/custom.css`.  

I have worked out a new color for this and tested it:
```css
.highlight .nf { color: #2648ee } /* Name.Function */
```
and it works beautifully for my eyes for both backgrounds.  See the results here.

[image]

[image]

Gábor, if you will, please confirm that this new color is acceptable.  (Call in any other opinions that might be interested.)  If it is acceptable, I am ready with a PR to submit.  I'm awaiting confirmation that this new color works for everyone involved.

Kind regards,
Vic

### How to reproduce?

See above.

kisvegabor, 2024-10-31:
Hi,

Than you for the investigation. 

It's okay for me to change it. Please send a PR

vwheeler63, 2024-10-31:
> Please send a PR

Done!  PR #7199

closed 2024-11-04 by kisvegabor