lvgl/lvgl #7196: `lv_calendar_set_showed_date()` Contains a Grammar Error

vwheeler63, 2024-10-30:
### LVGL version

v9.3

### What happened?

`lv_calendar_set_showed_date()` -- to be correct English -- should be `lv_calendar_set_shown_date()`, or even better (reflecting what the C code does in the function): `lv_calendar_set_month_shown()` (since the day number is hard-coded to 1 and it only accepts `year` and `month` arguments).

Current documentation contains this:
```
lv_calendar_set_shown_date(calendar, year, month)
```

I found this because it was partially corrected in the documentation, but not in the code.  I don't know which was first, but can attest that *at least* the code that needs to change, and the documentation as well if `lv_calendar_set_month_shown()` is selected instead of what is in the current documentation.  Perhaps this can be done with a back-patch using a macro so that existing apps are not broken?

And the brief description in the API documentation should change from:
```
Set the currently showed
```
to
```
Set month currently shown
```

### How to reproduce?

See above.

kissa96, 2024-10-31:
Thanks for pointing it out! Do you want to open a PR for this?

vwheeler63, 2024-10-31:
> Thanks for pointing it out! Do you want to open a PR for this?

I think I could probably do it, but was hoping to get Gábor's feedback, as I know he knows a lot more about the impact it might create than I do.

By the way, are you two related (family)?

vwheeler63, 2024-10-31:
Also observed:  `lv_chart_set_all_value()` also contains a grammar error:  should be `lv_chart_set_all_values()` since it sets all values in a data series to a passed value.

cc:  @kisvegabor 

Gábor -- it occurs to me that I might run into many of these (grammar errors) before this is over.  Reason:  I am proceeding through the Widgets alphabetically and am only down to Chart and have found 2 so far.  Perhaps they should be gathered together and done as one PR with one modification to the `lv_api_map_....h` files?

Can you also guide me so I understand the precise design-intention behind the `lv_api_map_....h` files?  They appear to all be a map from prior API syntax to the "current", so that each branch remains current with its back-mapping.  In other words, "If I have v8.x API calls in my application, I can include `lv_api_map_v8.h` to ease my migration from v8 to current?"  Is that correct?

Kind regards,
Vic

kisvegabor, 2024-10-31:
Hi Vic,

The name is just a coincident with @kissa96 :slightly_smiling_face: 

Regarding the calendar: I like `lv_calendar_set_month_shown`. Please do this:
- Rename `lv_calendar_set_shown_date` to `lv_calendar_set_month_shown`
- Add a `static inline lv_calendar_set_shown_date` function in `lv_api_map_v9.h` in which `lv_calendar_set_month_shown` is called. (basically a wrapper)

Regarding the chart:
- Please rename the the function to `lv_chart_set_all_values`
- And add `#define lv_chart_set_all_value lv_chart_set_all_values` in `lv_api_map_v9.h`

So we are using `lv_api_map` for these kind of things  Not everything can be covered in these API map files, but they can really lower the headache a little when migrating. :slightly_smiling_face:

kissa96, 2024-10-31:
Hey, haha we are not related. Kiss is a fairly common family name in our country.

vwheeler63, 2024-10-31:
> Hey, haha we are not related. Kiss is a fairly common family name in our country.

Thank you!  😄

vwheeler63, 2024-10-31:
Hi, Gábor!  ( @kisvegabor )

Okay!  I will accumulate these in their own branch and not submit it yet until I get past documentation-editing for the last Widget, as I have a feeling there might be more of these, then I can submit in 1 PR.  I am thinking it could make it easier to merge if docs are separated from actual code updates (not to be confused with code-documentation updates) wherever possible.  Just because code updates (I think) require a "different viewpoint" to approve than documentation updates.  Let me know if you think these should come instead in another "batch plan" (e.g. `chart.rst` with `lv_chart...` code).  I can do either plan equally easily.  (Seeking your confirmation or guidance.)

Another thing I need your confirmation/guidance on is this:  Remember when I described the frustration of learning LVGL v8.x due to missing bits in the documentation, and having to clarify things by reading the actual C code?  I'm running into that now (\*\*laughing at self\*\*) with Chart because [the documentation](https://docs.lvgl.io/master/details/widgets/chart.html#data-series) "hints at" data-value arrays being "external" or "normal", but the code (yes, I am having to clarify this by reading the code) indicates that X and Y coordinate values ONLY come from `series->y_points[i]` and `series->x_points[i]` where `y_points` and `x_points` are pointers and therefore they can ONLY come from external arrays.  In other words, there is no such thing as "normal" data-value arrays being different than "external" arrays.  Is this a hold-over from a prior implementation?  Can you confirm that I am seeing this correctly?  If I am — as a programmer, I will defer (honor) what's in the code before what's in the documentation every time — I very much need to clarify this in the documentation, because it *will lead programmers astray* having to do the same clarification that I am doing now (which is costly time-wise)....

Looking forward to your reply,
Vic

kisvegabor, 2024-11-07:
Both are fine for me. Please pick the direction that you feel better. :slightly_smiling_face:

PR #7340 (2024-11-29): feat(chart,calendar):  two grammar corrections
- lv_calendar_set_showed_date() => lv_calendar_set_month_shown()
- lv_chart_set_all_value() =>  lv_chart_set_all_values()

Both of these were discussed in Issue #7196 and guidance given by @kisvegabor therein so as to not break the API.

Resolves #7196

cc:  @kisvegabor 

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.  n/a
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.  n/a
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.  n/a
 [... cut]
closed 2024-11-29 by liamHowatt