lvgl/lvgl #7198: Where is the macro corresponding to LV-COLOR_16_SWAP in 9.0

CasualXiao, 2024-10-31:
### LVGL version

v9.1.0

### What happened?

How to configure lv_comlor_16uswap in version 9.0? My screen transfer type is SPI model st7789, hardware platform is ESP32, IDF version 5.1.3. I saw a related macro in lv_comf. h, but it didn't change after I opened it. Do I need to replace the screen driver with the one provided by lvgl and the dot drawing function?

### How to reproduce?

static void example_lvgl_flush_cb(lv_display_t *drv, const lv_area_t *area, void *px_map)
{
    esp_lcd_panel_handle_t panel_handle = (esp_lcd_panel_handle_t)lv_display_get_user_data(display);
    int offsetx1 = area->x1;
    int offsetx2 = area->x2;
    int offsety1 = area->y1;
    int offsety2 = area->y2;
    // copy a buffer's content to a specific area of the display
    esp_lcd_panel_draw_bitmap(panel_handle, offsetx1, offsety1, offsetx2 + 1, offsety2 + 1, px_map);
    /* IMPORTANT!!!
     * Inform LVGL that you are ready with the flushing and buf is not used anymore*/
    lv_display_flush_ready(display);
}

kissa96, 2024-10-31:
Hi, Im not sure what you are asking. Could you please elaborate? Im happy to help

kisvegabor, 2024-10-31:
Hi,

See [here](https://docs.lvgl.io/master/details/main-components/display.html#swapping-endian-ness).

CasualXiao, 2024-10-31:
> Hi,
> 
> See [here](https://docs.lvgl.io/master/details/main-components/display.html#swapping-endian-ness).
Is it like `this?I will try：    esp_lcd_panel_handle_t panel_handle = (esp_lcd_panel_handle_t)lv_display_get_user_data(display);
    int offsetx1 = area->x1;
    int offsetx2 = area->x2;
    int offsety1 = area->y1;
    int offsety2 = area->y2;
    // copy a buffer's content to a specific area of the display
    lv_draw_sw_rgb565_swap(px_map, 16);
    esp_lcd_panel_draw_bitmap(panel_handle, offsetx1, offsety1, offsetx2 + 1, offsety2 + 1, px_map);
    /* IMPORTANT!!!
     * Inform LVGL that you are ready with the flushing and buf is not used anymore*/
    lv_display_flush_ready(display);

CasualXiao, 2024-10-31:
After I add this code again, lv_rraw_stw_rgb565_stwap (px_map, 16); Still unable to display
[image]

liamHowatt, 2024-11-04:
You are calling `lv_draw_sw_rgb565_swap(px_map, 16)` which swaps only 16 pixels. You need to swap all of the pixels. Try `lv_draw_sw_rgb565_swap(px_map, lv_area_get_size(area))`.

CasualXiao, 2024-11-07:
> You are calling `lv_draw_sw_rgb565_swap(px_map, 16)` which swaps only 16 pixels. You need to swap all of the pixels. Try `lv_draw_sw_rgb565_swap(px_map, lv_area_get_size(area))`.
Thank you, the display is normal now

closed 2024-11-07 by CasualXiao