lvgl/lvgl #7203: ImageButton graphic doesn't change when pressed, unless you set an empty style

Windwoes, 2024-11-01:
### LVGL version

9.3.0

### What happened?

If I create an image button and give it graphics for pressed and released states, the button does not update when pressed, unless I also set an empty style on the image button, and then it works.

### How to reproduce?

```c
static void loadImgBtn(lv_obj_t* ptrImgBtn, uint16_t width, uint16_t height,
                       const void* imgUnpressed,const void* imgPressed, uint16_t x, uint16_t y)
{
    ptrImgBtn = lv_imagebutton_create(ui_screen);
    lv_imagebutton_set_src(ptrImgBtn, LV_IMAGEBUTTON_STATE_RELEASED, imgUnpressed, NULL, NULL);
    lv_imagebutton_set_src(ptrImgBtn, LV_IMAGEBUTTON_STATE_PRESSED, imgPressed, NULL, NULL);
    lv_obj_set_width(ptrImgBtn, width);
    lv_obj_set_height(ptrImgBtn, height);
    lv_obj_align(ptrImgBtn, LV_ALIGN_TOP_LEFT, x, y);

    // Workaround bug in LVGL where the button graphic won't change when pressed,
    // unless we set a style and then it seems work ?
    static lv_style_t style;
    lv_style_init(&style);
    lv_obj_add_style(ptrImgBtn, &style, LV_STATE_PRESSED | LV_STATE_DEFAULT);
}
```

lizhaoming634, 2024-11-01:
maybe you can try
 lv_obj_add_style(*ptrImgBtn, &style, LV_STATE_PRESSED | LV_PART_MAIN);

liamHowatt, 2024-11-05:
Thanks for reporting it. I was able to reproduce it.

You can make it work by simply applying the image source as the `src_mid` parameter instead of the `src_left` parameter.

```c
    lv_imagebutton_set_src(*ptrImgBtn, LV_IMAGEBUTTON_STATE_RELEASED, NULL, imgUnpressed, NULL);
    lv_imagebutton_set_src(*ptrImgBtn, LV_IMAGEBUTTON_STATE_PRESSED, NULL, imgPressed, NULL);
```

The problem is that the imagebutton refresh handler does nothing if the middle image is `NULL`. I opened a PR: #7224

Windwoes, 2024-11-06:
> You can make it work by simply applying the image source as the src_mid parameter instead of the src_left parameter.

Okay, I'll try that when I get a chance. Also, I think it would be helpful to update the docs to explain why there are 3 source parameters and the effects of specifying (or not) the various combinations. It looks like previous versions of LVGL only took a single source image parameter.

liamHowatt, 2024-11-06:
I added a line to the docs which says: "Typically the middle image should be one of the set image sources."

PR #7224 (2024-11-28): fix(imagebutton): warn if middle image is not set
Fixes #7203

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/code-format.py` (`astyle v3.4. [... cut]
closed 2024-11-28 by XuNeo