lvgl/lvgl #7204: when  LV_IMAGE_ALIGN_TILE is used,  lv_image is incorrect in vg_lite drawing enabled

huohongpeng, 2024-11-01:
### LVGL version

v9.2

### What happened?

I used lv_image to draw image in  LV_IMAGE_ALIGN_TILE and vg_lite enabled,  only one image was drawed , not tile more image.

liamHowatt, 2024-11-05:
Looks like image tiling is not supported with VG Lite yet.

See these generated ref images for the same test.

- Drawn by software renderer: https://github.com/lvgl/lvgl/blob/4aa74e15c3f21ff1a53fb5b699910c6dcf901f44/tests/ref_imgs/widgets/image_tile.png
- Drawn by VG Lite renderer: https://github.com/lvgl/lvgl/blob/4aa74e15c3f21ff1a53fb5b699910c6dcf901f44/tests/ref_imgs_vg_lite/widgets/image_tile.png

kisvegabor, 2024-11-05:
cc @FASTSHIFT @nicusorcitu

nicusorcitu, 2024-11-08:
@cosmindanielradu19

cosmindanielradu19, 2024-11-11:
Hi, we do support tiled images, and the benchmark results looks fine on our side.

https://github.com/lvgl/lvgl/blob/master/src/draw/nxp/vglite/lv_draw_vglite_img.c#L178

liamHowatt, 2024-11-11:
Right, if I recall correctly, tiling works on-device but not in the simulated environment.

kisvegabor, 2024-11-13:
In the ThorVG implementation tiling really doesn't work. See the test images:
- ThorVG/VGLite API: https://github.com/lvgl/lvgl/blob/master/tests/ref_imgs_vg_lite/widgets/image_tile.png
- SW render: https://github.com/lvgl/lvgl/blob/master/tests/ref_imgs/widgets/image_tile.png

kisvegabor, 2024-12-02:
@FASTSHIFT should it be fixed?

FASTSHIFT, 2024-12-03:
> @FASTSHIFT should it be fixed?

Yes, I'll take some time to look at it.

kisvegabor, 2025-02-07:
@FASTSHIFT Is it on your radar?

FASTSHIFT, 2025-02-08:
The image width of vg-lite needs to be aligned to 16px. For images that are not aligned to 16px, you may see non-image data used to fill the alignment. This may not work well in tiled mode and is not supported yet.

closed 2025-03-02 by lvgl-bot