lvgl/lvgl #7206: Chart Cursor Misplaced on Bar Graph with Multiple Series

vwheeler63, 2024-11-01:
### LVGL version

v9.3

### What happened?

This was done by adding a red LV_DIR_ALL cursor to `lv_chart_example_2.c`, and calling:
```c
lv_chart_set_cursor_point(chart, cursor, ser1, 2);
```
which should place the cursor on the 3rd value, at the top edge of the bar in series 1.  As you can see, it gets the right value and *height*, but the wrong bar (wrong series).
[image]

Likewise with series 2:
```c
lv_chart_set_cursor_point(chart, cursor, ser2, 2);
```
Same as before:  it gets the right *height*, but the wrong bar.
[image]

### How to reproduce?

Add this to `lv_example_chart_2.c` after the 2 data series have been created:

```c
    /* Add cursor to chart to see if it moves with scrolling. */
    lv_chart_cursor_t* cursor = lv_chart_add_cursor(chart, lv_color_make(255, 0, 0), LV_DIR_ALL);
    lv_chart_set_cursor_point(chart, cursor, ser1, 2);
```

vwheeler63, 2024-11-01:
Note:  this *may* also affect `lv_chart_get_point_pos_by_id()` which is intended to fetch the coordinates of that point.

lizhaoming634, 2024-11-01:
[image]

lizhaoming634, 2024-11-01:
This is another mistake in the chart, it's better to modify it together
https://github.com/lvgl/lvgl/issues/7184

vwheeler63, 2024-11-01:
> This is another mistake in the chart, it's better to modify it together #7184

What does that cause?

lizhaoming634, 2024-11-01:
> > This is another mistake in the chart, it's better to modify it together #7184
> 
> What does that cause?

This may lead to zero division errors

vwheeler63, 2024-11-01:
> This may lead to zero division errors

I see!  If you can handle both, you are welcome to.

vwheeler63, 2024-11-04:
Handled by PR #7211!  Thank you @lizhaoming634 !

closed 2024-11-04 by vwheeler63