lvgl/lvgl #7208: LVGL Image converter mangles iage palette

blkhawk, 2024-11-01:
### LVGL version

master

### What happened?

The Image converter mangles indexed based images by reordering and discarding entries from the palette.

### How to reproduce?

If given a palette based image both the web converter as well as the script LVGLImage.py discard and reorders palette entries. This is not causing any visible issue if you want to just display an image but if you want to say palette swap to create an animation effect the issue becomes apparent. 

Optimizing and discarding color entries has as far as I can tell no advantage for a indexed image and tracking down this unexpected behavior took me hours.  

I hope that this issue can be at least found in the future by people having the same problem when working with classic pixelart animations. For more context what this is used for there is a great online demo: 

http://www.effectgames.com/effect/article-Old_School_Color_Cycling_with_HTML5.html

I have attached a short correctly working script for 256 color indexed images.

[convert1.zip](https://github.com/user-attachments/files/17600111/convert1.zip)

liamHowatt, 2024-11-05:
Since you already have a fix, please open a PR and it can likely be merged :slightly_smiling_face:

blkhawk, 2024-11-05:
I took a short look at the code - I think it would be annoying to fix the code because it never cares about the format of the input image directly - If i see it correctly it just chucks the PNG bytes to PngQuant and that's what probably is to blame for the mangling (line 875 of LVGLImage). Then it force adds 0xffffff as the first palette entry (this is weird and might have some reason I don't understand) and builds the palette from there. 

A "fix" by me would involving detecting the actual format of the image first then bypassing PngQuant and the adding of white as an entry if the requested colour format is the existing colour format. I have no idea what side-effect the second part has.

liamHowatt, 2024-11-07:
Just speculation, but to me it looks like PngQuant will make the palette sparse if some of the palette entries were not used in the image and LVGLImage.py will pad out the palette with multiple white entries at the end of the palette (not just one white entry as the first, as you say). Did you observe it actually reordering entries? Or does it just remove entries and close the gaps by shifting the remaining entries?

https://github.com/lvgl/lvgl/blob/17d0169e5099a773140df1ffa5a3320143910c7c/scripts/LVGLImage.py#L891-L896

I see what you mean about the opaque nature of PngQuant. If we were to post-process the image after PngQuant runs, we'd need to find out what the full palette is somehow and then iterate through all the pixels in the output and adjust them up to point to the true palette indices.

cc @XuNeo, the original author of LVGLImage.py

blkhawk, 2024-11-07:
Yes, I got the order wrong and that is exactly what PngQuant does. It also automatically removes colour that are too "near" another colour automatically. I am not 100% sure if it scrambles the colour order but I got the impression that it reordered the palette into the order it detected the colours in the image data. 

In my use case these changes are devastating and I blamed my Image editor at first. The fact that I had to edit the Image also made the palette a moving unpredictable target. That said using PngQuant is a perfectly fine solution to normalize whatever input image if all you care about that it looks the same on screen afterwards. 

I think a good solution is to simply detect the Format and if the imput format matches the output format to simply bypass that step. 

For context I don't use palette manipulation for animation but to control about 250 display elements with complex shapes cheaply.

XuNeo, 2024-11-08:
Currently all PNG is firstly processed by `pngquant` for LVGL.Indexed image format. If the image comes in PNG 8bit mode, it could bypass the `pngquant` process by judging at [here](https://github.com/lvgl/lvgl/blob/f94b34fff12a4c6442f11a394f582296b1a6a140/scripts/LVGLImage.py#L875-L877).

https://github.com/lvgl/lvgl/blob/f94b34fff12a4c6442f11a394f582296b1a6a140/scripts/LVGLImage.py#L875-L877

LVGL support I1/2/4/8 and `pngquant` may output palette that has less colors than LVGL wants, thus palette is padded to meet LVGL.

blkhawk, 2024-11-08:
Yes, I was wondering why exactly the colour padding was needed. I assumed it was a requirement but i could see no obvious defects in my testing even without a full white. Maybe its only a factor if other UI elements are mixed in or  the image is used in a UI element. My use is very basic :)

blkhawk, 2024-11-08:
Weird that its a requirement then - since a palette would contain all black by default and that is a color as well so there is no reason to pad the palette for indexed images at all.

liamHowatt, 2024-11-11:
`IX` color format images in LVGL always begin with a (2 ^ `X`) * 4 byte palette. It's a constant known amount for LVGL to skip to get to the image pixel data. You're right, it doesn't matter what value is used for padding, but to be useable by LVGL, the padding must be there.

How are you swapping out palettes at runtime? I think you must already know what I've just described, since you'd need to know that in order to swap palettes at runtime.

blkhawk, 2024-11-11:
Currently I am simply editing each colour directly in the data structure then invalidating the image object for a redraw. In my project I arranged the colour palette in such a way that I can do a memcpy to set different patterns directly but i am not doing that yet. 

For context here is a short YT clip of what I am actually doing (its a replica of a rare 1980ies VW car instrument cluster): 
https://youtu.be/pl4Qp_JWD0k

XuNeo, 2024-11-27:
You means re-use the palette in PNG if it's a 8bit PNG file right? Sure, thank you!

blkhawk, 2024-11-27:
well the logic is not too complex: basically "if input is indexed and requested format is auto or identical then do not run pngquant.convert()". I need to clean it up a bit first and maybe add a comment explaining what/why it does it like that.

blkhawk, 2024-11-27:
Okay I did a pull request - is there a process to update the website?

XuNeo, 2024-11-29:
The website doesn't use this tool. See https://github.com/lvgl/lv_img_conv

blkhawk, 2024-11-29:
thanks because it needs fixing as well

closed 2024-12-02 by FASTSHIFT