lvgl/lvgl #7213: lv tiny ttf works so slow, is there another way to use vector fonts fastly?

lanistor, 2024-11-03:
### Introduce the problem

I had a vector font ttf file, it contains full text in Chinese, Japanese, English, German, French, Russian, etc.
The ttf file size is `8MB`, when using tiny ttf to dynamic load this vector font, for every render, it will need nearly 60 seconds.

Is there another way to use vector fonts fastly?

Env: `ESP32S3`, `1024*758` screen.

MacDragon, 2024-11-03:
tiny ttf basically has to parse through the whole font file every time it deals with an uncached charecter, as it keeps nothing dynamically in memory itself - you may be better off trying to use truetype if dealing with fonts that big I suspect

lanistor, 2024-11-04:
> tiny ttf basically has to parse through the whole font file every time it deals with an uncached charecter, as it keeps nothing dynamically in memory itself outside the amount of cached entries that lvgl wrapper is configured to deal with - you may be better off trying to use freetype perhaps if dealing with fonts that big I suspect, I believe it has more active cacheing of the font internal structures

Thanks, i will try it.

lanistor, 2024-11-05:
> tiny ttf basically has to parse through the whole font file every time it deals with an uncached charecter, as it keeps nothing dynamically in memory itself outside the amount of cached entries that [...]

@MacDragon Freetype works faster than ttf a lot, but on ESP32, it's behavior is still slow, whole paint of one screen need nearly 10 seconds.

I'm using chinese font, it has nearly 10,000 characters.

I think i have to give up vector font, instead of using pixcel font.

closed 2024-11-06 by lanistor

Satnet, 2024-11-10:
Here is an awesome lifehack I figured out. Generate a large font, let's say you make one 48px in the font converter.
Not TTF or FreeFont, just a usual LVGL font.

Then use lv_obj_set_style_transform_zoom(label, 255, 0);
255 will set it to 100% size. From there you can go downwards! 128 would be about half the size.

It works, no performance issues and you save space.

Silen-xt, 2024-12-25:
May I ask how the landlord solved this problem? Whether it is convenient to share.
I have also encountered this problem. It takes a long time to use the font through lv_tiny_ttf_create_data, but the font obtained through Satnet sharing is not clear enough. Is it convenient to share your way? I am looking forward to your reply.
 Thank you very much