lvgl/lvgl #7214: ThorVG internal file error

Gangwa-Labs, 2024-11-04:
### LVGL version

v9.1

### What happened?

When using the internal ThorVG library to play a Lottie animation it fails to build due to errors inside the thorVG library

"no instance of constructor "ColorStop::ColorStop" matches the argument list" & "no match for 'operator=' (operand types are 'ColorStop' and '<brace-enclosed initializer list>')"

I've placed the line of code that is causing it to fail. it can be found on line 721 in lvgl/src/libs/thorvg/tvgLottieProperty.h

### How to reproduce?

            const_cast<LottieColorStop&>(other).value = {nullptr, nullptr};

liamHowatt, 2024-11-06:
I could not reproduce the compilation error in `release/v9.1` or `master`. Are you able to update the LVGL? Otherwise, please send a more precise version description of LVGL which you are using. Thanks!

C47D, 2024-11-12:
I tried to search a similar error in the thorVG repo but couldn't find one. 

---

HUGE OFFTOPIC:

From a "C head" this looks like the `ColorStop` struct doesn't have the `=`operator OR it doesn't match the expected `{ }` initialization.

kisvegabor, 2024-11-13:
cc @hermet

mesopa, 2024-11-21:
Also run into the same problem, tried with LVGL builds 9.2 and 9.2.2 and got always the same errors when enabling the ThorVG on my project, inside PlatformIO.

hermet, 2024-11-21:
Hello, I guess this fix addresses the issue. Could someone verify it with this change?

721 in lvgl/src/libs/thorvg/tvgLottieProperty.h
```
//const_cast<LottieColorStop&>(other).value = {nullptr, nullptr};
const_cast<LottieColorStop&>(other).value = ColorStop();   
```

DblBee, 2024-11-22:
I tested this fix and can confirm this solves the compile issue with LVGL.

`const_cast<LottieColorStop&>(other).value = ColorStop(); `

hermet, 2024-11-23:
here is the patch: https://github.com/lvgl/lvgl/issues/7214, the fix will be in thorvg v0.15.5. thanks.

kisvegabor, 2024-11-28:
Thank you @hermet!

@Gangwa-Labs can you send a PR to LVGL with the fix?

closed 2024-12-21 by lvgl-bot