lvgl/lvgl #7217: Set different names for threads created by different units

cristian-stoica, 2024-11-04:
### Introduce the problem

The OS abstraction layer defines and uses a single name "lvglDraw" for all LVGL threads when LV_USE_OS is enabled.

Because a single thread name is used this can lead to confusion when identifying running threads. For example, during FreeRTOS profiling I've found that identifying SW and VGlite threads was confusing enough that I've made a patch to address it in my setup. This is resolved by adding a 'thread_name' argument to `lv_thread_init`.

Is this something you also find useful? I know there are a bunch of OSs that don't use thread names and there are also written applications that may need updating if the lv_thread_init signature is changed. If it shows interest, I'll submit for PR for review.

Thanks

### Proposal

```
-lv_result_t lv_thread_init(lv_thread_t * thread, lv_thread_prio_t prio, void (*callback)(void *), size_t stack_size,
+lv_result_t lv_thread_init(lv_thread_t * thread, const char * const pcName,
+		lv_thread_prio_t prio, void (*callback)(void *), size_t stack_size,
                            void * user_data);
```

kisvegabor, 2024-11-04:
I could be okay with it, but it brings up the question of whether `lv_thread` is a public or private API. My intention was to keep it private but it leaked to the public API space (i.e. it's available from `lvgl.h`)

I opened https://github.com/lvgl/lvgl/pull/7220 to prevent such leakage. If we can agree on that it should be private we can modify this API without any issues.

cristian-stoica, 2024-11-22:
Not stale

cristian-stoica, 2024-12-13:
not stale

uLipe, 2025-01-04:
Not stale

kisvegabor, 2025-01-07:
@cristian-stoica Could you comment [on this](https://github.com/lvgl/lvgl/issues/7217#issuecomment-2455450202)?

cristian-stoica, 2025-01-08:
Sure - I'll prepare a patch for review

closed 2025-01-15 by kisvegabor