lvgl/lvgl #7223: Rework and unify CMake

kisvegabor, 2024-11-05:
### Introduce the problem

Now our CMake "policy" is not clear. The way we link 3rd party lib or enable/disable demos and examples is messy. We need a unified way of handling all these.

### Proposal

After having a discussion with @etag4048 and @glebm I believe we should do this:
- CMake should use `lv_conf.h` or `Kconfig` configs
- From these the enabled options should be extracted
- CMake should compile and link accordingly. 

What do you think?

glebm, 2024-11-05:
That makes sense.

kisvegabor, 2024-11-26:
@etag4048 will be back next week and can look into it.

uLipe, 2024-11-26:
What about let cmake to handle only the source tree that is platform agnostic of the lvgl and register it in the package registry from cmake.

Zephyr and NuttX do that and this facilitates a lot the creation of out of tree projects that can just use the cmake's find package that creates lvgl as an archive file add to the target list then the platform specific project just links it against the final binary.

kisvegabor, 2024-11-28:
I'm open to any ideas/solution. My goal is to create something flexible and stable that can be used for a few years.

However note that it's not only about 3rd party libs, but internal LVGL configs too. E.g. if `LV_BUILD_EXAMPLES == 0` we can skip the building and linking of the examples. Same for each demo.

tdjastrzebski, 2024-12-01:
I am not a CMake expert but I used to build LVGL 8.x with my own build tool (cppbuild) and it worked just great. No issues at all.
Now LVGL 9.x got more sophisticated. `thorvag/rapidjson` library was added which contains two files (`stdint.h`, `inttypes.h`) causing conflicts. Includes like `#include LV_DRAW_DMA2D_HAL_INCLUDE` are used which my tool cannot resolve.
All in all, it might be easier to go with the mainstream and use CMake. LVGL contains (intends to contain?) a support document dedicated to CMake config (https://docs.lvgl.io/integration/build/cmake) but this page appears to be missing, 404 error is returned. Is it because it is in works?

kisvegabor, 2024-12-02:
@tdjastrzebski 
> here are only three above issues which, when fixed manually allow my tool to work correctly again

Do you remember what issues did you have? Could you add a build log?

@uLipe 
> One thing that is also could be useful here is to pick the lv_conf.h and generate a lvconf.cmake that can be included into the main cmake file allowing for example to select which directories should be considered during build system generation.

Sounds good to me!

tdjastrzebski, 2024-12-03:
@kisvegabor  One issue is that under `src\libs\thorvg\rapidjson\msinttypes` there are two headers (`inttypes.h` and `stdint.h`) which are supposed to be used only with Microsoft Visual C++ compiler. My [cppbuild](https://github.com/tdjastrzebski/cppbuild) tool does not know that `inttypes` and `stdint` are standard libraries and tries to include headers from `src\libs\thorvg\rapidjson\msinttypes` which ofc causes conflicts and errors.
Another issue is that my build tool cannot correctly resolve includes like `#include LV_DRAW_DMA2D_HAL_INCLUDE` using preprocessor variable defined somewhere else, that is, in `lv_conf.h`. Currently there is only one such instance in LVGL.
Ofc, this could be handled and [cppbuild](https://github.com/tdjastrzebski/cppbuild) could be improved but I just do not have capacity to do it and I'd rather keep it (relatively) simple so I am willing start using CMake for LVGL.

kisvegabor, 2024-12-22:
Not stale.

AndreCostaaa, 2025-01-30:
Would having a `CmakeLists.txt` per folder be a good idea to handle this?

Something like: 

1. Root CMakeLists loads config from `lv_conf.h`
2. It then includes the other `CMakeLists.txt` recursively based on config
    - Using `add_subdirectory()`
3. Each `CMakeLists.txt` then compiles the `.c` files in its own folder based on config
   - Using `add_library(FOLDER_NAME OBJECT module1.cpp)`
4. Repeat step 2 and 3 recursively

Doing this we can start small, eg: start with  a CMakeLists.txt inside `src/indev` (there's only 3 .c files there) and the biggest hurdle is checking which config/ combination of configs is/are needed to safely include a .c file / directory

uLipe, 2025-01-30:
@AndreCostaaa  I like this approach, this is how for example Zephyr handles that, the difference here it picks the output .config from the Kconfig subsystem and generate a set of symbols in a .cmake file and include as top file to all other cmake files. We can make the lv_conf.h to generate a cmake config file and them repeat the other steps above. and each module of LVGL can be treated as a library archive and just link the used ones to the final binary

etag4048, 2025-01-31:
While I like quickly editing ``lv_conf.h`` in the editor, as the project grows using KConfig with default configs is the better and more maintainable approach. Making it quite similar to what is found in the linux kernel or Zephyr.  The only downside is that it introduces some build dependencies.

uLipe, 2025-01-31:
Even though going to kconfig may add build dependency, since kconfiglib has been released it was simpler to manage this build requirement, we can make lvgl build system to depend on it:

https://github.com/ulfalizer/Kconfiglib

JojoS62, 2025-01-31:
But kconfig is not available in every environment?
One important thing I think is to remove the file globbing, by using this the lib is always compiling everything. Adding new files is not such a big deal, it is not too much work when the most files are already in (nested) CMakeLists.

uLipe, 2025-01-31:
Not in every env, the standalone build system from LVGL does not use it. The Kconfig infrastructure we have currently is just to make it easy to integrate on OS like Zephyr and NuttX. But at the end of build step the .config gets processed and supply a template `lv_conf.h` file for the target platfrm..

Yes about the globbing, this is something that really needs to be reworked if we need to have a scalable and configurable build system based on cmake nested files, one thing also would be good to create a sort of conditional functions to include a particular module depending on the output of the config file.

kisvegabor, 2025-02-03:
@AndreCostaaa Thank you for picking it up an opening a PR. My comments (note that I'm not an expert CMake user, so I might miss something):

- What problem does the CMakeFile-s in each folder solves I'm asking because I see a few drawbacks too:
  - It's a maintenance issue as CMakeFile-s needs to be added to each new folder and new files needs to be added manually as well. These are easy to miss on PR reviews.   (We might be able to check them with CI, but not having these kind of issues is clearly better)
  - LVGL should be build system agnostic. Therefore if we establish the practice of having CMake files in each directory, we might want to add meson, make, SCons, etc files as well. Which would be really hard to maintain.

- Because LVGL should be as platform independent as possible we need to keep  `lv_conf.h` too as this the most universal config interface. A header file is the most feature rich too. Kconfig is limited to string, bool, int, and choice. In H file we can do any macro magic if needed. E.g. [these](https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L63-L68).

tdjastrzebski, 2025-02-03:
What I dare to suggest is make sure, or at least try to make sure, that LVGL always builds but simply compiling **all** the c/cpp files with a minimum set of global macro 'defines'. Until recently it worked this way. Now it still almost does.

Once you start getting sophisticated using particular functions of this or the other build system there is no coming back.
Introduction of such dependency initially may seem very beneficial but long-term dramatically reduces product maintainability simply because the one who wants to maintain the product must be proficient not only in the used language but also in the particular build system.
Ofc, there are products which are so complex that there is no other way around. Is LVGL already one of them? I am not convinced.
Just my 3 cents.

tdjastrzebski, 2025-02-03:
@AndreCostaaa Let me present yet another perspective. I am able to build projects using LVGL using a simple build script like [this one here](https://github.com/tdjastrzebski/STM32F769I-DISCO-LVGL/blob/master/.vscode/c_cpp_build.json) and relatively simple [cppbuild](https://github.com/tdjastrzebski/cppbuild) tool I put together bored on vacation several years ago. It is very fast, probably as fast as it gets.
Such build requires almost no maintenance because LVGL (and my code) is structured the way it is possible to simply compile all the files. Using more sophisticated tool would inevitably encourage me to complicate things beyond the actual need. For this reason, as long as I can I consciously refrain from using more sophisticated tools being remained of a sentence I heard long time ago: 
_"A master can be recognized by the constraints he imposes on himself"_.
This sentence is attributed to Goethe and Bruce Lee.
Ofc, this is just my perspective. Modifying my cppbuild script to run cmake is not a problem at all and this is not the reason why I wrote the above.

AndreCostaaa, 2025-02-03:
Looks very cool, thank you for sharing !

But like you said :

> Modifying my cppbuild script to run cmake is not a problem

Plus, you don't even need to run cmake very often, once lvgl is compiled, the only thing left to do is to link it to your own project unless you modify lvgl/ the lvgl config which I, correct me if i'm wrong, would believe happens rarely

tdjastrzebski, 2025-02-03:
The co concern is rapidly increasing complexity. LVGL already starts looking like a Swiss Army knife. Probably SVG and XML stuff are good candidates for a separate repo.

kisvegabor, 2025-02-05:
In general I agree with [this comment](https://github.com/lvgl/lvgl/issues/7223#issuecomment-2631605337) of @tdjastrzebski.
One of the key differentiators of LVGL is the portability (that is system agnostic nature). We shouldn't sacrifice it.

@AndreCostaaa 
I still don't understand what's point of having many small CMake files. We don't have to resolve the "which files are included in a build based on configs" as it's already handled in LVGL by having `#if LV_USE_...` checked at beginning of the files. These `LV_USE_` defines are created in [`lv_conf_internal.h`](https://github.com/lvgl/lvgl/blob/master/src/lv_conf_internal.h#L310-L320) if they are
- set in `lv_conf.h` or
- passed as a compiler define (`-D`) or
- set in Kconfig or
- not defined (use a default value) 

So it's up to the user which way to use (`lv_conf.h` or `Kconfig` or none and use `-D`)  and LVGL will create the required symbols for itself internally.

However I do agree to allow completely disabling the inclusion of examples and demos. This way the core LVGL (`src` folder) is ~400 files which is not that bad. 

IMO the real problem that we need to solve is how to make sure that e.g. FreeType is linked if `LV_USE_FREETYPE` is enabled. So to automatically handle external lib linking. But as I'm writing it, I'm not sure that it's something that we should handle in LVGL but it can be user's responsibility to link the required libs as needed.

Note that LVGL is a library, not an app or framework.  So there is no point (or at least I can't see it) to build an app from LVGL directly. Always there should be a wrapper project which provides the configs somehow (`lv_conf.h`, `Kconfig`, `-D`).

One special case is when LVGL is compiled to static library (`.a`) or a dynamic library (`.so`).
- Static library: IMO it's normal that the user links the required 3rd party library and not LVGL tries to find an link them automatically
- Dynamic library: We can just link the 3rd party `.a` files into the `.so` file too, or we can modify LVGL 3rd party library support interfaces to use `dlopen` and `dlsym` to load 3rd party libs for so files at runtime. 

What do you think?

glebm, 2025-02-05:
> But as I'm writing it, I'm not sure that it's something that we should handle in LVGL but it can be user's responsibility to link the required libs as needed.

That's not how CMake targets work, it is not possible (and would be bad if it were possible) to add link dependencies to an external target.
If LVGL defines a library target (static or shared doesn't matter), it must link all of its dependencies.

glebm, 2025-02-05:
Note that one of my earlier attempts was a hack to allow the user to specify the list of extra libraries to link (https://github.com/lvgl/lvgl/pull/6958) but I later realized it's a terrible idea:

1. It requires the dependent library target to be GLOBAL.
2. It makes a future install configuration (e.g. to make a Debian package) very difficult.
3. It's not how every other CMake library out there works.

AndreCostaaa, 2025-02-05:
Hi @kisvegabor 

I guess I misunderstood what you meant in the proposal of the issue. By reworking i thought you saw the weakness of how things work right now and were looking for an alternative, so my proposition is based on what's currently done in projects like `zephyr`

**Multiple CMakelists**

I don't think having multiple `CMakelists` is problematic, it's really easy to maintain them and allows much more flexibility compared to file globbing. 

**#ifdef wrapping**

I'ts my opinion that not including a file inside the compilation process is a much cleaner way of handling the current `#ifdef` wrapping of whole .c files. For instance, currently `src/others/xml` contain 16 `.c` files that shouldn't ever be added to the compilation process and for that, every single one of these files is wrapped in `ifdef`. Wouldn't it be easier to just add

```cmake
# src/others/CMakeLists.txt
if(LV_USE_XML)
    list(APPEND SUBDIRS xml)
endif()
```

**lv_conf_internal**

As to comment on how `lv_conf_internal` works, like I said it's much error prone and harder to maintain, specially since the work could be done by `Kconfig` 

So I believe that it will keep getting harder to maintain as the project continues to grow but having said that maybe `lvgl` is not that big yet and if you don't feel it's time to go through such a change maybe it's because the change is not needed yet

kisvegabor, 2025-02-07:
> That's not how CMake targets work, it is not possible (and would be bad if it were possible) to add link dependencies to a target that comes from an external project (whether IMPORTED or FetchContent'd).

Wow, I'm still living in the `make` era. Thank you! With this we made a full circle and got back to the original issue/suggestion in the [first post](https://github.com/lvgl/lvgl/issues/7223#issue-2635223004). So we can extract the enabled options somehow from `lv_conf.h` or a Kconfig config file and link the libraries in CMake accordingly.  

> By reworking i thought you saw the weakness of how things work right now and were looking for an alternative, so my proposition is based on what's currently done in projects like zephyr

I think Zephyr is a bad reference for us as Zephyr is not embedded into any other systems, but other "stuff" are embedded into Zephyr. That is Zephyr is more like a framework and not a library (like LVGL). Our goal is to make sure that LVGL as library can be embedded into many frameworks:
- Zephyr 
- ESP-IDF
- Arduino (no Cmake, nor Kconfig)
- PlatformIO
- NuttX
- FreeRTOS
- Windows
- Linux
- Mac
- Bare metal make
- Bare metal CMake
- Managed Eclipse project
- UEFI BIOS
- Any the uncountable custom workflows of the users

> I don't think having multiple CMakelists is problematic, it's really easy to maintain them and allows much more flexibility compared to file globbing.

I still don't understand what problem it resolves once we have `#if LV_USE_...` It seems like a redundant features. 
 
> Wouldn't it be easier to just add something like ?
> ```
> # src/others/CMakeLists.txt
>  if(LV_USE_XML)
>      list(APPEND SUBDIRS xml)
>  endif()

We can't do it all systems. E.g. Arduino doesn't use Cmake.

AndreCostaaa, 2025-02-07:
> We can't do it all systems. E.g. Arduino doesn't use Cmake.

Totally forgot about arduino and you're right, conditional compiling with arduino is, afaik impossible to do without `#if LV_USE_...` in the source files.

> So we can extract the enabled options somehow from `lv_conf.h` or a Kconfig config file and link the libraries in CMake accordingly.

I understand the issue better now, i know see that i was diverging a lot from what you had i mind. Sorry about that

As per linking 3rd party libraries, examples and demos, it would imply loading the definitions from `lv_conf.h` or `kconfig`

I already had loading the `kconfig` inside `cmake` working in #7700 and that could be used to add/exclude libraries, examples and demos but since the plan is to only use `kconfig` for esp and zephyr, maybe it won't be necessary ?

Using `lv_conf.h` would not be so simple, I don't think `cmake` can load definitions from a `.h` file and use them. So there must be a better way to do this

uLipe, 2025-02-07:
@AndreCostaaa actually is possible to pick the kconfig symbols and generate a set of set(SYMBOL VALUE) zephyr uses that approach allowing to use the kconfig symbols as you would do in source files, probably the same can occur using a header file ( in  plain without comments)

AndreCostaaa, 2025-02-07:
> [@AndreCostaaa](https://github.com/AndreCostaaa) actually is possible to pick the kconfig symbols and generate a set of set(SYMBOL VALUE) zephyr uses that approach allowing to use the kconfig [...]

Yes, `kconfig` is easy as it was already working in #7700. Sorry if i wasn't clear 

I meant that doing the same thing using a `.h` file will be much more complicated. `kconfig` lines are simple `CONFIG_XXX=YYY` while `#defines` in `.h` can be much more complicated to parse as they can expand to other configs, they may be wrapped in `#if #else` etc... You basically need to write a c preprocessor that expands to cmake variables

kisvegabor, 2025-02-10:
> #defines in .h can be much more complicated to parse as they can expand to other configs, they may be wrapped in #if #else etc... You basically need to write a c preprocessor that generates cmake variables

We can have a lot of assumptions as this parser needs to support only `lv_conf.h` not any H files. For example top level defines to enable a module are never wrapped. E.g. `#define LV_USE_FREETYPE` is always on its own. 

As we are really writing the parse for ourselves, if needed,  we can write a separate parser for FreeType, FFmpeg, libPNG, etc, to handle special cases (if any). It'd be very flexible and as we have 5-10 external libraries it's also manageable.  

FYI, we already parse `lv_conf.h` to [generate `lv_conf_internal.h`](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and it's not the complicated.

etag4048, 2025-02-10:
Yep, parsing simple LV_USE_* defines should be simple enough. I've been reading up on CMake lately and now I am leaning towards agreeing with @glebm handling external dependencies within the main repo is the way to go. Doing it the 'CMake way' brings the possibility to build LVGL on many different platforms, linux, windows, macOS. And let CMake find/fetch dependencies.
We should of course keep the possibility of using LVGL from a top level directory. I am currently investigating the cleanest way to accomplish this.

glebm, 2025-02-10:
You're proposing to parse `lv_conf.h` rather than kconfig? The best way to parse a C header is to run the C compiler, but I think using kconfig is better.

When using CMake, kconfig is better as the source of truth because:

1. Parsing it is trivial.
2. A whole set of tools already exists for kconfig, including:
  1. Graphical configurators (e.g. `menuconfig` CLI and GUI)
  2. Merging tool
  3. Minimization tool (`savedefconfig`)
3. Ease of migration: because a minimization tool exists, one can only specify settings different from the default. This makes updating to a new version of lvgl much easier. My PR https://github.com/lvgl/lvgl/pull/6974 shows how to go from minimized config to the full one using CMake.
4. kconfig is used to configure the Linux kernel, so it is familiar to many already.

kisvegabor, 2025-02-10:
To find the 5-10 configs that we need is really as simple as `cat lv_conf_template.h | grep -P "#define +LV_USE_FREETYPE"`

For this introducing any extra tools and step feel overkill to me.

glebm, 2025-02-10:
Regardless of how CMake support is handled, using defconfig is much nicer than `lv_conf.h`, so it'd be good to add defconfig support to CMake anyway. At which point there is no reason to parse `lv_conf.h`.

AndreCostaaa, 2025-02-10:
It's also a good oportunity to unify the way lvgl is configured instead of having to support both lv_conf.h and kconfig

glebm, 2025-02-11:
Yeah, that's the idea. Only Kconfig when using CMake directly. Without CMake, one can still provide an lv_conf.h file directly.
This is very similar to what https://github.com/lvgl/lvgl/pull/6974 does if you set the `LV_CONF_DEFCONFIG_PATH` option.

kisvegabor, 2025-02-11:
How would it look like in the practice? I mean if I'm using CMake to cross compile an application to e.g. Linux  based MPU project what shall I type into the terminal? I guess
1 `cd lvgl`
1. run menuconfig (how can we run it on Mac/Windows/Linux)?
1. `cd ../build`
1. `cmake ..`
1. `make -j` 

Is it correct?

glebm, 2025-02-11:
> I mean if I'm using CMake to cross compile an application to e.g. Linux based MPU project what shall I type into the terminal

That would work but in CMake you wouldn't usually build `lvgl` separately but build is as a dependency of your application.
Most people would `lvgl` as a subproject via `FetchContent` and simply set the path to the `defconfig` file. I have an example here (that example generates `lvgl_conf.h` by itself because lvgl does not yet have support for doing that, so disregard that bit): https://github.com/glebm/lvgl-cmake-example

A `defconfig` file (`defconfig` refers to a minimized Kconfig configuration file) looks like this:

https://github.com/glebm/lvgl-cmake-example/blob/173b7dae95ed9b0d03d3c0295c364718514b56ac/src/platforms/sdl2/lvgl_defconfig#L1-L10

There are implementations of Kconfig. For example, on Linux the `kconfig-conf` [package](https://manpages.debian.org/testing/kconfig-frontends/kconfig-conf.1.en.html) is commonly available in distributions:

1. Run `kconfig-mconf Kconfig` in the `lvgl` source directory and select the desired options. This will produce a `.config` file (the full configuration file with all the option values).
2. Run `kconfig-conf --savedefconfig .config Kconfig` to produce a minimized config file (where defaults are omitted).

For Windows and Mac the easiest is probably to use the pure Python one https://github.com/ulfalizer/Kconfiglib (it is equivalently powerful, including `menuconfig`):

```python
pip install kconfiglib
menuconfig
savedefconfig
```

kisvegabor, 2025-02-14:
Thank you! Note that, we already have similar solution to defconfig: You need to create [lv_conf.defaults](https://github.com/lvgl/lv_port_pc_eclipse/blob/master/lv_conf.defaults) and [this script](https://github.com/lvgl/lvgl/blob/master/scripts/generate_lv_conf.py) can generate `lv_conf.h` from it.

> Run kconfig-mconf Kconfig in the lvgl source directory and select the desired options. This will produce a .config file (the full configuration file with all the option values).

It means with CMakethe users **have to use Kconfig**, right? And `lv_conf.h` won't be supported.

It's a personal preference, but I don't like using Kconfig. Maybe this is partly the reason why I'm not that supportive in adding it.

glebm, 2025-02-14:
> It means with CMake the users have to use Kconfig, right? And lv_conf.h won't be supported.

lv_conf.h could still be supported but with the caveat that the user then needs to specify the corresponding CMake options that pull in dependencies manually.

kisvegabor, 2025-02-17:
What is the problem with doing this:
1. If there a defconfig file, use it
2. Otherwise extract the configs from lv_conf.h

[Here](https://github.com/lvgl/lv_port_linux/blob/master/scripts/backend_conf.sh) we already have script that can read lv_conf.h, it's really very simple.

AndreCostaaa, 2025-02-17:
That script isn't entirely correct. For example, consider this code:

```c
#define FOO 0
#if FOO
#define BAR 1
#endif
```

The script will incorrectly report that BAR is defined when it isn't.

I understand that, right now, you only care about external libraries, and those aren't wrapped inside #if directives. But will you remember this when you eventually decide to wrap these lib configs in something like `#if LV_ALLOW_EXTERNAL_LIBS` ?

You said: 

> My goal is to create something flexible and stable that can be used for a few years.

I don’t think building on top of non-standardized solutions is the best way to achieve stability. kconfig is widely used in large projects for a reason—the tooling already exists and provides everything you need.

That said, kudos to everyone involved in creating these scripts. It’s clear that a lot of effort has gone into them, and they do work. I’m only trying to point out that they may not be the most scalable or stable solution in the long run

kisvegabor, 2025-02-18:
I see your point but we can say:
1. KConfig is the pro solution
2. and `lv_conf.h` is the 99% working simpler one.

The user can pick. 

You might argue if `lv_conf.h` is really the simpler, but it's being used by many people and they are ok with it.

glebm, 2025-02-19:
Rather than parsing `lv_conf.h`, which can fail in numerous and unobvious ways, I think it is far more sensible to require the user to specify the CMake options manually in that scenario. Anyway, I think that's a minor point in the overall design.

XuNeo, 2025-02-19:
Hi,

I'm also looking forward to have better support of Kconfig and defconfig.
As discussed in https://github.com/lvgl/lvgl/pull/6522#issuecomment-2268581293, using the existing and widely used Kconfig tool sets instead of inventing a new method(lvgl's configuration generator) is more reasonable and should be more consistent with variety of RTOS(NuttX, Zephyr, RT-Thread etc) where Kconfig is also used.

I'm imagining that LVGL provides a easy to use python tool, to generate `lv_conf.h`(or `lv_conf_kconfig.h`?) from defconfig together with Kconfig. User can change the generated `lv_conf.h` directly too, if they feel it's more easier.

In this way, we can provide multiple `defconfig` file, for different scenarios like with debug enabled, minimal config etc.

Please correct me if I'm missing something.

kisvegabor, 2025-02-19:
@XuNeo Multiple default settings would be really welcome. This way we could provide a configs like
- typical (same as the current `lv_conf_template.h`)
- minimal (very basic config)
- linux (for MPUs)

We have added [`lv_conf.defaults`
](https://github.com/lvgl/lv_port_pc_eclipse/blob/master/lv_conf.defaults) which does basically the same that we could do with Kconfig.

Anyhow, I still believe that `lv_conf.h` and Kconfig support needs to coexist. Kconfig for integration with OS-es and `lv_conf.h` as a dependency free, always working option. Obviously it's be nice to handle as much as possible with common code. 

If we have `lv_conf.defaults` it's easy to generate defconfig from it. The other direction is qorking as well (`defconfig`->`lv_conf.defaults`), however in LVGL we don't use `CONFIG_*` prefixed defines but use `LV_USE_*`, so it makes more sense that user interacts with those. 

In light of that my suggestion:
- Store the default in files like `lv_conf.minimal.defaults` and have script which can create `lv_conf.h` or defconfig from it. It's an updated version of [this script](https://github.com/lvgl/lvgl/blob/master/scripts/generate_lv_conf.py).
- If there is an `lv_conf.h` file, CMake extract the values from there. If there is a defconfig file, CMake uses it. 

This way we can keep both worlds happy.  What do you think?

Daxidz, 2025-02-25:
Hi everyone! After some talk with @kisvegabor, it was decided to push forward a first implementation of the `Kconfig`/`lv_conf.h` coexistence to be able to configure the project. I will be pushing a proposal in the next weeks, which we can discuss and improve on if needed.
It was also decided that splitting the `CMakeLists.txt` in the subfolders will be on hold for now, as we need to focus on Kconfig integration. This may be taken over again after the first part is done.

PR #7934 (2025-04-03): arch(cmake): Add native Kconfig support to cmake
Fixes #7223 (first part of the work)

It allow to build lvgl with Kconfig directly, without the need of an integration on top. It also creates the CONFIG_LV_USE_* as cmake variable, to be used in future work.

Allow to use the `-DLV_USE_KCONFIG` cmake option to use Kconfig instead of `lv_conf.h`.
It now uses a preprocessed `lv_conf_internal.h` to extract the `CONFIG_LV_USE_*` and offer them as cmake variables. 

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
  -  To be discussed what to update/ad [... cut]
closed 2025-04-03 by kisvegabor