lvgl/lvgl #7226: lv_anmt-a new animation api

cnjhb, 2024-11-05:
### Introduce the problem

Hello everyone.
I implemented a tweener style animation system that is based on the current lv_anim.
I was wondering if it would be worth merging, it currently relies on glib.I'll rewrite it if it's merge-worthy.
Example of use:
```c
lv_anmt_anm(obj,
    "x",500,
    "y",300,
    "h",30
    "tm",300,
    NULL);
```
The above code will generate a 300ms animation about the obj.
This animation will eventually set the x of the obj to 500, the y to 300 and the height to 30.
The lv_anmt_anm parameter is variable, but must end in NULL.

kisvegabor, 2024-11-06:
Hi,

What is tweener? :sweat_smile: I could find it Google with quick search.

cnjhb, 2024-11-06:
> Hi,
> 
> What is tweener? 😅 I could find it Google with quick search.

https://discourse.gnome.org/t/gtk4-animations-a-strawman-proposal/239

liamHowatt, 2024-11-06:
Would `<stdarg.h>` be used to read alternating `const char *` and `int` parameters until `NULL` is encountered?

cnjhb, 2024-11-06:
> Would `<stdarg.h>` be used to read alternating `const char *` and `int` parameters until `NULL` is encountered?

Yes,you can.
https://gitlab.gnome.org/GNOME/glib/-/blob/main/gobject/gobject.c#L3288

closed 2024-12-01 by lvgl-bot