lvgl/lvgl #7230: Slider-Knob Delayed Updates in Examples

vwheeler63, 2024-11-05:
### LVGL version

v9.3

### What happened?

Updates to Slider-Knob position are being delayed in current examples (current = 5-Nov-2024 15:40 GMT).  This is occurring in both FireFox and Chrome browsers (the only 2 I have tested this with), in both LIGHT and DARK modes, and is occurring everywhere Sliders are used in examples (including on the Slider page).  The following screenshots illustrate the problem, from https://docs.lvgl.io/master/details/widgets/msgbox.html 

Original Display (Pre-Slider-Change):
[image]

After Slider Movement:
[image]

After Scrolling (which causes the display of the knob positions to be updated again):
[image]

This "feels" like a failure to invalidate the KNOB after sliding, but I am guessing.

### How to reproduce?

See above.

lizhaoming634, 2024-11-06:
The slider example runs well in version 9.2, but it encounters issues when running on the master version
[image]
[image]

lhdjply, 2024-11-06:
Maybe need revert #6638

kschwan, 2024-11-06:
Same issue here. Happens since #6638

lizhaoming634, 2024-11-07:
@kisvegabor Could you please take a look at this issue?

vwheeler63, 2024-11-07:
Perhaps it is really simple like 1 line of code to invalidate the knob....  At the right place.

lizhaoming634, 2024-11-08:
The roller is also experiencing issues.
[image]
When the roller scrolls to 10, further upward scrolling causes issues.

kisvegabor, 2024-11-08:
I found a bug in #6638 [here](https://github.com/lvgl/lvgl/pull/6638/files#diff-87868f4fa80d2fbd02400750545f9d5db3581d6d096b84487e1421d6464f9e6bR883):
`*area = obj->coords;` should be `*area = obj_coords`. 

Could you also try it?

vwheeler63, 2024-11-08:
> I found a bug in #6638 [here](https://github.com/lvgl/lvgl/pull/6638/files#diff-87868f4fa80d2fbd02400750545f9d5db3581d6d096b84487e1421d6464f9e6bR883): `*area = obj->coords;` should be `*area = obj_coords`.
> 
> Could you also try it?

Beautiful!  Yes, that solves it for BOTH the Knob (on Slider) and the Roller!  Thankfully that bug only arrived a few days ago!

vwheeler63, 2024-11-08:
@kisvegabor I didn't see the fix already in the pull requests, so I went ahead and submitted it in PR #7264.

PR #7264 (2024-11-11): fix(draw):  fix bug introduced by PR #6638
PR #6638 introduced a bug that caused issue #7230.  This bug was found by @kisvegabor and is tested by myself under `lvgl/lv_port_pc_visual_studio` and it resolves both the issue seen with the Knob as well as with the Roller Widget display updates after the Widget's value has been changed.  (It is probable that there were other Widgets impacted, but these were the only ones I was aware of.)  This PR was not requested, but since I already had it on my screen, and since I did not see a fix already submitted for it, I am submitting it here.

Resolves #72 [... cut]
closed 2024-11-11 by liamHowatt