lvgl/lvgl #7231: `lv_msgbox_close_async()` Will Delete Default Display's Top Layer if It Is Modal

vwheeler63, 2024-11-05:
### LVGL version

v9.3

### What happened?

When a message box is created via `lv_msgbox_create()` passing `NULL` as the parent, instead of becoming a `Screen` like most widgets, it grabs the Default Display's Top Layer (itself a Screen) as the parent, and `LV_MSGBOX_FLAG_AUTO_PARENT` flag is added to the created Message Box widget.  So when this call gets executed:

```c
void lv_msgbox_close_async(lv_obj_t * obj)
{
    if(lv_obj_has_flag(obj, LV_MSGBOX_FLAG_AUTO_PARENT)) lv_obj_delete_async(lv_obj_get_parent(obj));
    else lv_obj_delete_async(obj);
}
```

As you can see, get the Message Box's parent (Default Display's Top Layer) and pass it to `lv_obj_delete_async()`, and I am pretty sure that was not the intention.

Suggested fix:  
```c
void lv_msgbox_close_async(lv_obj_t * obj)
{
    lv_obj_delete_async(obj);
}
```

But I need someone who has more internal knowledge than me to know this is the correct fix.

### How to reproduce?

See above.

kisvegabor, 2024-11-08:
Here
```c
lv_obj_t * lv_msgbox_create(lv_obj_t * parent)
{
    LV_LOG_INFO("begin");
    bool auto_parent = false;
    if(parent == NULL) {
        auto_parent = true;
        parent = lv_obj_class_create_obj(&lv_msgbox_backdrop_class, lv_layer_top());
        LV_ASSERT_MALLOC(parent);
        lv_obj_class_init_obj(parent);
        lv_obj_remove_flag(parent, LV_OBJ_FLAG_IGNORE_LAYOUT);
        lv_obj_set_size(parent, LV_PCT(100), LV_PCT(100));
    }
```

If there is no `parent` a new parent will be created on `lv_layer_top()`. So the message box is created on the top layer directly.

vwheeler63, 2024-11-08:
> If there is no `parent` a new parent will be created on `lv_layer_top()`. So the message box is created on the top layer directly.

Hi, Gábor!  This is exactly my point -- that when the current `lv_msgbox_close_async()` -- it deletes the PARENT!  I.e. it will cause the *Top Layer* to be deleted through the call to `lv_obj_delete_async(lv_obj_get_parent(obj));` !

kisvegabor, 2024-11-11:
Sorry, I meant:
> So the message box is **NOT** created on the top layer directly.

And it seems it really isn't deleted.

vwheeler63, 2024-11-11:
> Sorry, I meant:
> 
> > So the message box is **NOT** created on the top layer directly.
> 
> And it seems it really isn't deleted.

`**`embarassed`**`  Wow, I see it creates a "backdrop" which is a child of `top layer` and becomes the parent of the `MessageBox`.

Silly me:  I was reading that line as:
```
    parent = lv_top_layer();
```
which would have been a problem upon deletion.  `**`laughint at self`**`

Thank you for the clarification.

You know...  this *does* indicate another area that probably *should* be documented for maintainers:  the run-time object hierarchy that is built when the Widget is complex.  Sometimes it is explained in the documentation, and sometimes it is not, but a *diagram* would be a beautiful thing....  In my system for the M32 dashboard, I did diagrams of each screen in the documentation above the `..._create_...()` function for the screen — the inherited-and-redefined function each screen has.  I found it a HUGE help when it comes to maintenance.  Like now, a year later, I don't remember the details I created in October 2023....  Here is an example for my screen that showed the status of the on-board GPS (required to be operational to do lap-timing, so this screen helps Tech Support people find out whether the GPS is working properly or not, as well as providing interesting information for the end user).

Note that `wdtSkyView` is its own widget and provides a view of the GPS satellites in the sky.  This was a fun screen to build.

Definitions:
- HDOP, Horizontal dilution of precision (below 2.0 is quite accurate, above 9.0, the GPS doesn't know where it is yet).
- Hdg, Heading.

```
/**------------------------------------------------------------------------
 * \brief  Create and return pointer to LVGL screen.
 *
 * Object Tree:
 *
 * \verbatim
 *          Screen
 * ---------------------------
 *     |        |           |
 *     |     InfoTable      |
 *   Title             wdtSkyView
 *
 * \endverbatim
 *
 * ### InfoTable:
 *
 * -         Time: YY-MM-DD hh:mm:ss
 * -          Lat: 39.1645
 * -          Lon: -105.7169
 * -          Alt: 1639.1 m
 * -          Hdg: 271.5 °
 * -          Vel: 77.2 kph
 * - Sats Tracked: 20
 * -    Sats Used: 14
 * -         HDOP: 1.22
 * -        Other: 1:A:2:3
 *
 *//*----------------------------------------------------------------------*/
```

vwheeler63, 2024-11-11:
Closing since this is not a bug.

closed 2024-11-11 by vwheeler63

vwheeler63, 2024-11-11:
@kisvegabor This is what the screen looked like:

[image]

vwheeler63, 2024-11-11:
@kisvegabor What do you think about the idea of adding run-time object hierarchy diagrams for the maintainers?

kisvegabor, 2024-11-13:
> run-time object hierarchy diagrams

Do you mean drawing the object tree with LVGL on an other display or so? E.g.
```
- screen
   - obj
      - label
      - button
  - chart
  - slider         
```

vwheeler63, 2024-11-13:
> Do you mean drawing the object tree with LVGL on an other display or so?

No, simply documenting it in comments so that maintainers can see the parts and envision them in their minds so that they can *understand* the run-time structure more quickly so whatever they do (bug fixes, enhancements, etc.) can be done faster and with higher likelihood of getting it right the first time.

kisvegabor, 2024-11-14:
Could you show how an example issue could look like with this suggestion?

vwheeler63, 2024-11-14:
> Could you show how an example issue could look like with this?

Certainly.  What I am basing off is how many times while I was learning LVGL v8.3 that I needed to understand the run-time object structure of a Widget so that I could work with it.  And if it had been in the documentation, how much time it would have saved me *not* having to study the C code to find out about it.

Here is a (human-generated) example, which could be in `./src/widgets/win/lv_win.h`:

```c
/**
 * @brief  Create a Window Widget.
 *
 * Resulting object structure:                  \
 * - window (Base Widget, LV_FLEX_FLOW_COLUMN)   \__ this paragraph just over the `@param`s and `@return`.
 *     - header (Base Widget, LV_FLEX_FLOW_ROW)  /
 *     - content (Base Widget)                  /
 *
 * @param parent    pointer to a parent widget
 * @return          the created window
 */
lv_obj_t * lv_win_create(lv_obj_t * parent);
```

But in thinking about your earlier question, "Do you mean drawing the object tree with LVGL on an other display or so?" -- in theory, a utility function could walk the run-time object structure and output something to STDOUT or to a file to document it....  It would eliminate the potential of me overlooking something (i.e. if I was tired).

Example:  In `lvgl/lv_port_pc_visual-studio`, let's take `lv_example_win_1.c` as an example.  After such a utility is written like the above, a call to it could be made after the line:
```c
    lv_obj_t * win = lv_win_create(lv_screen_active());
    document_object_hierarchy(win);
```

I could insert that line (temporarily) in all the examples, and then run ALL of them to generate a file to have its parts copy/pasted into the API comments for the `..._create()` functions....

I'm just sharing ideas at this point, but wondered what you thought.

kisvegabor, 2024-11-15:
> Here is a (human-generated) example, which could be in ./src/widgets/win/lv_win.h:

I think it's a fantastic idea! Let's handle it a new pass.

 > document_object_hierarchy(win);
 
It's also interesting, however we will add a similar feature to our UI editor too. It doesn't mean that we can't add it to LVGL itself too, but I'd like to avoid overlapping functionalities. Let's get back to this later when we know more about the editor.

vwheeler63, 2024-11-15:
> I think it's a fantastic idea! Let's handle it a new pass.

It's added to my list.  👍 

> Agreed.  And to clarify what I was thinking -- it doesn't have to be a permanent addition.  It could be a "quickie" that I reset with Git.  It's sole point would be to place strings into my hands, though I LOVE the idea of it being in the editor!  Very useful!