lvgl/lvgl #7232: lvgl load binary font file, will mapping to wrong character.

lanistor, 2024-11-05:
### LVGL version

8.4

### What happened?

When convert lots of characters, lv_font_conv will mapping wrong character, such as: using `子` replaced `屏`, using `尘` replaced `密`.

Test cases:
1. When reducing the count of characters, such as 2000, the problem disappear, https://lvgl.io/tools/fontconverter has the same problem
2. Increase memory of node `node --max-old-space-size=10240 lv_font_conv.js` ... also have the problem.
3. Using `--format lvgl` works fine, the problems appears on binary file by `--format bin`.
4. `--no-compress`, `--full-info` has no effect.

Env
OS: MacOSX
Lvgl: 8.4
Node: v18.19.1

Referer: https://github.com/lvgl/lv_font_conv/issues/126

lanistor, 2024-11-06:
@kisvegabor Could you help me see this issue?

lanistor, 2024-11-08:
Finally, i use freetype vector fonts resolved it.

closed 2024-11-08 by lanistor

kisvegabor, 2024-11-08:
Could you check if enabling `LV_FONT_FMT_TXT_LARGE` has any effect?

lanistor, 2024-11-08:
> Could you check if enabling `LV_FONT_FMT_TXT_LARGE` has any effect?

It's defined to `LV_FONT_FMT_TXT_LARGE=1`