lvgl/lvgl #7246: Allow specifying indexed image palette depth

bryghtlabs-richard, 2024-11-06:
### Problem to solve

When I try to scroll a component with 12 77x77 indexed images, the screen stutters a lot. Profiling LVGL 8.4.0 shows that LVGL is spending a lot of time going from index-byte to RGBA8888 to RGB565. And I think indexed images on 9.x work the same way.

I implemented a quick fix using LV_IMG_CF_USER_ENCODED_0 with RGB565 palette, and I can now draw the 12x image screen in about 65% of the time it took before. And it saves ~60KB of image resource space.

### Success criteria

30%+ faster image draws.

### Solution outline

1. Allow specifying the palette color type separately from the indexed pixel-size
2. Old images are considered to be same indexed-pixel-size as they were before, with LV_IMG_CF_RGBA8888 palette type
3. If necessary, after de-paletteization, run through built-in pixel format converter. For most cases, this will not be needed.

### Rabbit holes

Some possible corner-cases involve indexed images with a destination type that is not supported by the display - these would need converted before drawn.

This would be an breaking change - existing indexed images would need tagged with RGBA8888 palette type.

### Testing

With unit-tests that compare indexed images against an uncompressed image to verify they match.

### Teaching

Documentation and a change-log note.

kisvegabor, 2024-11-13:
Interesting, I didn't know that this conversation is such a bottle neck. 

In v9 is works slightly differently: the indexed images are always decoded to ARGB8888 and that image is blended to any destination color format (unlike in v8, v9 and blend images to different destination color format)

Anyway, still there is a conversation, but in v9 during blending instead of image decoding. 

---

I'm wondering what would be the best way of handling it. 
1. Introduce color formats like `LV_COLOR_FORMAT_I1_RGB565`? 
2. Or always use an ARGB8888 palette, but tell LVGL to  cache the palette as RGB565 and produce RGB565 images?
3. Or leave it to the user?  Actually that's why we have a public image decider interface. :slightly_smiling_face:

bryghtlabs-richard, 2024-11-18:
> Introduce color formats like LV_COLOR_FORMAT_I1_RGB565?

This is what I did, just with USER1 meaning FORMAT_I8_RGB565. But if we want to be flexible maybe we need one with chrome-keying and one without. And maybe one with alpha and one without. Each one could use its own software path, but then that gets quite unwieldy to maintain.

>Or always use an ARGB8888 palette, but tell LVGL to cache the palette as RGB565 and produce RGB565 images?

Perhaps this could be stored in the image-cache, where it could be kept as long as it has been recently used? For my application though, it's nice to pre-bake the cache to RGB565, so that it can be smaller.

>Or leave it to the user? Actually that's why we have a public image decider interface. 🙂

This one works for me just fine too 🙂

kisvegabor, 2024-11-27:
>> Or leave it to the user? Actually that's why we have a public image decider interface. 🙂

> This one works for me just fine too 🙂

Let's do it this for now then. We can get back to this if there will be more requests.