lvgl/lvgl #7249: We are hiring LVGL developers :hugs:

kisvegabor, 2024-11-07:
### Introduce the problem

Hey,

At LVGL LLC, we are extending our team! :tada:

Simply put, if you are an active contributor here and looking for a job, you are the perfect candidate for us. The daily tasks involve maintaining and developing LVGL and supporting our chip/GPU vendor partners.

To put it more officially: :slightly_smiling_face:

For the LVGL Developer position, we’re looking for candidates who:
- Have experience with LVGL
- Are experts in C
- Maintain an active GitHub account
- Ideally have experience with the NXP and/or Renesas ecosystem and their GPUs

The tasks include:
- Supporting our chip vendor partners
- Reviewing GitHub pull requests and issues
- Developing new LVGL features

If this sounds like a great fit for you and you can imagine yourself working on these tasks as part of the core LVGL team, don't hesitate to contact us at lvgl@lvgl.io!

Read more about the other open positions and our culture here: https://lvgl.io/jobs#about

kisvegabor, 2024-11-07:
cc @PGNetHun  :slightly_smiling_face:

vwheeler63, 2024-11-22:
Hi, Gábor!  ( @kisvegabor )

How about PIC?  I have a long history with:

- PIC32MZ (probably the choice for LVGL projects since it features more program space and RAM)
- PIC32MX
- dsPIC33
- PIC24

😄

kisvegabor, 2024-11-26:
@vwheeler63 we don't have an official connection to Microchip yet, but I'll definitely keep it in mind :wink:

uLipe, 2024-11-26:
I'm also a huge fan from Microchip microcontrollers but I admit their way of doing things for their pic line is somewhat weird in a time where are standard compilers and development tools out there.

The ex-Atmel side already took the direction of not being too tied on proprietary tools, and the Risc-V side is also going on the same direction.

:)

kisvegabor, 2024-11-28:
I didn't know that 64 bit RISC-V MPUs. Not bad!

kisvegabor, 2024-12-03:
Good news guys! You might have seen @uLipe commenting a lot recently. It was because he (Felipe) started to work at LVGL and started to look in to issues and PRs. 

Felipe is coming form Zephyr so he has a great open-source and embedded experience too. I really love to work with him and I feel confident with **providing write access to @uLipe** for the `lvgl` repo. 

It means now we will have 5 active maintainers:
1. @XuNeo 
2. @FASTSHIFT 
3. @liamHowatt 
4. @uLipe 
5. @kisvegabor 

---

As we have more and more Chinese partners **I'd love to have a Chinese speaking person in our team** as well. It would help us a lot to better manage and support our Chinese partners and customers. If you have Chinese roots, but speaks English fluently, and looking for new challenges, please write to gabor@lvgl.io

joebnb, 2024-12-10:
will hire some rust developer to make it work in rust?maybe it's a good deal to get seats in rust comminute

kisvegabor, 2024-12-16:
Right now I don't see that Rust is a popular target for LVGL. But maybe it's because our Rust binding is stale. :smile: 

Our main focus now is on our UI editor. Maybe if hings gets calmed down we can steer our attention to Rust as well.

joebnb, 2024-12-17:
> Right now I don't see that Rust is a popular target for LVGL. But maybe it's because our Rust binding is stale. 😄
> 
> Our main focus now is on our UI editor. Maybe if hings gets calmed down we can steer our attention to Rust as well.

i found slint on rust https://www.toradex.com/pt-br/videos/slint-torizon-embedded-systems-ui-ew-2024 , and think lvgl may reference some idea on it may better for mutiple language development（slint create a DSL layer for each language interact with themself）

kisvegabor, 2024-12-18:
Oh yeah, Slint it really great to develop Embedded, Web, Mobile and Desktop with the same code base. However I'm not sure if Rust is an advantage in case of Embedded UIs as Rust doesn't have a wide support among chip vendors.

joebnb, 2024-12-18:
> Oh yeah, Slint it really great to develop Embedded, Web, Mobile and Desktop with the same code base. However I'm not sure if Rust is an advantage in case of Embedded UIs as Rust doesn't have a wide support among chip vendors.

yes, there is still a lot of work to be done, but things are changing quickly. For example, [esp32-rs](https://github.com/esp-rs), [stm32-rs](https://github.com/stm32-rs), [imxrt-rs](https://github.com/imxrt-rs/imxrt-hal), and [Infineon](https://hightec-rt.com/rust-evaluation) — most of these repositories were created in the past 3 years and have started to support Rust. However, the ecosystem is still lacking and will need a long time to develop.i‘m also is new one year rustacean

kisvegabor, 2025-01-08:
As  @uLipe joined our team, I'm closing this issue now.

closed 2025-01-08 by kisvegabor