lvgl/lvgl #7255: GLFW3 fails on a compositor without xwayland support.

etag4048, 2024-11-07:
### LVGL version

master

### What happened?

@liamHowatt I am attempting to add glfw3 to lv_port_linux. I've decided to try on my system
before attempting to run it on other embedded linux distros, my development host doesn't have X11/Xwayland support.

I've added the glfw3 backend to lv_port_linux in an attempt to run on Ubuntu in this PR : https://github.com/lvgl/lv_port_linux/pull/64

On a desktop-based distribution dynamic linking is used, on newer Ubuntu versions
One has to install the following packages:

`libglfw3` - X11 Libraries, **only necessary if the target system uses X11**

`libglfw3-dev` - Header files

`libglfw3-wayland` - Wayland Libraries

`libglew-dev` - OpenGL Extension Wrangler, header files

`libglew2.2` - OpenGL Extension Wrangler - Runtime environment

If the X11 libraries are installed and the compositor has XWayland support,
GLFW3 will select X11 which yields the following error on start:

```text
[Error]      (0.000, +0)      glfw_error_cb: GLFW Error 65544: X11: The DISPLAY environment variable is missing lv_glfw_window.c:320
[Error]      (0.000, +0)      lv_glfw_init: glfwInit fail. lv_glfw_window.c:200
```

See the XWayland on Demand section:
[https://blogs.gnome.org/shell-dev/2019/11/22/a-review-of-gnome-shell-mutter-3-34/](https://blogs.gnome.org/shell-dev/2019/11/22/a-review-of-gnome-shell-mutter-3-34/)

**But there are problems when using `libglew2.2` and `libglfw3-wayland`**

After installing the dependencies above and modifying the `CMakeLists.txt` file
the following errors are present in the logs when running the simulator

```text
[Error]      (0.000, +0)      glfw_error_cb: GLFW Error 65544: Wayland: Failed to create XKB compose table lv_glfw_window.c:320
[Error]      (0.000, +0)      glfw_error_cb: GLFW Error 65544: Wayland: Focusing a window requires user interaction lv_glfw_window.c:320
[Error]      (0.022, +22)     lv_glew_init: glewInit fail: 4. lv_glfw_window.c:218
```

There is a keyboard issue and a failure in `lv_glew_init`

Enabling `LV_USE_DRAW_OPENGLES` results in a black window and the failure in `lv_glew_init`
still hanppens, error 4 - apparently is an ‘Unknown error’

After a quick internet search, the following patch was applied on lvgl

```text
diff --git a/src/drivers/glfw/lv_glfw_window.c b/src/drivers/glfw/lv_glfw_window.c
index d6287999a..a307ec786 100644
--- a/src/drivers/glfw/lv_glfw_window.c
+++ b/src/drivers/glfw/lv_glfw_window.c
@@ -78,6 +78,8 @@ lv_glfw_window_t * lv_glfw_window_create(int32_t hor_res, int32_t ver_res, bool
     if(window == NULL) return NULL;
     lv_memzero(window, sizeof(*window));

+    //glfwWindowHint(GLFW_CLIENT_API, GLFW_OPENGL_ES_API);
+
     /* Create window with graphics context */
     lv_glfw_window_t * existing_window = lv_ll_get_head(&glfw_window_ll);
     window->window = glfwCreateWindow(hor_res, ver_res, "LVGL Simulator", NULL,
@@ -97,8 +99,8 @@ lv_glfw_window_t * lv_glfw_window_create(int32_t hor_res, int32_t ver_res, bool
     glfwSetWindowUserPointer(window->window, window);
     lv_glfw_timer_init();
     lv_glfw_window_config(window->window, use_mouse_indev);
-    lv_glew_init();
     glfwMakeContextCurrent(window->window);
+    lv_glew_init();
     lv_opengles_init();

     return window;
@@ -213,9 +215,11 @@ static int lv_glew_init(void)
         return 0;
     }

+    glewExperimental = GL_TRUE;
+
     GLenum ret = glewInit();
     if(ret != GLEW_OK) {
-        LV_LOG_ERROR("glewInit fail: %d.", ret);
+        LV_LOG_ERROR("glewInit fail: %s.", glewGetErrorString(ret));
         return ret;
     }
```

With the patch applied, it results in the following output on stdout:

```text
[Error]      (0.000, +0)      glfw_error_cb: GLFW Error 65543: EGL: Failed to create context: EGL failed to allocate resources for the requested operation lv_glfw_window.c:322
[Error]      (0.000, +0)      lv_glfw_window_create: glfwCreateWindow fail. lv_glfw_window.c:88
```

With the `glfwWindowHint` call - Execution doesn’t reach `glewInit`.

Most answers on Stack-overflow didn’t shine any light on the matter

The call to `glewErrorString` says it’s an Unknown error.

`[Error]       (0.025, +25)     lv_glew_init: glewInit fail: Unknown error. lv_glfw_window.c:222`

Details of the system:

```text
Desktop environment: Sway without xwayland support.

Distributor ID:      Ubuntu
Description: Ubuntu 22.04.4 LTS
Release:     22.04
Codename:    jammy
Linux etagxps 6.5.0-1027-oem #28-Ubuntu SMP PREEMPT_DYNAMIC Thu Jul 25 13:32:46 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
```

This occurs when LV_USE_DRAW_OPENGLES is set to 0
https://github.com/user-attachments/assets/d7565fda-2202-450a-8924-2cac532fab0d

This occurs when LV_USE_DRAW_OPENGLES is set to 1
[image]

kisvegabor, 2024-11-13:
We will also take a look.

liamHowatt, 2024-11-25:
Sorry for the delay. I plan to look at this soon.

kisvegabor, 2024-12-16:
Not stale

kisvegabor, 2025-01-03:
Not stale

etag4048, 2025-01-09:
Not stale.

kisvegabor, 2025-02-14:
Any updates here?

closed 2025-03-11 by lvgl-bot