lvgl/lvgl #7265: No publicly known method to change Meter's indicator needle color after it's created

TothFrank, 2024-11-08:
### LVGL version

v8.4.0

### What happened?

I went thru many examples, internet knowledge and even the lv_examples, but I cannot find any method to change the Meter object's needle line color once it is created.

I would like to have matching needle color of the meter's scale as the picture shows, not just the color of the creation time (light blue). Is there any way to do that (Just included the code how I create the scale and the indicator needle, maybe that helps)?
[image]

### How to reproduce?

`static void meter_cb(lv_event_t * e)
{
	lv_obj_draw_part_dsc_t* dsc  = (lv_obj_draw_part_dsc_t *)lv_event_get_param(e);
	double value;

	if (dsc->text != NULL)
  {
		value = dsc->value / 10;
		snprintf(dsc->text, sizeof(dsc->text), "%5.1f", value);
	}
}

void add_scale()
{
  lv_meter_indicator_t * indic;

  scale = lv_meter_add_scale(objects.simple_meter);
  lv_meter_set_scale_ticks(objects.simple_meter, scale, 51, 1, 7, lv_color_hex(0xffc8c8c8));
  lv_meter_set_scale_major_ticks(objects.simple_meter, scale, 5, 3, 10, lv_color_hex(0xffffffff), 10);
  lv_meter_set_scale_range(objects.simple_meter, scale, 160, 260, 300, 120);

  /*Add a blue arc to the start*/
  indic = lv_meter_add_arc(objects.simple_meter, scale, 3, lv_palette_main(LV_PALETTE_BLUE), 0);
  lv_meter_set_indicator_start_value(objects.simple_meter, indic, 160);
  lv_meter_set_indicator_end_value(objects.simple_meter, indic, 190);

  /*Make the tick lines blue at the end of the scale*/
  indic = lv_meter_add_scale_lines(objects.simple_meter, scale, lv_palette_main(LV_PALETTE_BLUE), lv_palette_main(LV_PALETTE_BLUE), false, 0);
  lv_meter_set_indicator_start_value(objects.simple_meter, indic, 160);
  lv_meter_set_indicator_end_value(objects.simple_meter, indic, 190);

  /*Add a green arc to the start*/
  indic = lv_meter_add_arc(objects.simple_meter, scale, 3, lv_palette_main(LV_PALETTE_GREEN), 0);
  lv_meter_set_indicator_start_value(objects.simple_meter, indic, 190);
  lv_meter_set_indicator_end_value(objects.simple_meter, indic, 240);

  /*Make the tick lines blue at the end of the scale*/
  indic = lv_meter_add_scale_lines(objects.simple_meter, scale, lv_palette_main(LV_PALETTE_GREEN), lv_palette_main(LV_PALETTE_GREEN), false, 0);
  lv_meter_set_indicator_start_value(objects.simple_meter, indic, 190);
  lv_meter_set_indicator_end_value(objects.simple_meter, indic, 240);

  /*Add a red arc to the end*/
  indic = lv_meter_add_arc(objects.simple_meter, scale, 3, lv_palette_main(LV_PALETTE_RED), 0);
  lv_meter_set_indicator_start_value(objects.simple_meter, indic, 240);
  lv_meter_set_indicator_end_value(objects.simple_meter, indic, 260);

  /*Make the tick lines red at the end of the scale*/
  indic = lv_meter_add_scale_lines(objects.simple_meter, scale, lv_palette_main(LV_PALETTE_RED), lv_palette_main(LV_PALETTE_RED), false, 0);
  lv_meter_set_indicator_start_value(objects.simple_meter, indic, 240);
  lv_meter_set_indicator_end_value(objects.simple_meter, indic, 260);  
  

  indicator = lv_meter_add_needle_line(objects.simple_meter, scale, 3, lv_color_hex(0xff009dff), -28);
  lv_meter_set_indicator_value(objects.simple_meter, indicator, 0);

  lv_obj_set_style_bg_color(objects.simple_meter, lv_palette_darken(LV_PALETTE_GREY, 4), LV_PART_INDICATOR);  ///<--- Here?
  lv_obj_set_style_line_color(objects.simple_meter, lv_color_white(), LV_PART_INDICATOR);  //<--- Or here?
  lv_obj_set_style_line_width(objects.simple_meter, 3, LV_PART_INDICATOR);  //<--- Or where?

  lv_obj_add_event_cb(objects.simple_meter, meter_cb, LV_EVENT_DRAW_PART_BEGIN, NULL);
}`

lizhaoming634, 2024-11-09:
[image]

TothFrank, 2024-11-09:
> indic->type_data.needle_line.color = lv_color_make(255, 0, 0);

@lizhaoming634 > Highly appreciate this help! Staying with my example above, here is how I changed the color of then needle:

`indicator->type_data.needle_line.color = lv_color_make(255, 0, 0);`

to red.

closed 2024-11-09 by TothFrank