lvgl/lvgl #7267: Massive issues with PNG, BMP and GIF

Satnet, 2024-11-09:
### LVGL version

V9.2

### What happened?

I've got file system running and it can successfully open JPG files. BMP, PNG or GIF's wont work.
GIF shows up as a fixed color and slows down the device so much that it's almost frozen.
BMP file tries to load from file system, I can tell because the graphics on the screen just went bonkers and then the device crashes. Memory is set very high to make sure no memory issues.

This is a custom board fully operational with SDRAM of 32 Megabytes.

So I moved on to debug by testing PNG as code instead, using the LVGL 9 Image Converter.
Nothing happens on the screen, I have plenty of memory and the LVGL buffer is set to a whopping 4 Megabytes. The image I test is just a small one. Also made sure to set Opacity to 255.

This is the output, I've set the log to Warn.
```
[Info]	(6.374, +6374)	 lv_obj_create: begin lv_obj.c:215

[Info]	(6.374, +0)	 lv_image_create: begin lv_image.c:138

[Info]	(6.374, +0)	 image_decoder_get_info: Image decoder didn't set stride. Calculate it from width. lv_image_decoder.c:349

[Info]	(6.374, +0)	 lv_obj_create: begin lv_obj.c:215

[Info]	(6.375, +1)	 lv_label_create: begin lv_label.c:125

[Info]	(6.375, +0)	 lv_label_create: begin lv_label.c:125

[Info]	(6.375, +0)	 lv_label_create: begin lv_label.c:125

[Info]	(6.376, +1)	 lv_label_create: begin lv_label.c:125

[Info]	(6.376, +0)	 lv_label_create: begin lv_label.c:125
```

Am I the only one or does other people also have issues with Images and Gif's in V9?
I've used V8 since it was released and it works fine with images and Gif's, very happy with it in fact.

I can't for the life of me figure out what's going on.

### How to reproduce?

  LV_IMAGE_DECLARE(jpg);
  lv_obj_t * image = lv_img_create(lv_layer_top());
  lv_obj_set_style_opa(image , 255, 0);
  lv_img_set_src(image , &jpg);  // No return value to check
  lv_obj_align(image , LV_ALIGN_CENTER, 0, 0);

kisvegabor, 2024-11-10:
Hi,

Could you attach original and the whole converted image so that we can test it?

Have you tried it in a simulator?

Satnet, 2024-11-10:
Gabor, thanks for replying. I've kept testing and testing, I tried the TFF font by loading from C Array (works) but when loading from the SD Card the graphics on the screen just look like garbage, just dots and colors all over the place. Same happens with BMP loaded from SD Card. I am not sure if it's the Teensy file system, it should not be, but I cannot be sure.

The Gif used is this one, I've tried others and also used online tools to "clean" them just to be sure: 
[image]

Converted to Array:
https://paste.ee/p/L1V3v

The LVGL converter has been broken for a long time it seems,

kisvegabor, 2024-11-11:
Thank you! Just in case you can test it earlier than me: please try with this more complete [image converter script](https://github.com/lvgl/lvgl/blob/master/scripts/LVGLImage.py).

Satnet, 2024-11-13:
> Thank you! Just in case you can test it earlier than me: please try converting the image with this more complete [image converter script](https://github.com/lvgl/lvgl/blob/master/scripts/LVGLImage.py).

I have not been able to try this, I am on Windows and I've never played with Python.
If someone could send a GIF that is "correct" that I can place on my SD Card for testing.

What I do know is that the SD Card works fine for streaming audio files, it's the Teensy ecosystem.

kisvegabor, 2024-11-14:
I've converted it andn it works for me, although it's very slow even on my desktop and still don't know why. 

[banana.c.zip](https://github.com/user-attachments/files/17756668/banana.c.zip)

kisvegabor, 2024-11-14:
I see a lot of warnings like: `read_image_data: LZW table token overflows the frame buffer` which are coming deeply from the gif decoder library. I've reconverted the gif with ezgif.com but it didn't help. 

We are using [this library](https://github.com/lecram/gifdec) I've opened an issue [here](https://github.com/lecram/gifdec/issues/29).

Satnet, 2024-11-27:
> I see a lot of warnings like: `read_image_data: LZW table token overflows the frame buffer` which are coming deeply from the gif decoder library. I've reconverted the gif with ezgif.com but it didn't help.
> 
> We are using [this library](https://github.com/lecram/gifdec) and I've opened an issue [here](https://github.com/lecram/gifdec/issues/29).

Stupid question maybe but how come Gif's worked well in LVGL 8 for me. Perhaps LVGL 9 uses a newer version of gifdec ?

Satnet, 2024-12-12:
> We need some feedback on this issue.
> 
> Now we mark this as "Abandoned" because there was no activity here for 14 days.
> 
> Remove the "Stale" label or comment else this will be closed in 7 days.

Not abandoned, issue opened here by Gabor: https://github.com/lecram/gifdec/issues/29
Waiting on that.

huoxingdigua, 2024-12-14:
if(frm_off + str_len >= frm_size){
    LV_LOG_WARN("LZW table token overflows the frame buffer");
    return -1;
}
should be
if(frm_off + str_len > frm_size){
    LV_LOG_WARN("LZW table token overflows the frame buffer");
    return -1;
}

kisvegabor, 2025-01-03:
Not stale. @liamHowatt or @huoxingdigua could you open a PR to fix it?

innovandal, 2025-01-08:
FYI i tested the code snippet on top of 9.2.2 and it fixed the issue for me when viewing a gif

liamHowatt, 2025-01-28:
PR open: #7675

closed 2025-01-29 by uLipe