lvgl/lvgl #7274: lv_arc point rotation

andresvalda, 2024-11-11:
### LVGL version

v9.2.1

### What happened?

from 9.2.1 commit: 
3a96eaa08a7c895a7cd6b053871ab64b7091ac4d ignore hits that are outside drawn background arc 

we are unable to rotate the arc point from the PC using the mouse or from the device with a touch screen.

The problem disappears by modifying the lv_arc_angle_within_bg_bounds function, changing:

from while(bounds_angle >= 360) bounds_angle -= 360; to while(bounds_angle > 360) bounds_angle -= 360;

liamHowatt, 2024-11-12:
Thanks for opening the issue. I see some buggy behavior when clicking the bottom arc and it doesn't seem to disappear with the `while(bounds_angle > 360) bounds_angle -= 360;` change, for me.

To reproduce:

```c
static void arc_issue(void)
{
    lv_obj_t * arc = lv_arc_create(lv_screen_active());
    lv_obj_t * arc2 = lv_arc_create(lv_screen_active());

    lv_arc_set_value(arc, 0);
    lv_arc_set_value(arc2, 0);
    lv_obj_set_size(arc, 100, 100);
    lv_obj_set_size(arc2, 100, 100);
    lv_arc_set_bg_angles(arc, 20, 160);
    lv_arc_set_bg_angles(arc2, 200, 340);
    lv_obj_add_flag(arc, LV_OBJ_FLAG_ADV_HITTEST);
    lv_obj_add_flag(arc2, LV_OBJ_FLAG_ADV_HITTEST);
    lv_arc_set_value(arc, 10);
    lv_arc_set_value(arc2, 10);
    lv_arc_set_rotation(arc, 355);
    lv_arc_set_rotation(arc2, 355);
    lv_obj_center(arc);
    lv_obj_center(arc2);
}
```

kisvegabor, 2024-11-13:
cc @C47D

kisvegabor, 2024-12-02:
I've also noticed that clicking is not working at all when the the arc is full circle.

liamHowatt, 2024-12-03:
> I've also noticed that clicking is not working at all when the arc is a full circle.

Me too.

kisvegabor, 2024-12-18:
@liamHowatt can you take a look at it in the near future?

C47D, 2024-12-20:
Sorry, i didn't looked at my notifications. Full circles was taken into conskderation in the initial implementation, I can try to look at it when i come back from vacations in 2 weeks, sorry.

kisvegabor, 2024-12-22:
No worries, thank you Carlos!

kisvegabor, 2025-01-30:
@C47D will you have time for this in near future?

C47D, 2025-01-30:
> [@C47D](https://github.com/C47D) will you have time for this in the near future?

Hi, sorry I totally forgot about this issue. I'll take a look this weekend, we have holidays over here

C47D, 2025-02-15:
I've been having issues after updating my local repo. It's taking longer than expected to fix the compilation 😅 
Just for confirmation, the issue is that the arc indicator is not being updated when clicking 'inside' the arc?

C47D, 2025-02-15:
@kisvegabor For the full circle issue can you try this patch?

```patch
diff --git a/src/widgets/arc/lv_arc.c b/src/widgets/arc/lv_arc.c
index 1c566c566..16007e645 100644
--- a/src/widgets/arc/lv_arc.c
+++ b/src/widgets/arc/lv_arc.c
@@ -517,7 +517,7 @@ static void lv_arc_event(const lv_obj_class_t * class_p, lv_event_t * e)
         const lv_value_precise_t tolerance_deg = (360 * lv_dpx(50U)) / circumference;
         const uint32_t min_close_prev = (uint32_t) arc->min_close;
 
-        const bool is_angle_within_bg_bounds = lv_arc_angle_within_bg_bounds(obj, angle, tolerance_deg);
+        bool is_angle_within_bg_bounds = lv_arc_angle_within_bg_bounds(obj, angle, tolerance_deg);
         if(!is_angle_within_bg_bounds) {
             return;
         }
@@ -968,6 +968,12 @@ static bool lv_arc_angle_within_bg_bounds(lv_obj_t * obj, const lv_value_precise
 
     lv_value_precise_t bounds_angle = arc->bg_angle_end - arc->bg_angle_start;
 
+    if(arc->bg_angle_end + tolerance_deg > 360) {
+        arc->in_out = CLICK_INSIDE_BG_ANGLES;
+        arc->min_close = CLICK_CLOSER_TO_MIN_END;
+        return true;
+    }
+
     /* ensure the angle is in the range [0, 360) */
     while(bounds_angle < 0) bounds_angle += 360;
     while(bounds_angle >= 360) bounds_angle -= 360;
```

andresvalda, 2025-03-18:
Thank you all. Will this fix be available in v9.2.x?

kisvegabor, 2025-04-15:
I've opened https://github.com/lvgl/lvgl/pull/8106 which seems to working well for me. 

What do you think?

PR #8106 (2025-04-24): fix(arc): handle clicks on a full circle
fixes #7274
related to #7825

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/code-format.py` (`asty [... cut]
closed 2025-04-24 by liamHowatt