lvgl/lvgl #7275: Visualization of indexed images seems to be corrupted on ST when NEMA GFX is enabled

williams-one, 2024-11-11:
### LVGL version

v9.3

### What happened?

I have a simple application showing an image on a STM32U5 board.  I've enabled `LTDC` and `NEMA_GFX` in `lv_conf.h` and I'm testing the visualization of an image exported with the LVGLImage.py tool using different image format.

If the image is exported using e.g. RGB565 format it is displayed correctly
[image]

If the image is exported using e.g. I8 format it is not displayed correctly
[image]

If I set `NEMA_GFX` acceleration to `0`, then both images are displayed correctly

kisvegabor, 2024-11-17:
Hi,

Please share both the original and the converted image. 

cc @liamHowatt

williams-one, 2024-11-19:
Hi, thanks for the feedback.

Attached you can find the original image `background.png` and the converted images
* `background_i8.c` created using `python3 lvgl/scripts/LVGLImage.py hal_stm_lvgl/image/background.png -o hal_stm_lvgl/image --ofmt C --cf I8`
* `background_rgb565.c` created using `python3 lvgl/scripts/LVGLImage.py hal_stm_lvgl/image/background.png -o hal_stm_lvgl/image --ofmt C --cf RGB565`

[background](https://github.com/user-attachments/assets/3ca9db37-08e6-4081-bc7e-e0249d73a45f)
[background_i8.zip](https://github.com/user-attachments/files/17816172/background_i8.zip)
[background_rgb565.zip](https://github.com/user-attachments/files/17816173/background_rgb565.zip)

Not sure if this relates somehow, but I have also been looking around for additional information and I found conflicting information.
From this link https://support.touchgfx.com/docs/development/scenarios/touchgfx-on-gpu2d#neochrom-limitations it looks like that NeoChrom is not supporting L8 images.
But in Chapter 9 of "NemaGFX_API_Manual.pdf", it mentions the support for Look Up Table (LUT) based Textures that is configured using the function `nema_bind_lut_tex`

kisvegabor, 2024-11-27:
Thanks, we will try it out in a few days. 

@MarkIoanAmat might be also interested in this issue.

kisvegabor, 2024-12-22:
Not stale

kisvegabor, 2025-01-13:
Not stale

liamHowatt, 2025-01-29:
Thanks @williams-one for the tip about `nema_bind_lut_tex`. I've downloaded the manual too and I see what you mean.

This must be very close to what is needed to make it work. But unfortunately the result is still not correct. @MarkIoanAmat do you know what's missing? The parameters are all correct in the debugger.

(using hardcoded `NEMA_L8` for now)

```diff
diff --git a/src/draw/nema_gfx/lv_draw_nema_gfx_img.c b/src/draw/nema_gfx/lv_draw_nema_gfx_img.c
index 846602d69..a4d521a85 100644
--- a/src/draw/nema_gfx/lv_draw_nema_gfx_img.c
+++ b/src/draw/nema_gfx/lv_draw_nema_gfx_img.c
@@ -152,8 +152,14 @@ static void _draw_nema_gfx_img(lv_draw_unit_t * draw_unit, const lv_draw_image_d
                       lv_area_get_height(&(layer->buf_area)), dst_nema_cf,
                       lv_area_get_width(&(layer->buf_area))*lv_color_format_get_size(dst_cf));
 
-    nema_bind_src_tex((uintptr_t)(src_buf), tex_w, tex_h, src_nema_cf, src_stride,
-                      dsc->antialias ? NEMA_FILTER_BL : NEMA_FILTER_PS);
+    if(!LV_COLOR_FORMAT_IS_INDEXED(src_cf)) {
+        nema_bind_src_tex((uintptr_t)(src_buf), tex_w, tex_h, src_nema_cf, src_stride,
+                        dsc->antialias ? NEMA_FILTER_BL : NEMA_FILTER_PS);
+    }
+    else {
+        nema_bind_lut_tex((uintptr_t)(src_buf + LV_COLOR_INDEXED_PALETTE_SIZE(src_cf) * 4), tex_w, tex_h,
+        NEMA_L8, src_stride, dsc->antialias ? NEMA_FILTER_BL : NEMA_FILTER_PS, (uintptr_t)(src_buf), NEMA_BGRA8888);
+    }
 
     /*Guard for previous NemaGFX Version*/
 #ifdef NEMA_BLOP_RECOLOR
```

https://github.com/lvgl/lvgl/blob/d1302001bae07b708cff38333591ec4cabaa8b5a/libs/nema_gfx/include/nema_graphics.h#L403-L420

From the manual

> ### 5.2.2 Look Up Table (LUT) based Textures
> NEMA| GFX provides the option to use LUTs (look up tables) in texture blitting.
To translate an image to a LUT, a color palette table and an indices table are needed.
The supported palettes are 2,4,16,256 colors. Color palette format can be any supported
by NEMA| GFX color format. The indices table format is L* and the size depends on the
color palette table format.
For example, when you use a 4 color palette, the index needs 2 bits for representation.
So the index table needs to be either of L2 or L2 little endian. So, the supported indices
tables formats are:
• NEMA_L1,NEMA_L1LE
• NEMA_L2,NEMA_L2LE
• NEMA_L4,NEMA_L4LE
• NEMA_L8
NEMA| pix-presso has the functionality to transform a texture to palette and indices
tables (target format LUT-2, LUT-4, LUT-16, LUT-256). For further information, refer to the
NEMA| pix-presso manual. NEMA| pix-presso produces the palette table to .rgba format
and the indices table to .gray format.
The following API call is used to bind the color palette and the indices table: ne-
ma_bind_lut_tex(). In Table 3 the relation between color palette table and indices ta-
ble is mentioned.

Get the manual by submitting the form, confirming your email, and they will send the download immediately: https://www.think-silicon.com/nema-gfx-api

[image]

williams-one, 2025-02-07:
Today I spent some time trying to understand the issue based on the comments and work by @liamHowatt.
I was able to make some steps forward...  Sort of...  But not not with `I8` images exported by LVGLImage...
I try to describe anyway what I did since it may give you some ideas. Sorry for not using the same image above...  If you think it will help I could do the same steps with the original image

I have downloaded the official tool from ThinkSilicon to convert images for NEMA https://www.think-silicon.com/nema-pix-presso

I have used this image 

[image]

I have converted the image using the pixpresso tool and exported it as a header file

[image]

That gave the following two files

[image-header.zip](https://github.com/user-attachments/files/18711193/image-header.zip)

Now I have injected directly those data in the `nema_bind_lut_tex` in the following way.  The important bit was also adding `blending_mode |= NEMA_BLOP_LUT;`

```c
        nema_bind_lut_tex((uintptr_t)internal_1_800x480_indices, tex_w, tex_h,
        		NEMA_L8, src_stride, NEMA_FILTER_PS, (uintptr_t)internal_1_800x480_palette, NEMA_RGBA8888);  // Palette exported by pixpresso is RGBA
    	blending_mode |= NEMA_BLOP_LUT;  // <-- Without this nothing works
```

I have also removed `NEMA_FILTER_BL`, since the manual for `nema_bind_lut_tex` says (not sure if it matters though...)

[image]

With all this changes the image showed up correctly on the screen.

[image]

However if I just use an I8 image exported by LVGLImage using the code that Liam showed, I still get some garbage on the screen...  Or at least some colored garbage after the addition of NEMA_BLOP_LUT 😄

[image]

williams-one, 2025-02-08:
I'm back with another iteration that fixed the problem I think 😃 

I have created a very simple image 11x6 pixel wide and exported it with `LVGLImage` as `I8` and with Pixpresso as `LUT-256`

[image]

This was the output

**LVGLImage**
```c
uint8_t test_minimal_map[] = {
    // palette
    0x00,0xff,0x00,0xff,    // [0x00]: BGRA green
    0x00,0x00,0xff,0xff,    // [0x01]: BGRA red
    0xff,0xff,0xff,0xff,    // [0x02]: BGRA white
    0xff,0x00,0x00,0xff,    // [0x03]: BGRA blue
    0xff,0xff,0xff,0x00,    // ...

    // indexes
    0x02,0x02,0x02,0x02,0x02,0x02,0x02,0x02,0x02,0x02,0x02,
    0x02,0x00,0x00,0x00,0x00,0x00,0x02,0x02,0x02,0x01,0x02,
    0x02,0x01,0x01,0x01,0x01,0x01,0x02,0x02,0x00,0x01,0x02,
    0x02,0x02,0x02,0x02,0x02,0x02,0x02,0x03,0x00,0x01,0x02,
    0x02,0x03,0x03,0x03,0x03,0x03,0x02,0x03,0x00,0x02,0x02,
    0x02,0x02,0x02,0x02,0x02,0x02,0x02,0x03,0x02,0x02,0x02,
};
```

**Pixpresso**
```c
// palette
const unsigned char test_minimal_palette[] = {
0xff, 0xff, 0xff, 0xff,     // [0x00]: RGBA white
// ...
0xff, 0x00, 0x00, 0xff,     // [0x10]: RGBA red
// ...
0x00, 0x00, 0xff, 0xff,     // [0x20]: RGBA blue
// ...
0x00, 0xff, 0x00, 0xff,     // [0x30]: RGBA green
};

// indexes
const unsigned char test_minimal_indices[] = {
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
0x00,0x03,0x03,0x03,0x03,0x03,0x00,0x00,0x00,0x01,0x00,
0x00,0x01,0x01,0x01,0x01,0x01,0x00,0x00,0x03,0x01,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x02,0x03,0x01,0x00,
0x00,0x02,0x02,0x02,0x02,0x02,0x00,0x02,0x03,0x00,0x00,
0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x02,0x00,0x00,0x00
};
```

Now the problem is clear.  For some reasons `NEMA_L8` format semms to be assuming that the color with index `0x03` for example is in position `0x30` of the palette array.

Changing the export script to swap high and low nibbles as described below
https://github.com/lvgl/lvgl/blob/b9e0c4daaf6f4c86ccb8bbe0a0e7710900dccda1/scripts/LVGLImage.py#L919-L922

```python
        if cf == ColorFormat.I8:
            for e in rows:
                e = bytearray((x >> 4) | ((x & 0x0F) << 4) for x in e)
                rawdata += e
```

And changing the code in `lv_draw_nema_gfx_img.c` as suggested by @liamHowatt in the post above and adding `blending_mode |= NEMA_BLOP_LUT` as follows
```c
    if(!LV_COLOR_FORMAT_IS_INDEXED(src_cf)) {
        nema_bind_src_tex((uintptr_t)(src_buf), tex_w, tex_h, src_nema_cf, src_stride,
                        dsc->antialias ? NEMA_FILTER_BL : NEMA_FILTER_PS);
    }
    else {
       nema_bind_lut_tex((uintptr_t)((uint8_t*)src_buf + LV_COLOR_INDEXED_PALETTE_SIZE(src_cf) * 4), tex_w, tex_h,
       		NEMA_L8, src_stride, NEMA_FILTER_PS, (uintptr_t)(src_buf), NEMA_BGRA8888);
    	blending_mode |= NEMA_BLOP_LUT; // <-- This is the important bit
    }
```

That seemed to do the trick 😄 

I have also removed `NEMA_FILTER_BL`, since the manual for `nema_bind_lut_tex` says the following (not sure if it matters though...) 
https://github.com/lvgl/lvgl/blob/b9e0c4daaf6f4c86ccb8bbe0a0e7710900dccda1/libs/nema_gfx/include/nema_graphics.h#L410-L411

I guess that maybe one solution could be to add an option to the `LVGLImage` script to specify that NEMA is used and in that case if I8 images are exported just swap the nibbles in the indexes

williams-one, 2025-02-10:
I have opened this PR https://github.com/lvgl/lvgl/pull/7744 to better show what I had in mind.
I have tested it with I8, I4, I2 and I1 images and it seems to be working correctly.
I'm not sure if the `--nemagfx` option in the exporter is the best solution.  Or if it would be better something like a `I8_NEMA`

In any case, I'm waiting for your feedback :)

closed 2025-02-12 by liamHowatt

KlemenDEV, 2025-06-11:
Hi! Does anyone here have a download link or the file for PIX-Presso or the TSC converter binary? https://www.think-silicon.com/ domain is gone, and so are all the downloads?

PR #7744 (2025-02-12): fix(nema): fix indexed image error
Fixes #7275

The visualization of indexed images (in particular I8 images) is corrupt when NEMA acceleration is enabled.

The root cause of the problem seems to be related to the way in which NEMA indexes the color palette in I8 format.  See the following comments for a detailed explanation
* https://github.com/lvgl/lvgl/issues/7275#issuecomment-2644550771
* https://github.com/lvgl/lvgl/issues/7275#issuecomment-2643781026

Based on the analysis, NEMA_L8 format seems to be assuming that for example the color with index 0x03 is in position 0x30 of the palette a [... cut]