lvgl/lvgl #7276: [NemaGFX] Spinner animation rotates strangley

KlemenDEV, 2024-11-11:
### LVGL version

current master branch

### What happened?

We use code from https://docs.lvgl.io/master/widgets/spinner.html to define the spinner, a completely default implementation.

The animation looks like this when LV_USE_NEMA_GFX is 1, no OS enabled:

[image]

liamHowatt, 2024-11-12:
Thanks, will take a look soon!

KlemenDEV, 2024-11-13:
Perfect, if you need any more info, let me know.

I am testing master branch on STM32U5G9 with DSI and LTDC

KlemenDEV, 2024-11-13:
I know this is unrelated, but I figured this may be a place to ask since there is no "Discussions" section on this repo.

https://docs.lvgl.io/master/details/integration/renderers/nema_gfx.html says OS support is experimental. What part should I be careful of, since I tried it and it seemed to work at first glance (using LV_OS_CMSIS_RTOS2).

But still this bug reported in this ticket happens with and without OS

KlemenDEV, 2024-11-15:
#7307 fixes this issue. Thank you for a quick fix! :)

PR #7307 (2024-11-18): Nema fix arc issue and update library (do not squash)
Fixes #7276

Fix #7276:

- Shift arc angles for Nema to draw arcs correctly.

Update the Nema library:

- lv_draw_nema_gfx now accepts arc draw tasks with non-rounded ends.
- Contents of guards added in #7245 will now be compiled
  - `NEMA_TSC6AP` is now defined.
  - `NEMA_BLOP_RECOLOR` is now defined

cc @MarkIoanAmat

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blo [... cut]
closed 2024-11-18 by XuNeo