lvgl/lvgl #7279: Use a single ID for both task->preferred_draw_unit and parallel debugging

cristian-stoica, 2024-11-12:
### Introduce the problem

- Currently draw units use macros like DRAW_UNIT_ID_SW, DRAW_UNIT_ID_PXP etc. for self-identification when setting task->preferred_draw_unit to themselves. There are also LV_DRAW_UNIT_NONE (0) and a LV_DRAW_UNIT_IDLE (-1)

- SW draw unit has a single ID (DRAW_UNIT_ID_SW) although it can instantiate more than one draw units. Although not used, it is setting for itself an ->idx member with values starting from 0 to LV_DRAW_SW_DRAW_UNIT_CNT-1

- When parallel debug is enabled, objects on screen get a numbered and colored overlay for easier identification. The number is a draw unit number that has no relation with the above macros or with the ->idx of the SW unit. Instead, the number is the position of the acting draw unit in the linked list of initialized draw units.

Although not exactly a subject for frequent confusion, having a self-assigned ID for draw unit identification, a set and not used sw->idx value and an arbitrary list ID for debug are an opportunity for simplification.

### Proposal

My question is whether a single draw unit ID initialized during `lv_draw_create_unit` would serve all current purposes.

Initialization:
```
--- a/src/draw/lv_draw.c
+++ b/src/draw/lv_draw.c
@@ -82,6 +82,9 @@ void * lv_draw_create_unit(size_t size)
     lv_draw_unit_t * new_unit = lv_malloc_zeroed(size);
     LV_ASSERT_MALLOC(new_unit);
     new_unit->next = _draw_info.unit_head;
     _draw_info.unit_head = new_unit;
     _draw_info.unit_cnt++;
+
+    new_unit->idx = _draw_info.unit_cnt;
+
```

Parallel debug:
```
@@ -581,12 +580,8 @@ static void execute_drawing(lv_draw_sw_unit_t * u)
         lv_area_t draw_area;
         if(!lv_area_intersect(&draw_area, &t->area, u->base_unit.clip_area)) return;
 
-        int32_t idx = 0;
-        lv_draw_unit_t * draw_unit_tmp = _draw_info.unit_head;
-        while(draw_unit_tmp != (lv_draw_unit_t *)u) {
-            draw_unit_tmp = draw_unit_tmp->next;
-            idx++;
-        }
+        int32_t idx = u->base_unit.idx;
```

evaluate / other
```
@@ -455,7 +455,7 @@ static int32_t evaluate(lv_draw_unit_t * draw_unit, lv_draw_task_t * task)
 
     if(task->preference_score >= 100) {
         task->preference_score = 100;
-        task->preferred_draw_unit_id = DRAW_UNIT_ID_SW;
+        task->preferred_draw_unit_id = draw_unit->idx;
     }
 ```

kisvegabor, 2024-11-17:
Imagine that there are 2 SW draw units. During evaluation or dispatching they should be treated equal. With the suggested change we will assign all draw tasks to either the first or second core. 

E.g. in 
```c
    lv_draw_task_t * t = NULL;
    t = lv_draw_get_next_available_task(layer, NULL, DRAW_UNIT_ID_SW);
```

All SW draw unit related tasks are fine, not matter which SW draw unit asks them.

cristian-stoica, 2024-11-18:
> All SW draw unit related tasks are fine, not matter which SW draw unit asks them.

Indeed, using the draw_unit->id for preferred_task will not preserve this behavior, however it should not lock the task execution to one core because the dispatch will check for the same ID that evaluate set previously - a different ID for each unit, but the same on the same unit.

On the other hand, it prevents sw draw units to help with the tasks unless they evaluated them themselves - not ideal.

If we drop this part of the question, a patch that deals with the rest is pushed for review in another pull request:
https://github.com/lvgl/lvgl/pull/7240/commits/01f477745bd9eb015ef35fa736e7b6fbd3b817c4

kisvegabor, 2024-11-21:
> Indeed, using the draw_unit->id for preferred_task will not preserve this behavior, however it should not lock the task execution to one core because the dispatch will check for the same ID that evaluate set previously - a different ID for each unit, but the same on the same unit.

If we have 2 SW draw units with ID 1 and 2 how can evaluate decide which ID to assign to a draw task?

cristian-stoica, 2024-11-22:
> If we have 2 SW draw units with ID 1 and 2 how can evaluate decide which ID to assign to a draw task?

I see what you mean now. If the task was taken by the first unit in the list, the next ones will take it only if they assume a stronger preference. For SW that would mean the second unit will not even get the chance to take a task unless a different round-robin mechanism would be implemented - which makes no sense.

kisvegabor, 2024-11-22:
Yes, that's it.

I can't really imagine cases other then SW render where we can have multiple draw units of the same kind. So maybe we can do some special handling with SW rendering.

closed 2024-12-17 by lvgl-bot