lvgl/lvgl #7282: Does LVGL support something like Soft hyphen to break words?

Viatorus, 2024-11-12:
### Introduce the problem

See: https://en.wikipedia.org/wiki/Soft_hyphen

kisvegabor, 2024-11-17:
Just to be sure that I understand it correctly. With such an update we could write a string like
```
fantastically
fan-tas-ti-cal-ly
fan [SHY] tas [SHY] ti [SHY] cal [SHY] ly
```   
As LVGL  performs work wrapping it should add as many syllables as possible and also show a `-` if needed.

If so, I think we need to do these things to make it work:
1. Improve [`lv_text_get_next_word`](https://github.com/lvgl/lvgl/blob/master/src/misc/lv_text.c#L201) to support SHY. Also take into account that the `-` will take space.
2. Somehewre [here](https://github.com/lvgl/lvgl/blob/master/src/draw/lv_draw_label.c#L437) check if the last character is a SHY and if so manually draw a `-`.

If you agree, would you be interested in working on it?

liamHowatt, 2024-11-25:
So there would be some rules that would decide where in "`seasaw`" the word is allowed to be broken? The user would still call `label_set_text("seasaw")`, but it could be displayed as below?

```
sea-
saw
```

Viatorus, 2024-11-25:
@kisvegabor Yes, I think you understand it correctly.

I am interested on working on it, yes. But there are to ways for me to solve this: with [SHY] and a new LVGL feature or by providing multiple text translation which may or may not contain an explicit `-` to manually force a line break.

I have to think about it. The first solution seems to be more reliable but time consuming... :(

@liamHowatt No, the user has to write `label_set_text("sea" "\xad" "saw")`.

kisvegabor, 2024-11-28:
Great! Please let me know if you need anything else from our side.

closed 2024-12-25 by lvgl-bot