lvgl/lvgl #7283: How to stop lottie animation and change the animation file to play again?

SSEHX, 2024-11-12:
### LVGL version

v9...

### What happened?

We need to play several different lottie animations according to our requirements, and we need to switch lottie json files. At present, the file system has been docked on esp32p4 platform and is running normally. How can I stop lottie animations and switch files to play again? It seems that no such api is provided.

kisvegabor, 2024-11-20:
Hi,

You can just call  `lv_lottie_set_src_data/file` again with a different source.

SSEHX, 2024-12-04:
After testing with a computer LVGL simulator, it is impossible to switch animations through this API interface. Are there any other interfaces that can be used? Thank you!

kisvegabor, 2024-12-22:
Please send a code snippet to reproduce the issue.

closed 2025-01-18 by lvgl-bot