lvgl/lvgl #7285: Scale -- Suggested Refactor

vwheeler63, 2024-11-13:
### Introduce the problem

- `_lv_scale_section_t.minor_range` and
- `_lv_scale_section_t.major_range`

should really be called

- `_lv_scale_section_t.range_min` and
- `_lv_scale_section_t.range_max`

because their use is exclusively and only to compare against `_lv_scale_t.range_min` and `_lv_scale_t.range_max` respectively, and they have the *exact same semantics*.

To put it simply, `minor_range` and `major_range` are names chosen early by the original writer where more perfect, more appropriate names are found later, that more accurately reflect the meaning of their content.  (As you all know, this is a common occurrence in software development, and is a classic case for refactoring by renaming fields of a custom data type.)

C47D, 2024-11-14:
Hi @vwheeler63 ,

Thanks for reading through the `lv_scale` mess, I agree with your observations and I would like to ask @kisvegabor if we can proceed with the updates, I'm not really aware of the release schedule.

vwheeler63, 2024-11-14:
> mess

Please don't read this as "fault finding"!  I've developed several dozen components like this (for different environments) and can attest they go through idea phases and change.  This simply looks like part of its evolution to me.  👍  And my proposal here doesn't break the API, so if Gábor likes it, I would think we could do it immediately.

kisvegabor, 2024-11-14:
Hey,

The rename and type change make perfect sense to me, however I'm a little bit hesitant about the default value. 
Usually we use a default value which is already visible. E.g. the default label text is `"Text"` instead of `""`. However, in this case an not visible section can make sense too, especially if it's documented in
- the docs
- and `lv_scale_add_section` API comments (e.g. `lv_scale_section_set range` needs to be called to set a valid range)

We ave plenty of time until v9.3 (2025 Q1), so it's not a concern.

vwheeler63, 2024-11-14:
> The rename and type change make perfect sense to me, however....

Hi, Gábor!  

I'm glad you like the idea.  Default range of [0..0] can also make sense, since it doesn't have any styles added to it yet, it still won't be drawn.  So the existing initialization value does no harm and it cannot "accidentally" cause something to be drawn that was not intended.

When you say, "it's not a concern" -- does that mean it is okay to implement now?

Kind regards,
Vic

C47D, 2024-11-15:
> Please don't read this as "fault finding"! I've developed several dozen components like this (for different environments) and can attest they go through idea phases and change. This simply looks like part of its evolution to me.

No worries, I see it as an evolution of the SW. I will start to work on this in the weekend, I'm in a tight schedule at work right now :smile:

vwheeler63, 2024-11-15:
@C47D

If it helps, I can crank it out in about 5 minutes.

vwheeler63, 2024-11-15:
@kisvegabor I found out something important!  Even with no styles set, the [0..0] range STILL causes it to be drawn, as it inherits the styles from the Scale!  It isn't visible, but the drawing *is being executed*!  So initial range [0..-1] is actually a *correct* way to initialize it since it eliminates that CPU overhead until the user has set it otherwise.

C47D, 2024-11-15:
> @C47D
> 
> If it helps, I can crank it out in about 5 minutes.

Sure, you can take it 👍

vwheeler63, 2024-11-15:
> Sure, you can take it

Okay!  I'm currently awaiting another PR to commit where `lv_scale` got involved (PR #7277), and will add these refactors afterwards.

vwheeler63, 2024-11-25:
@kisvegabor @C47D  Note that I just added item `#D` in my proposal above -- I ran into it while I was clarifying the documentation in `lv_scale.h` -- when reading the documentation for those items made no sense to me.  So I clarified it by seeing how they were used in the C code and updated the documentation accordingly to be crisply clear about what those functions (and struct member) do.

kisvegabor, 2024-11-26:
> rename member field `angle_range` => `needle_max_angle`  (which is what it actually is)

I don't think it's a good idea. The needle is optional for the scale. So we should describe the scale in general, and if there is a needle it can use these values. 

Am I missing something?

vwheeler63, 2024-11-27:
> > rename member field `angle_range` => `needle_max_angle`  (which is what it actually is)
> 
> I don't think it's a good idea. The needle is optional for the scale. So we should describe the scale in general, and if there is a needle it can use these values.
> 
> Am I missing something?

You are actually correct!  I discovered in the process of updating the documentation that the name "angle_range" is actually correct, but it was not well documented.  It is now well documented in my PR, and I did not change the names.  I will remove "D" from the above because it was not valid.

PR #7362 (2024-12-02): fix(scale): fix angle calculation error
This step started as an implementation of the refactor recommended in #7285.  In doing so, I also documented the fields of `struct _lv_scale_t` and `struct _lv_scale_section_t` since I was now intimately familiar with them.  In the process of updating the documentation in `lv_scale.h`, I discovered:

- There was an important bug in `lv_scale_set_rotation()` that caused part of the main scale line to be omitted in the drawing when negative values were passed, however drawing worked as intended for all values in range [0..360].  So instead of solving the d [... cut]
closed 2024-12-02 by FASTSHIFT