lvgl/lvgl #7286: Load fonts and images with offset (external flash dual imaging)

bhspyder, 2024-11-13:
### Problem to solve

I have a use case where the external flash which is used to store large assets such as fonts and images is being split into 2 different slots to support OTA updates. When one slot is in use by the running firmware, the new update is written to the other slot. Once it has been successfully written, the bootloader switches to the updated slot, leaving the previous slot as a backup in case there is an issue. This of course offsets all of the memory addresses of the data stored in external flash depending on which slot is active. This only affects rodata as the text section is copied from external flash to internal and executed from a consistent address. The only thing that needs to change are the memory addresses used to access the data that is loaded in place from external flash.

I have a fork of 8.4 that is currently doing this in a rudimentary way, but would prefer to use the main stream and it seems like functionality that others may desire as well.

### Success criteria

Provide the ability to set an address offset at runtime that LVGL will append to the addresses of fonts and images.

### Solution outline

Implement the following functions:
```C
void lv_resource_set_offset(uint32_t offset); // set global resource offset (lv_resource_offset)
uint32_t lv_resource_get_offset(void); // get global resource offset (lv_resource_offset)
```

Notes:
* Naming the functions/variables and finding a home for this that make sense is likely going to be the hardest part
* The global resource offset should be 0 by default
* Ability to enable/disable this functionality with a flag in lv_conf.h
* Add the global offset to the memory address of images and fonts
* 

### Rabbit holes

Images are relatively easy to add the offset to (Just add it to .data of lv_img_dsc_t before loading), but fonts (lv_font_t) have a number of structures pointing to other structures that would need to be offset.

### Testing

Store an image and font in 2 different locations of memory with a known offset. Use the setter function to set the offset and see that LVGL successfully loads the image and label (using the font) from both locations (0 and the offset).

### Teaching

Some basic documentation explaining the use case and how to use the feature should suffice.

kisvegabor, 2024-11-14:
I'm don't fully understand the memory layout of your app. Is it like:
- bootloader
- slot1:
  - app
  - images
  - fonts
- slot2:
  - app
  - images
  - fonts  
  
  or
- bootloader
- slot1:
  - app
- slot2:
  - app
- images
- fonts

bhspyder, 2024-11-15:
It is closer to the first:

## Internal Flash
* bootloader
* app (copied from active slot in external flash)

## External Flash 32 MB
### Slot 1: (0x900000000, 16 MB)
* app (copied to consistent location in internal flash)
* images (used in place; offset from originally linked location)
* fonts (used in place; offset from originally linked location)
### Slot 2: (0x91000000, 16 MB)
* app  (copied to consistent location in internal flash)
* images (used in place; offset from originally linked location)
* fonts (used in place; offset from originally linked location)

Example: An image linked and stored at address 0x900000100 would actually be stored at 0x910000100 if slot 2 is active. So an offset of 0x1000000 would need to be added to the image pointer before using it.

kisvegabor, 2024-11-20:
Thanks! Maybe I'm still missing something, but how could the same app work in a different location. E.g. `lv_label_set_text(lebel "Hello");` will compile to e.g. `lv_label_set_text(label, 0x900000400)` which is not valid in case of slot 2.

bhspyder, 2024-11-20:
The app is linked for the internal flash, but stored in external flash so all of the memory addresses for rodata (like the string "Hello") or code in the text section will be internal flash addresses. The bootloader copies the app section from external flash to the location in internal flash that the application was linked for at compile time and executes it from there. Images and fonts (since they are larger than what will fit in internal flash) will be the only data used in place in external flash and will need to be offset.

kisvegabor, 2024-11-28:
I perfectly understand now, thank you. As this is the first time such a request came up, I'd want to wait a little to se if others might be interested in it too. 

I don't know exactly how you workaround it now, but on startup I'd create new `lv_image_dsc_t`s and `lv_font_t`s in RAM (just the descriptors so it's only a few bytes), copy the `rodata` structures there and add the offset to the pointer. After it just use the RAM image and font descriptor in the app.  E.g.
```c
lv_image_dsc_t img1_ext = img1;
img1_ext.map += offset;

lv_image_set_src(logo, &img1_ext);
```

bhspyder, 2024-11-28:
The only issue with that approach is that it doesn't play well with code generators (SquareLine for example). I was thinking something more global, but maybe this is to niche of a thing to support something like that natively. I am currently working around it by patching the pointer access in the lvgl library. Images for example:
```
void lv_draw_img_decoded(lv_draw_ctx_t * draw_ctx, const lv_draw_img_dsc_t * dsc,
                         const lv_area_t * coords, const uint8_t * map_p, lv_img_cf_t color_format)
{
    if(draw_ctx->draw_img_decoded == NULL) return;

    draw_ctx->draw_img_decoded(draw_ctx, dsc, coords, GUI_DATA_OFFSET_PTR(map_p), color_format);
}
```
**GUI_DATA_OFFSET_PTR** is just a macro that adds the offset to the pointer. It's a hack, but it does meet the needs of my use case.

closed 2024-12-28 by lvgl-bot

kisvegabor, 2025-01-03:
Sorry for the delay. Before adding something like this, I'd like to wait a little to see if other people will also report similar requests. 

> The only issue with that approach is that it doesn't play well with code generators (SquareLine for example).

I see the problem. SquareLine generates and uses `img1` but it should use `img1_ext`. It seems like something that we might support in LVGL's UI editor (still under development). See a preview video here: https://www.youtube.com/watch?v=ntAuLhuK3Ck

bhspyder, 2025-01-03:
No problem. We can pursue this another time if there is more buy in. For now, I can continue to use my workaround.