lvgl/lvgl #7297: Remove links from documentation to "lvgl.100ask.net" server or show only in China region

PGNetHun, 2024-11-14:
### Introduce the problem

In the documentation every page has a link to "lvgl.100ask.net" server, which is hosted in China: 
https://ipgeolocation.io/what-is-my-ip/lvgl.100ask.net

Clicking on a link (from Europe) opens nginx error page 404 from that chinese server:

[image]

[image]

### Proposal

**TODO:**
1. Do not add links to documentation that are not accessible from everywhere (a simple bot/script could go through links to verify that they direct to still available sites)
2. Keep links that point to **only verified sites**, remove any suspicious links (to avoid browser based viruses and attack types)
3. We need to regularly verify links added to documentation
 
LVGL documentation (and source code) should not be compromised to possible attacks and misuse (like XZ was).

kisvegabor, 2024-11-15:
It makes sense to add a link only on the home page. It would me the docs more resilient. Now it's broken as a lot of sites were moved to a new URL and it was not followed in the Chinese docs. 

Adding a more docs checks to the CI is definitely on the roadmap. cc @vwheeler63 

> LVGL documentation (and source code) should not be compromised to possible attacks and misuse (like XZ was).

Could you tell what was XZ issue? I think I haven't heard about it.

cc @100ask

vwheeler63, 2024-11-15:
@kisvegabor Thank you for the CC.  Do we need to remove some of the translation links?  I am coming up to speed now in Python so am well equipped to do this if needed, or have the translation links appear only on selected docs if needed.  Your call.

100ask, 2024-11-16:
This issue is caused by the document not being synchronized yet. I will synchronize it as soon as possible in the next few days.

100ask, 2024-11-16:
> which is hosted in China

Great suggestion! I will try it.

kisvegabor, 2024-11-17:
@100ask Thank you! However, if I think about it more it will cause a constant maintenance issues as pages are moved or created.

Having a single link on the home page should help the users to find the Chinese docs and it would be way simpler. What do you think?

kisvegabor, 2024-11-17:
@vwheeler63 Do we need to remove some of the translation links? 

Let's discuss what would be the best move :slightly_smiling_face:

M1NGS, 2024-11-18:
> This issue is caused by the document not being [synchronized](https://github.com/100askTeam/100ask_lvgl_docs) yet. I will synchronize it as soon as possible in the next few days.

首页js有问题，会卡死浏览器

100ask, 2024-11-18:
> > This issue is caused by the document not being [synchronized](https://github.com/100askTeam/100ask_lvgl_docs) yet. I will synchronize it as soon as possible in the next few days.
> 
> lvgl.100ask.net首页js有问题，会卡死浏览器

[这里](https://github.com/100askTeam/100ask_lvgl_docs) 只是LVGL库的副本，我们只是添加了中文翻译而已，之前`https://docs.lvgl.io/master`也有这个问题，同步之后再看有没有这个问题。
[This ](https://github.com/100askTeam/100ask_lvgl_docs) is just a copy of the LVGL docs, we just added a Chinese translation. `https://docs.lvgl.io/master` also had this issue before, we will check if there is any problem after synchronization.

100ask, 2024-11-18:
> @100ask Thank you! However, if I think about it more it will cause a constant maintenance issues as pages are moved or created.
> 
> Having a single link on the home page should help the users to find the Chinese docs and it would be way simpler. What do you think?

@kisvegabor  @vwheeler63 
I think we can add two buttons here and I will try. What do you think?

[image]

vwheeler63, 2024-11-18:
> What do you think?

Through what mechanism?  (I'm unfamiliar with such things at my stage of web learning.)  Would that be through a Sphinx extension?

100ask, 2024-11-20:
> > What do you think?
> 
> Through what mechanism? (I'm unfamiliar with such things at my stage of web learning.) Would that be through a Sphinx extension?

Yes, I will try to write a Sphinx extension.

kisvegabor, 2024-11-20:
What about just making it more prominent:
[image]

Adding the Chinese link to all site on such a large button can be disruptive for non-Chinese readers, while the Chinese people should find it on the home page too.

PGNetHun, 2024-11-21:
@100ask - adding language button to FAB ([Floating Action Buttons](https://m3.material.io/components/floating-action-button/overview)) would be also OK if there would be more translations. I guess the FAB button would redirect to a new page where user would select language from a list of buttons, in our case English or Chinese. 
My personal preference is having the language selector button (or link) on home page or in top bar of page or in footer (as most websites implement it).
**But it's up to you how to add language button.**

@kisvegabor - about the security concerns:
Unfortunately it seems cyber attacks and "hijacks" increased recently, Github and even open source projects on Github were also affected, so it is important to protect LVGL community by checking what goes into the code / docs / github actions. 
Obviously not the LVGL runtime is directly compromittable (as stability depends on host code), but the links to suspicious websites in docs/code, or 3rd party libraries used by code, and 3rd party packages used in github actions and local dev env build process. 
**Basically malwares and viruses (from not verified sources) could infect the machines of developers.**
For example there were malwares in NodeJS/Nuget packages, but there could be also malware in Python packages (`pip` packages), which are used (if used) in local dev environment build process.
Unfortunately that's one additional step that reviewers need to verify during PRs, that nothing suspicious goes into the git repo.
LVGL is amazing, and I hope more and more makers and business partners discover it and use it.

PS:
About the XZ: https://www.youtube.com/watch?v=MllrK4XSJxc
(it's also worth noting the sophisticated social engineering)

kisvegabor, 2024-11-22:
Thank you @PGNetHun!

100ask, 2024-11-27:
> This issue is caused by the document not being [synchronized](https://github.com/100askTeam/100ask_lvgl_docs) yet. I will synchronize it as soon as possible in the next few days.

@PGNetHun @kisvegabor Synchronized already.

100ask, 2024-11-27:
> Adding the Chinese link to all site on such a large button can be disruptive for non-Chinese readers, while the Chinese people should find it on the home page too.

@kisvegabor @PGNetHun How about having a Chinese translation link only on the [homepage](https://docs.lvgl.io/master/index.html)?

kisvegabor, 2024-11-28:
> How about having a Chinese translation link only on the [homepage](https://docs.lvgl.io/master/index.html)? This ensures that the link will not be lost.

Okay for me!

vwheeler63, 2024-11-28:
> > How about having a Chinese translation link only on the [homepage](https://docs.lvgl.io/master/index.html)? This ensures that the link will not be lost.
> 
> Okay for me!

Can do!

PGNetHun, 2024-11-28:
> > Adding the Chinese link to all site on such a large button can be disruptive for non-Chinese readers, while the Chinese people should find it on the home page too.
> 
> @kisvegabor @PGNetHun How about having a Chinese translation link only on the [homepage](https://docs.lvgl.io/master/index.html)? This ensures that the link will not be lost.

Yes, that is a good idea. 
Please CC me also in PR.
Thank you!

liamHowatt, 2024-12-03:
Fixed by #7370

closed 2024-12-03 by liamHowatt

kisvegabor, 2024-12-03:
Thank you all!