lvgl/lvgl #7298: lv_binding_rust development stale

AlixANNERAUD, 2024-11-14:
### Introduce the problem

Hi,  
It appears that the development of `lv_binding_rust` has stalled:  
- The last commit was five months ago.  
- It has been broken for the same duration.  
- There are still major bugs.  
- There's no plan to upgrade to v9 in the near future.  

To be clear, this isn't a criticism,I completely understand that people have many other priorities. I just wanted to start a conversation about the future of Rust support for LVGL.

### Proposal

As mentioned [here](https://github.com/lvgl/lv_binding_rust/issues/176), decoupling the `sys` part of Rust LVGL (which includes LVGL and the build script) from the generated bindings is a simple solution that allows users to work with LVGL without concerns about the build script. I have already done that here: [lvgl_rust_sys](https://github.com/Xila-Project/lvgl_rust_sys). That would solve a part of the problem.

kisvegabor, 2024-11-17:
Thank you for opening this issue. 

The Rust binding is really community maintained now. Although I think it would have a great value to have an LVGL Rust binding I cannot assign an LVGL employee to it in the near future. 

@nia-e please comment on your plans with `lv_binding_rust`. If your priorities have been shifted towards other things, it's also find, but in this case we should look for new maintainers.

C47D, 2024-11-20:
@AlixANNERAUD Hi, Is moving the sys part of Rust LVGL to a separated repo the first step of many others? If so, is there any low hanging fruit for me to start fixing issues in the binding repo?

AlixANNERAUD, 2024-11-20:
> @AlixANNERAUD Hi, Is moving the sys part of Rust LVGL to a separated repo the first step of many others? If so, is there any low hanging fruit for me to start fixing issues in the binding repo?

Hi, yes, this approach would allow people who want to use the latest version of LVGL or aren't concerned about the generated bindings to proceed. It's a quick fix. For the complete Rust bindings, please refer to the PRs and issues at [lvgl_binding_rust](https://github.com/lvgl/lv_binding_rust)—there's still a lot to do and merge.

C47D, 2024-11-21:
@AlixANNERAUD I'm not really familiar with Rust, so I don't think I'm the correct person to merge pull requests. I will take a look at the issues tho.

C47D, 2024-11-23:
@AlixANNERAUD @kisvegabor I've opened an issue to keeo track of what I had to do in order to get the demo example up and running https://github.com/lvgl/lv_binding_rust/issues/182

There's 2 pull requests in the bindings repo that were needed to make it work, I think it would be nice to merge them and re-start the development of that repo, what do you think?

C47D, 2024-11-28:
@AlixANNERAUD Dummy question, how (and when) are the rust bindings being created, are they manually created or they're manually created?

AlixANNERAUD, 2024-11-28:
> @AlixANNERAUD @kisvegabor I've opened an issue to keep track of what I had to do in order to get the demo example up and running, see https://github.com/lvgl/lv_binding_rust/issues/182
> 
> 
> 
> There's 2 pull requests in the bindings repo that were needed to make it work, I think it would be nice to merge them and re-start the development of that repo, what do you think?

Sure! Here’s a revised version:

That’s a great start! I think the redevelopment should focus on supporting LVGL v9. LMK if you need any help.

C47D, 2024-11-28:
> > There's 2 pull requests in the bindings repo that were needed to make it work, I think it would be nice to merge them and re-start the development of that repo, what do you think?
> 
> That’s a great start! I think the redevelopment should focus on supporting LVGL v9. LMK if you need any help.

Hi, yes, maybe we can add it to the Readme as well. What's the purpose of the lvgl-sys? I was trying to fix the Gauge example and didn't find it in the generated bindings.

@kisvegabor do you happen to know who has merge rights so we can integrate the fixes into master?

AlixANNERAUD, 2024-11-29:
Here’s how the bindings generation works:  
1. `lvgl-sys` generates the raw, unsafe C-to-Rust bindings (using raw pointers) with `bindgen`. Refer to the `build.rs` file in the `lvgl-sys` folder.  
2. The proper bindings (with Rust structures, objects, and methods) are generated in the `lvgl` crate using `lvgl-codegen`. See the `build.rs` file in the `lvgl` folder.

C47D, 2024-11-29:
> Here’s how the bindings generation works:
> 
>     1. `lvgl-sys` generates the raw, unsafe C-to-Rust bindings (using raw pointers) with `bindgen`. Refer to the `build.rs` file in the `lvgl-sys` folder.
> 
>     2. The proper bindings (with Rust structures, objects, and methods) are generated in the `lvgl` crate using `lvgl-codegen`. See the `build.rs` file in the `lvgl` folder.
>        For the gauge example, I think you have to take a look at what is output by `lvgl-codegen` in the `OUT` directory.

Thanks for the reply, I'll take a look at those directories. Do you mind if we add that explanation into the README? It might help others to understand how the bindings are created.

AlixANNERAUD, 2024-12-16:
@C47D Any update ?

C47D, 2024-12-19:
Hi @kisvegabor and @AlixANNERAUD , I also had a busy week but I just freed up myself, almost ready for vacations.
I've accepted the invitation @kisvegabor , I've opened https://github.com/lvgl/lv_binding_rust/issues/184 to discuss further actions.

AlixANNERAUD, 2024-12-19:
> Hi @kisvegabor and @AlixANNERAUD , I also had a busy week but I just freed up myself, almost ready for vacations. I've accepted the invitation, I've opened [lvgl/lv_binding_rust#184](https://github.com/lvgl/lv_binding_rust/issues/184) to discuss further actions.

Yeah, don't worry, me too.

C47D, 2024-12-22:
Hi @kisvegabor , I've noticed the required revision of a pull request has to be from an specific set of users, I asked @AlixANNERAUD to help me review my pull request and the merge isn't enabled with their review. How can we proceed?
Remove the required review and letting the users with enough repo rights to merge the requests?

kisvegabor, 2024-12-22:
@C47D I've jsut approved your PR so it can be merged now. I also opened [this issue](https://github.com/lvgl/lv_binding_rust/issues/185) to discuss how to move forward.

kisvegabor, 2025-01-09:
As @C47D and @AlixANNERAUD are added as maintainers of the Rust repo I close this issue. 

Thank you guys! :heart:

closed 2025-01-09 by kisvegabor