lvgl/lvgl #7303: Alignment of image not working when image is scaled

Pavi92, 2024-11-15:
### LVGL version

v9.2

### What happened?

The image is placed at the position where ist should be placed without scaling
[image]

### How to reproduce?

```
LV_IMAGE_DECLARE(Symbol_Inc);
lv_obj_t *button = lv_button_create(lv_screen_active());
lv_obj_t *image = lv_image_create(button);
lv_image_set_src(image, &Symbol_Inc);
lv_obj_center(image);
lv_image_set_style_transform_scale(image, 128,128,LV_STATE_DEFAULT);
```

kisvegabor, 2024-11-21:
Hi,

When a widget is scaled the pivot point (origin of the transformation) is the top left corner by default. It means this point stays in place, and the other parts scales around it. 

You can change the pivot like this:
```c
  lv_obj_set_style_transform_scale(image, 128,LV_STATE_DEFAULT);
  lv_obj_set_style_transform_pivot_x(image, lv_pct(50),LV_STATE_DEFAULT);
  lv_obj_set_style_transform_pivot_y(image, lv_pct(50),LV_STATE_DEFAULT);
```

However not that, if you are scaling a simple image it's faster:
```c
  lv_image_set_scale(image, 128);
//  lv_image_set_pivot(image, lv_pct(50),lv_pct(50)); /*In case of the images, center is the default*/
```
as LVGL doesn't need to allocate a layer for widget transformation.

Pavi92, 2024-11-22:
Worked.

Thank you very much !

closed 2024-11-22 by Pavi92