lvgl/lvgl #7312: Alpha Array in RGB565A8 Not Written

jheinztrij, 2024-11-15:
### LVGL version

v9.2.0

### What happened?

After setting color depth to 16 and display color format to LV_COLOR_FORMAT_RGB565A8, the alpha array is not being generated.
Alpha is generated fine with 32 bit color depth and display color format LV_COLOR_FORMAT_ARGB8888.
Verified the alpha array section of my display buffers (whole screen direct render) was not being modified by LVGL by memsetting the last third of the buffers to 0x7F and the data remains untouched when inspected during flush function calls.

### How to reproduce?

  lv_display_t * disp = lv_display_create(MY_DISP_HOR_RES, MY_DISP_VER_RES);
    lv_display_set_color_format(disp, LV_COLOR_FORMAT_RGB565A8);
    lv_display_set_flush_cb(disp, disp_flush);
    static uint8_t buf_3_1[MY_DISP_HOR_RES * MY_DISP_VER_RES * BYTE_PER_PIXEL];    //hor*ver*3bytes per pixel
    static uint8_t buf_3_2[MY_DISP_HOR_RES * MY_DISP_VER_RES * BYTE_PER_PIXEL];    //hor*ver*3bytes per pixel
    lv_display_set_buffers(disp, buf_3_1, buf_3_2, sizeof(buf_3_1), LV_DISPLAY_RENDER_MODE_DIRECT);

static void disp_flush(lv_display_t * disp_drv, const lv_area_t * area, uint8_t * px_map)
{
      setRGBData(px_map);
  //All values in alpha array are always 0 unless set manually on buffer creation, then will remain the initial value
      setAlphaData(px_map + MY_DISP_HOR_RES * MY_DISP_VER_RES * (BYTE_PER_PIXEL-1));
      /*IMPORTANT!!!
       *Inform the graphics library that you are ready with the flushing*/
      lv_display_flush_ready(disp_drv);
}

liamHowatt, 2024-11-25:
Do you have interest in opening a PR? :slightly_smiling_face: It sounds like you understand the issue and what's missing.

If I'm not mistaken, this color format was added somewhat recently so it may be missing these blending functions, which should be added as required.

The interesting part is that there are already transformation functions for this color format, just not blending.

If you add it, please guard it with the `LV_DRAW_SW_SUPPORT_RGB565A8` config.

kisvegabor, 2024-11-28:
`LV_COLOR_FORMAT_RGB565A8` is really not supported as a destination color format now. What kind of hardware does support RGB565A8 format?

GorgonMeducer, 2024-12-03:
> `LV_COLOR_FORMAT_RGB565A8` is really not supported as a destination color format now. What kind of hardware does support RGB565A8 format?

I have the same question as @kisvegabor  asked.

closed 2024-12-28 by lvgl-bot