lvgl/lvgl #7313: Is running `lv_tick_inc` in another thread thread-safe?

JohanChane, 2024-11-17:
### Introduce the problem

use `lv_tick_inc` in another thread:

```cpp
main {
  thread {
    lv_tick_inc(5);
    usleep(5000);
  }

  while(1) {
    uint32_t time_till_next = lv_timer_handler();
    my_delay_ms(time_till_next);
  }
}
```

`lv_tick_inc` source:

```c
static uint32_t sys_time = 0;
static volatile uint8_t tick_irq_flag;
LV_ATTRIBUTE_TICK_INC void lv_tick_inc(uint32_t tick_period)
{
    tick_irq_flag = 0;
    sys_time += tick_period;
}

uint32_t lv_tick_get(void)
{
    uint32_t result;
    do {
        tick_irq_flag = 1;
        result = sys_time;
    } while(!tick_irq_flag);

    return result;
}
```

Based on:
- Memory ordering depends on both the order of instructions generated by the compiler at compile time and the execution order of the CPU at runtime. See [ref](https://en.wikipedia.org/wiki/Memory_ordering).
- Additionally, `sys_time` and `tick_irq_flag` do not have [Dependency-ordered before](https://en.cppreference.com/w/cpp/atomic/memory_order).

The following situations are possible:

```c
LV_ATTRIBUTE_TICK_INC void lv_tick_inc(uint32_t tick_period)
{
    sys_time += tick_period;
    tick_irq_flag = 0;
}
```

```c
LV_ATTRIBUTE_TICK_INC void lv_tick_inc(uint32_t tick_period)
{
    sys_time += tick_period;  // Half of the statement
    tick_irq_flag = 0;
    sys_time += tick_period;  // The other half of the statement
}
```

In conclusion, I believe that using `volatile` cannot solve the atomicity issue of `sys_time`.

References:
- [Optimisation question](https://github.com/lvgl/lvgl/issues/375)
- [Threading Considerations](https://docs.lvgl.io/master/intro/add-lvgl-to-your-project/threading.html#lvgl-and-threads)
- [Tick Interface](https://docs.lvgl.io/8.4/porting/tick.html)
-   [Why is volatile not considered useful in multithreaded C or C++ programming?](https://stackoverflow.com/questions/2484980/why-is-volatile-not-considered-useful-in-multithreaded-c-or-c-programming)

vwheeler63, 2024-11-17:
This is a *really* interesting point!  It begs for a cross-platform method of memory fencing around the lines of code in the 2 functions or reading `sys_time` directly on a system on which it is known to be an atomic value.  But *I believe* that it is safe conditionally — *as long as* — `sys_time` is indeed an atomic value, and not so if it is not.

There is a forgiving factor about possible instruction re-ordering, in that every use of `lv_tick_get()` that I am aware of is used to trigger an action (e.g. a timer or update of "last_activity_time" for an `lv_display_t` that if it is not done NOW, it will be done later, and the delay between the two will not cause any damage.

However, it is *not forgiving* if `sys_time` is *not* an atomic value and a read of it becomes corrupted.  Some kind of guarantee of instruction ordering would be needed inside the `do-while` loop to make it safe.

@kisvegabor  Thoughts?

kisvegabor, 2024-11-17:
This simple protection mechanism is really not designed for complex architectures with e.g. instruction reordering.

I think it can be solved in 3 ways:
- We introduce some define for at least `atomic_uint32_t`
- We assume that basic `uint32_t` operations are always atomic. See [here](https://stackoverflow.com/a/36955947/1999969)
- Use `lv_tick_set_cb` to set a better tick source

vwheeler63, 2024-11-17:
> I think it can be solved in 3 ways

And I could add to the documentation about an example of how to handle if you have a 16-bit platform.  Example:  the PIC 16-bit processors use the Harvard instruction set and it has a genius instruction to disable interrupts for a fixed number of SYSCLK cycles, and that number was an argument of the instruction (a field in the machine code).  So for a very convenient way to "protect" a 32-bit value like the system millisecond count (which got updated in an ISR context) to protect it like this (`__builtin_disi(2)` is translated by the compiler into that assembly instruction -- much more efficient than a MUTEX):
```c
/**------------------------------------------------------------------------
 * \brief  Safe read from 'gui32ElapsedPwrOnTimeInMs'
 *-------------------------------------------------------------------------*/
#define mSafeCopyOfElapsedTime(dest) {                                       \
        register volatile uint32_t  lui32Temp;                               \
        /* Disable priority 1-6 interrupts for 2 Fcys. */                    \
        __builtin_disi(2);                                                   \
        lui32Temp = gui32ElapsedPwrOnTimeInMs;   /* Cost: 2 Fcys */          \
            /* Generally looks like this in assembly:      */                \
            /*     MOV gui32ElapsedPwrOnTimeInMs, W0       */                \
            /*     MOV 0x7898, W1                          */                \
            /* Now value is copied to registers W0:W1      */                \
            /* where it can be written to any destination. */                \
        dest = lui32Temp;                        /* Cost: doesn't matter */  \
}
```

which works very nicely in one of our dashboards.

uLipe, 2024-11-19:
Just giving 2-cents what about creation of weak defined functions atomic_set and atomic_get, provide them default implementations but let's available to users to provide the actual way of doing that on their platforms?

kisvegabor, 2024-11-20:
@vwheeler63 I'm ok with adding such an example to the docs.

@uLipe "weak" functions are not C standard and not supported by all compilers. :slightly_frowning_face:

uLipe, 2024-11-20:
@kisvegabor , okay I was considering just GCC, Clang and IAR variants, but instead, the default / platform defined functions could be selected at build time, using some configuration option? If that option is not set, just use the default implementation that disables the irq, update the atomic variable, and then re-enable it (disable and re-enable here are meant to save latest irq state and restore it).

vwheeler63, 2024-11-20:
> I'm ok with adding such an example to the docs.

Done in PR #7342.

uLipe, 2024-12-06:
So, since now we have a documentation that explicitly explains the details related to the lv_tick_inc, can we close this?

kisvegabor, 2024-12-18:
As the #7342 is merged I close this issue.

closed 2024-12-18 by kisvegabor