lvgl/lvgl #7326: LVGL slider/memory issue

kschwan, 2024-11-19:
### LVGL version

v9.2.0-265-ge85059d82

### Platform

Custom platform with NXP RT1170

### What happened?

Hi,
I'm experiencing a crash/assert with the slider widget. When dragging the slider close to the lowest level, it looks like in the following image, with corresponding log output (from two separate restarts) below.

[image]

```
[Info]  (6.962, +55)     indev_proc_press: pressed at x:111 y:388 lv_indev.c:1165
[info]  Slider value: 15
[Info]  (7.018, +56)     indev_proc_press: pressed at x:84 y:385 lv_indev.c:1165
[info]  Slider value: 12
[Info]  (7.073, +55)     indev_proc_press: pressed at x:59 y:381 lv_indev.c:1165
[info]  Slider value: 8
[Info]  (7.084, +11)     lv_draw_layer_alloc_buf: Layer memory used: 13 kB
 lv_draw.c:467
[Info]  (7.129, +45)     lv_draw_dispatch_layer: Layer memory used: 0 kB
 lv_draw.c:223
[Info]  (7.145, +16)     indev_proc_press: pressed at x:31 y:369 lv_indev.c:1165
[info]  Slider value: 4
[Info]  (7.156, +11)     lv_draw_layer_alloc_buf: Layer memory used: 6 kB
 lv_draw.c:467
[Error] (7.200, +44)     integrity_walker: Asserted at expression: integ->prev_status == this_prev_status && "prev status incorrect" lv_tlsf.c:831
```

```
[Info]  (27.200, +10)    lv_draw_dispatch_layer: Layer memory used: 0 kB
 lv_draw.c:223
[Info]  (27.320, +120)   indev_proc_press: pressed at x:64 y:431 lv_indev.c:1165
[Info]  (27.328, +8)     lv_draw_layer_alloc_buf: Layer memory used: 11 kB
 lv_draw.c:467
[Info]  (27.374, +46)    lv_draw_dispatch_layer: Layer memory used: 0 kB
 lv_draw.c:223
[Info]  (27.391, +17)    indev_proc_press: pressed at x:64 y:431 lv_indev.c:1165
[Info]  (27.399, +8)     lv_draw_layer_alloc_buf: Layer memory used: 11 kB
 lv_draw.c:467
[Info]  (27.444, +45)    lv_draw_dispatch_layer: Layer memory used: 0 kB
 lv_draw.c:223
[Info]  (27.461, +17)    indev_proc_press: pressed at x:64 y:431 lv_indev.c:1165
[Info]  (27.469, +8)     lv_draw_layer_alloc_buf: Layer memory used: 11 kB
 lv_draw.c:467
[Info]  (27.514, +45)    lv_draw_dispatch_layer: Layer memory used: 0 kB
 lv_draw.c:223
[Error] (27.531, +17)    block_next: Asserted at expression: !block_is_last(block) lv_tlsf.c:457
```

### How to reproduce?

Sorry, the example is in C++. I hope you'll get the idea.

```c++
  auto screen = lv_obj_create(nullptr);
  lv_obj_set_size(screen, 800, 480); // this is the size of the display

  auto slider = lv_slider_create(screen);
  lv_obj_set_size(slider, 800 - 80, 60);
  lv_obj_set_pos(slider, 40, 80);
  lv_slider_set_range(slider, 0, 100);
  lv_slider_set_value(slider, 50, LV_ANIM_OFF);

  lv_screen_load(screen);

  static freertospp::MutexStatic mutex; // Just a C++ wrapper around a FreeRTOS mutex
  static lv_point_t point{0, 0};
  static lv_indev_state_t state = LV_INDEV_STATE_RELEASED;

  lv_indev_t* indev = lv_indev_create();
  lv_indev_set_type(indev, LV_INDEV_TYPE_POINTER);
  lv_indev_set_read_cb(
      indev,
      [](lv_indev_t* indev, lv_indev_data_t* data) {
        freertospp::LockGuard lock{mutex}; // RAII style lock
        data->point = point;
        data->state = state;
      }
  );

  // touchDriver was instantiated earlier
  touchDriver.subscribeToTouchEvents([&](const auto& event) {
    if (event.id != 0)
      return;

    freertospp::LockGuard lock{mutex};
    point.x = event.point.x;
    point.y = event.point.y;
    state   = event.state;
  });
```

kisvegabor, 2024-11-27:
Hi,

I can't see any obvious issues in the code. You can insert `LV_USE_ASSERT_MEM_INTEGRITY` many places in your code to bisect where the issue could be. 

Probably it's an out of bounds write somewhere.

kschwan, 2024-12-04:
I've discovered that this issue happens with `LV_USE_THEME_DEFAULT==1`, but not with `LV_USE_THEME_DEFAULT==1` (and `LV_USE_THEME_SIMPLE==1`)

kschwan, 2024-12-20:
Not stale. Still haven't gotten around to git bisect'ing :/

kisvegabor, 2025-01-09:
Does it help if you increase `LV_MEM_SIZE`?
Do you see anything special in the logs?

kschwan, 2025-01-14:
> Does it help if you increase `LV_MEM_SIZE`? Do you see anything special in the logs?

No and yes. I attached some log output above - the errors shown were the only unusual thing.

I was planning to try this without PXP enabled as I have a suspicion it has to do with this issue, but I haven't gotten around to it yet as I'm currently working on a different part of our system :/

closed 2025-02-07 by lvgl-bot

kschwan, 2025-02-11:
Not abandoned :o