lvgl/lvgl #7333: lv_font_conv：Use the tool to generate Chinese font bin file，lv_font_load Failed to load bin file

DragonYY123, 2024-11-20:
lv_font_conv --font MiSans-Medium.ttf -r 0x20-0x7F --size 40 --format bin --bpp 2 --no-compress -o cour40_2.bin
This bin file ok.
lv_font_conv --font MiSans-Medium.ttf -r 0x20-0x7F -r 0x4E00-0x9FA5 --size 40 --format bin --bpp 2 --no-compress -o cour40_2unicode.bin
Use lv_font_load() functions.Failed to load ambiq_cour40_2unicode.bin file.
Please advise how to deal with it?

kisvegabor, 2024-11-20:
I've just moved the issue to the main lvgl repo, as it seems like an LVGL issue.

closed 2024-12-15 by lvgl-bot

kisvegabor, 2024-12-18:
I confirm that the font can be loaded, but you need a lot of RAM to store the loaded font. Using [TinyTTF](https://docs.lvgl.io/master/details/libs/tiny_ttf.html) might be a better option