lvgl/lvgl #7334: Need for Drawing Events in `lv_scale`

vwheeler63, 2024-11-20:
### Problem to solve

When `lv_meter` went away in LVGL v9.0.0 in favor of `lv_scale`, some important drawing events went away, and in previous firmware projects, I made heavy use of those events in a tachometer (rev counter in British), for example to:

- change the text of the major-tick labels about to be rendered -- div by 1000 so "5" = "5000 RPM", and
- to change the color of the needle just before it was rendered, to match a user-configured color representing the power range of the engine.

The drawing events sent by `lv_meter` were both powerful and valuable.  They were well documented in the v8.4 documentation, and they came with vital drawing descriptor objects attached to the event (`e` argument) which gave write access to the `text` pointer, `color` and other things relevant to the drawing about to happen:

[image]

It would be a powerful addition to `lv_scale` to implement these drawing events for the same reasons that they were powerful and valuable for `lv_meter`.

### Success criteria

That the documentation of events for `lv_meter` in LVGL v8.4 will apply in full to the current version.  Since it is a feature ADDITION, no API will be broken.

### Solution outline

- All the events with drawing descriptors be harvested out of the `lv_meter` and added to `lv_scale`.
- Documentation of events from v8.4 be copy/pasted into current documentation for `lv_scale`.

### Rabbit holes

None known.

### Testing

- In one of the `lv_scale` (or perhaps in a new example, so examples don't get too complex), demonstrate changing actual text and color of major-tick labels, and
- perhaps this same (new) example can demonstrate changing something about the needle (e.g. color?)

using these drawing events.

### Teaching

- Documentation of events from v8.4 be copy/pasted into current documentation for `lv_scale`.

### Considerations

The code that populated the descriptors for the documented in events in `lv_meter` were quite straightforward and can be copied nearly verbatim if the drawing descriptors have not changed between LVGL v8.4 and v9.3.

kisvegabor, 2024-11-21:
Aren't it way simpler in with scale?

> change the text of the major-tick labels about to be rendered -- div by 1000 so "5" = "5000 RPM"

```c
    static const char * texts[] = {"a", "b", "c", "d", "e", "f", "g", NULL};
    lv_scale_set_text_src(scale, texts);
```

> to change the color of the needle just before it was rendered, to match a user-configured color representing the power range of the engine.

```c
    lv_obj_t * needle_line = lv_line_create(scale_line);
    
    //Later
    lv_scale_set_line_needle_value(scale, line, 100, value);
    lv_obj_set_style_line_color(needle_line, value_to_color(value), LV_PART_MAIN);

```

vwheeler63, 2024-11-22:
> Aren't it way simpler in with scale?

Indeed, those DO handle what I was using them for.... In which case, the "need" for them drops to near zero....

vwheeler63, 2024-11-22:
> Great! Note that you can still use the `DRAW_TASK_ADDED` event to adjust e.g. all the tick lines one by one.

Excellent!  In that case, I will close this since the need appears to have been filled in other ways!  👍

closed 2024-11-22 by vwheeler63