lvgl/lvgl #7337: Update NEMA library seems to break bar drawing

williams-one, 2024-11-20:
### LVGL version

v9.3

### Platform

STM32U5 demo board with NEMA GFX and VG enabled

### What happened?

Hello, I've been testing the last version of master and it seems to me that the update of the NEMA library has broken the drawing of a bar (see the darker blue triangle at the right end of the bar)

[image]

Up to the previous commit [fix(NemaGFX): arc angles](https://github.com/lvgl/lvgl/pull/7307/commits/539d7294656e72147e7abe4dc204db5934e78db6) the drawing was fine
After the commit [chore(NemaGFX): update library](https://github.com/lvgl/lvgl/pull/7307/commits/280177a5a7b10845017076bf28718a7786e3f421) the drawing is broken

kisvegabor, 2024-11-21:
cc @liamHowatt @MarkIoanAmat

williams-one, 2024-12-09:
Just adding a comment to prevent it going stale.  If you think it's not relevant/reproduceable please close it.

kisvegabor, 2024-12-18:
It's relevant. @liamHowatt please take a look in the near future.

kisvegabor, 2025-01-03:
Not stale.

liamHowatt, 2025-01-07:
I'm inclined to say it's a bug with Nema's `nema_fill_rounded_rect_aa`. cc @MarkIoanAmat

[image]

Only two draw tasks are executed to draw this. One fill for the screen background and one fill for the bar background, which has a non-zero radius.

https://github.com/lvgl/lvgl/blob/1f9f5f475ffef8bcbc6f8c84e7a75fa32f697c59/src/draw/nema_gfx/lv_draw_nema_gfx_fill.c#L90

The passed parameters are reasonable. `coords_bg_h` is `384` and `radius` is `192`.

I also tried this:

```c
lv_obj_t * o1 = lv_obj_create(lv_screen_active());
lv_obj_set_size(o1, LV_PCT(70), LV_PCT(45));
lv_obj_set_style_border_width(o1, 0, 0);
lv_obj_set_style_bg_opa(o1, 51, 0);
lv_obj_set_style_bg_color(o1, lv_color_black(), 0);
lv_obj_set_style_radius(o1, LV_COORD_MAX, 0);

lv_obj_t * o2 = lv_obj_create(lv_screen_active());
/* ... */
/* (other parts same as for o1) */
/* ... */
lv_obj_set_style_radius(o2, 40, 0);
lv_obj_set_align(o2, LV_ALIGN_BOTTOM_LEFT);
```

It's not an issue with using a maximum radius, as you can see for the `o2` obj with a non-maximum radius.

[image]

KlemenDEV, 2025-02-20:
I think similar issue can happen without NEMA too

MarkIoanAmat, 2025-02-20:
@KlemenDEV Do you have a relevant example ? Screenshot and example code will help.

KlemenDEV, 2025-02-20:
I will try to prepare isolated example

liamHowatt, 2025-04-08:
We are connecting with the related vendors about it. Here is an isolated example to reproduce without LVGL.

```c
nema_init();
nema_vg_init(800, 480);
nema_cmdlist_t cl = nema_cl_create();
nema_cl_bind_circular(&cl);
memset((void *) 0x20000000, 0xff, 800 * 480 * 2);
nema_set_clip(0, 0, 800, 480);
nema_bind_dst_tex((uintptr_t)NEMA_VIRT2PHYS((void *)0x20000000), 800,
                  480, NEMA_RGB565, 1600);
uint32_t bg_color = nema_rgba(255, 0, 0, 127);
nema_set_blend_fill(NEMA_BL_SRC_OVER);
bg_color = nema_premultiply_rgba(bg_color);
nema_fill_rounded_rect_aa(0, 0, 799, 479, 200, bg_color);
nema_cl_submit(&cl);
nema_cl_wait(&cl);
```

PR #8138 (2025-04-26): fix(NemaGFX): update STM32U5 nema lib to revC
Fixes #7337

cc @MarkIoanAmat

Replace `cortex_m33` with `cortex_m33_revC`. It has been discussed that there is no reason to keep the previous version.

Fixes transparent rounded rectangle issues and improves performance.

The https://github.com/lvgl/lv_port_riverdi_stm32u5 library path will be updated.

Below is a the performance **difference** between the old and new versions of the library.

#### Riverdi STM32U5 Nema (revC) vs Riverdi STM32U5 Nema (non-revC)

Name | Avg. CPU | Avg. FPS | Avg. time | render time | flush time
-- | -- | -- | -- | - [... cut]
closed 2025-04-26 by kisvegabor