lvgl/lvgl #7345: Drawing text using font where bpp is 3

RomanLuchyshyn, 2024-11-21:
### LVGL version

9.2.2

### Platform

any

### What happened?

After migration from 8.3.0 to 9.2.2 I faced with problem there are no any texts on most gui controls. My investigation show that difference is only in generated data by lv_font_conv below
`#if LVGL_VERSION_MAJOR >= 8
static const lv_font_fmt_txt_dsc_t font_dsc = {
#else
static lv_font_fmt_txt_dsc_t font_dsc = {
#endif
    .glyph_bitmap = glyph_bitmap,
    .glyph_dsc = glyph_dsc,
    .cmaps = cmaps,
    .kern_dsc = &kern_classes,
    .kern_scale = 16,
    .cmap_num = 3,
    **.bpp = 3,**
    .kern_classes = 1,
    .bitmap_format = 1,
#if LVGL_VERSION_MAJOR == 8
    .cache = &cache
#endif
};`
This rle compressed data. And other key is bpp = 3. I would like to emphasize that this code is fine for 8.3.0.
During debug i found in function draw_letter_cb (file lv_draw_sw_letter.c:67) there is no implementation for drawing letter of bpp = 3. In fact glyph_draw_dsc->format is value of font bpp. (See in code above). draw_letter_cb just skips this case of bpp 3. In version 8.3.0 letter drawing is implemented in function draw_letter_normal (lv_draw_sw_letter.c:165) there is a line which solve problem of bpp = 3 in line 175:
`if(bpp == 3) bpp = 4;`
I tried to add similar code in 9.2.2 and it solved problem.

kisvegabor, 2024-11-28:
Hi, 

Can you send a font to reproduce the issue? Could you also show exactly what code needed to be added to make it work again?

kisvegabor, 2024-11-28:
I've just found the related PR #7350

closed 2024-12-24 by lvgl-bot