lvgl/lvgl #7347: Custom fonts with 8 Bpp is not visible

keerthansys22, 2024-11-21:
### LVGL version

v9.2.2

### Platform

Custom board with Linux OS

### What happened?

I'm trying to convert my custom font using [LVGL font converter](https://lvgl.io/tools/fontconverter) and I tried with 4 bpp (bits-per-pixel) and the font is not clear and sharper.

So, I generated the fonts with higher bpp value i.e **8 bpp** but the fonts are not visible. Is there any option to be enabled in the configuration file?

My custom fonts are not super clear and sharper, I'm thinking of using compressed fonts, but the document suggests the compression is more effective with larger fonts and higher bpp. In my case, the **maximum font size is 10 pt**. Would you suggest me to go for compressed fonts? 

**With 4 bpp : -**
[image]

**With 8 bpp: -**
[image]

liamHowatt, 2024-11-21:
Is the version you're on older or newer than 417415f (#7100)? It was added then (October 26).

keerthansys22, 2024-11-21:
I'm on version V9.2.2, are you suggesting me to check with v9.3.0-dev ?

liamHowatt, 2024-11-21:
> are you suggesting me to check with v9.3.0-dev ?

Yes

> Any suggestions to use compressed fonts for my use-case?

What for? See the docs: https://docs.lvgl.io/master/details/main-components/font.html#compressed-fonts

keerthansys22, 2024-11-26:
Yes, its working in V9.3.0-dev, thanks.

closed 2024-11-26 by keerthansys22