lvgl/lvgl #7348: Migrate the two VG-Lite drivers

kisvegabor, 2024-11-21:
### Introduce the problem

Now we have 2 VG-Lite drivers which is no ideal. I asked @uLipe to test the performance and get familiar with the internals/pro/cons of each implementations so that we can start working on a single unified solution. 

To check what's going on I've created and SDK example for NXP i.MX1170 (with FreeRTOS as there is no VGLite in the Bare metal version) and updated it to LVGL v9. It compiles and works mostly  with software rendering:
- there are some cache related artifacts
- it's very slow. -O3 and [linking the app to RAM](https://github.com/user-attachments/assets/886cd361-535b-4c07-b11b-05a5f4517b28) could make it fast

When I enable **Xiaomi's VGLite** I get this error (`LV_USE_DRAW_VG_LITE`):
```
../lvgl/lvgl/src/draw/vg_lite/lv_vg_lite_utils.h:96:40: error: unknown type name 'vg_lite_stroke_t'
   96 | void lv_vg_lite_stroke_dump_info(const vg_lite_stroke_t * stroke);
      |                                        ^~~~~~~~~~~~~~~~
make[1]: *** [lvgl/lvgl/src/draw/vg_lite/subdir.mk:75: lvgl/lvgl/src/draw/vg_lite/lv_vg_lite_stroke.o] Error 1
../lvgl/lvgl/src/draw/vg_lite/lv_vg_lite_path.c: In function 'lv_vg_lite_path_reset':
../lvgl/lvgl/src/draw/vg_lite/lv_vg_lite_path.c:143:28: error: 'VG_LITE_DRAW_ZERO' undeclared (first use in this function)
  143 |     path->base.path_type = VG_LITE_DRAW_ZERO;
```

When I enable **NXP's VGLite** I get this error (`LV_USE_DRAW_VGLITE`):
```
../lvgl/lvgl/src/draw/nxp/vglite/lv_vglite_utils.h:28:10: fatal error: vg_lite_options.h: No such file or directory
   28 | #include "vg_lite_options.h"
```

I pushed my current project here: 
[image]

cc @FASTSHIFT @onecoolx @nicusorcitu @anaGrad @cosmindanielradu19 

Could you take a look at this issue to point us into right direction?

Context: We need to create a demo for CES and I'd like to use some cool perspective matrix transformation effects.

FASTSHIFT, 2024-11-21:
This is because the VG-Lite driver version is not new enough. It is recommended to update the VG-Lite driver to this [version](https://github.com/lvgl/lvgl/blob/master/src/others/vg_lite_tvg/vg_lite.h#L64) (or a later version) to maintain the consistency of the code and the most full functionality and performance.

```bash
../lvgl/lvgl/src/draw/vg_lite/lv_vg_lite_utils.h:96:40: error: unknown type name 'vg_lite_stroke_t'
   96 | void lv_vg_lite_stroke_dump_info(const vg_lite_stroke_t * stroke);
      |                                        ^~~~~~~~~~~~~~~~
make[1]: *** [lvgl/lvgl/src/draw/vg_lite/subdir.mk:75: lvgl/lvgl/src/draw/vg_lite/lv_vg_lite_stroke.o] Error 1
../lvgl/lvgl/src/draw/vg_lite/lv_vg_lite_path.c: In function 'lv_vg_lite_path_reset':
../lvgl/lvgl/src/draw/vg_lite/lv_vg_lite_path.c:143:28: error: 'VG_LITE_DRAW_ZERO' undeclared (first use in this function)
  143 |     path->base.path_type = VG_LITE_DRAW_ZERO;
```

nicusorcitu, 2024-11-21:
@cristian-stoica

nicusorcitu, 2024-11-21:
hi @kisvegabor I see you are testing lvgl 9 on rt1170. The board is not yet supported. I mean it is supported on our end but code have not been pushed on github.
When do you need this to be done? Hopefully in the meanwhile we push the driver to NXP/github for 24.12 release. I will let you know when is done.

nicusorcitu, 2024-11-21:
If you need it just now, @cosmindanielradu19 can provide you an archive with everything is done for 1170.

kisvegabor, 2024-11-21:
Thank you for the quick reply. It'd be great if you could send an archive.

kisvegabor, 2024-11-21:
I've just noticed that link to application RAM is not working now. It crashes immediately. Maybe is it also fixed?

kisvegabor, 2024-11-21:
> When do you need this to be done? 

In 2 weeks. 🙁 Using an other could be an option too, but we need to address the performance issue (ideally by having both the GPU and linking to RAM working)

cristian-stoica, 2024-11-21:
> I've just noticed that link application to RAM is not working now. It crashes immediately. Maybe is it also fixed?

I've had mixed results running from RAM on 1170. Typically, I run from FLASH for testing and OCRAM for performance. If the application is small enough to fit OCRAM I make the OCRAM1 section as large as possible at the expense of OCRAM2.
Besides the "link application from RAM" checkbox, you need to put the RAM section (OCRAM1) the first in the list in MCU_SETTINGS. If it doesn't work like this there are likely some initialization issues the SDK (doesn't) take care of. An archive from Cosmin will probably fix that :)

cosmindanielradu19, 2024-11-21:
> Thank you for the quick reply. It'd be great if you could send an archive. 

I'll come back with the archive of the project tomorrow.

kisvegabor, 2024-11-21:
I'm planning to use the ebike demo as a fallback plan, but the background image is quite large there (1.8MB), which won't fit to OCRAM. Can we link the app to OCRAM but the background image  to external RAM? I assume that external RAM it's faster than the external flash.

nicusorcitu, 2024-11-22:
Hi @kisvegabor 
What I would suggest is:

1. Put the code binary in internal RAM (OCRAM) - mandatory.
2. Let the FBs in external RAM (SDRAM) - mandatory for RT1170. On RT700 the FB can fit as well in internal RAM (SRAM).
3. Try to put all the assets (source images) into OCRAM as well - recommended. If there is no place in OCRAM, put them in SDRAM (less performance than previous).
4. Don't use FLASH at all :).

kisvegabor, 2024-11-25:
That's my plan too.  :slightly_smiling_face: 

@cosmindanielradu19  Do you see when can you send the project?

cosmindanielradu19, 2024-11-26:
> That's my plan too. 🙂
> 
> @cosmindanielradu19 Do you see when can you send the project?

Sorry for the delay. I'll talk with @nicusorcitu to check how to provide the archive, and I'll come back to you.

uLipe, 2024-11-26:
Just a comment, using the current project, the file that seems to be missing is: ` vglite_options.h`

cosmindanielradu19, 2024-11-26:
@kisvegabor I've just shared the project. You should've received an email related to this.

kisvegabor, 2024-11-26:
Thank you, received! @uLipe is trying it out now.

uLipe, 2024-12-02:
Hello Everyone, last week I went deeper on the VGLite stuff, and using the current private project i was able to create a setup to probe the VGLite implementations.

One immediate issue I found, is the project was made to run on the IMXRT1170 EVK and not on the EVKB, so a patch into the flex_spi_nor_config.c must be applied since the EVKB (the current version of this board) has a different QSP chip.

By editing link options I was able to link the firmware to both SDRAM and OCRAM, the last one is the most suitable since the benchmark demo project fits to it, allowing to achieve best memory related performance.

The other immediate issue is in respect of usage of the both VGLite client APIs, I had to provide some modifications at board support level to get both client APIs working, the NXP VGLite port allows the display driver to get initialized after the client API which is not true for the other VGLite client API which gathers the GPU controller information during its initialization, I patched the project to include this mechanism, and now we can just use one of another VGLite client API by just editing the lv_conf.h file.

Solved the setup issues, I just ran the benchmark project in following:

* Using either VGLite implementation + OCRAM + -O3;
* Using software rendering + OCRAM+ -O3.

The non-NXP VGLite client API renders just some of the scenes correctly, and while checking the console the faulty scenes mentioned on the console that some options were not set into the current GPU, which makes me to suppose there are still configuration issues to solve in this scenario, In any case for the scenes  rendered correctly the report on the console was the following:

[image]

Using the NXP VGLite client API I get correct rendering, without any modifications, except for the container scenes which are being slowly rendering, the general performance looks good, with low CPU usage  in the majority of the cases, for the slow scenes there might be issues with the drawing dispatching on the VGLite Draw layer which needs some more investigation, here is the console report:

[image]

And the third one is the SW rendering, which as expected draws everything correctly, but it consumes more CPU bandwidth than the other scenarios;

 
[image]

The next immediate issues to investigate is what is making the container scene to get rendered so slow before proceed with the VGLite drivers, and also improve their integration with the underlying kernel driver.

kisvegabor, 2024-12-02:
Thank you @uLipe ! This slow container rendering is really strange. I'm very curious what's causing it.

nicusorcitu, 2024-12-03:
Please state clearly which results are with NXP VGLite and which one with the other implementation.
From the logs, it looks like the first image with results is NXP. NXP does not have logs like `lv_vg_lite_dump_info...`

nicusorcitu, 2024-12-03:
Our latest result on RT1170 from OCRAM with optimization -O3:
[image]

kisvegabor, 2024-12-03:
@nicusorcitu could you share a bin/hex with us, so that we can try reproducing it. We can also use it on CES, if we can confirm it's working on out end too.

cosmindanielradu19, 2024-12-03:
> @nicusorcitu could you share a bin/hex with us, so that we can try reproducing it. We can also use it on CES, if we can confirm it's working on out end too.

It's the project that I've sent you. You have to change the optimization from -Og to -O3 / -Ofast, and use LVGL v9.2.1. Our results were obtained using RT1170 EVK with that specific project.

kisvegabor, 2024-12-03:
Okay, thank you! @uLipe please try it out, and if it's fast on your end too, send me the binary.

cosmindanielradu19, 2024-12-03:
> It's the project that I've sent you. You have to change the optimization from -Og to -O3 / -Ofast, and use LVGL v9.2.1. Our results were obtained using RT1170 EVK with that specific project.

Just to be sure that there is no confusion, the results were obtained using RT1170 EVK <ins>A</ins>

kisvegabor, 2024-12-04:
Are there any important differences between A and B?

cosmindanielradu19, 2024-12-04:
No, flash is the only difference

uLipe, 2024-12-04:
@cosmindanielradu19 thanks for pointing that, the images were swapped I corrected the post

uLipe, 2024-12-04:
I just ran a new round of the benchmark on EVKB, OCRAM+O3+VGLITE, the results seems very similar:

```
Name, Avg. CPU, Avg. FPS, Avg. time, render time, flush time

Empty screen, 8%, 65, 9, 4, 5

Moving wallpaper, 6%, 33, 25, 15, 10

Single rectangle, 10%, 66, 10, 5, 5

Multiple rectangles, 12%, 66, 10, 7, 3

Multiple RGB images, 22%, 66, 8, 8, 0

Multiple ARGB images, 22%, 66, 8, 8, 0

Rotated ARGB images, 14%, 66, 10, 7, 3

Multiple labels, 33%, 33,Multiple arcs, 24%, 63, 8, 7, 1

Containers, 84%, 5, 160, 155, 5

Containers with overlay, 77%, 5, 174, 164, 10

Containers with scrolling, 79%, 7, 122, 113, 9

Widgets demo, 16%, 51, 54, 48, 6

All scenes avg.,33%, 43, 46, 41, 5
```

Also a screenshot:

[image]

cosmindanielradu19, 2024-12-04:
@uLipe, are these results obtained with LVGL V9.2.x or with master? The containers tests are using the CPU too much. There is a usage difference of ~40%, and the render time is too high 50-60ms vs 150+ ms

uLipe, 2024-12-04:
@cosmindanielradu19 , I'm using this revision:

https://github.com/lvgl/lvgl/tree/bb2a24e8a10875e356757b31a333498da610ddda

Which is the current one attached to the project

uLipe, 2024-12-04:
@cosmindanielradu19 in your lv_conf.h did you decreased the SW renders count to 0?

cosmindanielradu19, 2024-12-04:
> @cosmindanielradu19 , I'm using this revision:
> 
> https://github.com/lvgl/lvgl/tree/bb2a24e8a10875e356757b31a333498da610ddda
> 
> Which is the current one attached to the project

I see now... the version of the LVGL on this tree contains multiple commits that are not present in the version that contains the results that @nicusorcitu attached previously. Maybe that would be the difference.

I will try to run it on my side tomorrow and see what are the differences.

> @cosmindanielradu19 in your lv_conf.h did you decreased the SW renders count to 0?

Yes, our results are obtained only with VGLITE. The SW renders count is set to 0

uLipe, 2024-12-05:
@cosmindanielradu19 , is it possible to you to share the tree of the LVGL you are being using internally? Maybe i can integrate to mine and do a second round of tests.

Thank you!

cosmindanielradu19, 2024-12-06:
> ![NXPVGLITE_DRAW_OCRAM_O3_RESULTS](https://private-user-images.githubusercontent.com/5991413/392988513-1a47a559-7cc0-4049-afee-3134c98a3d12.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM0NzEyOTAsIm5iZiI6MTczMzQ3MDk5MCwicGF0aCI6Ii81OTkxNDEzLzM5Mjk4ODUxMy0xYTQ3YTU1OS03Y2MwLTQwNDktYWZlZS0zMTM0Yzk4YTNkMTIuanBlZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjA2VDA3NDMxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNjYmZlYjM4YjQ4YjM3MjlkZmFhOGI0YjM4M2M0ZTFhZDc0YzNkYWVjMWQyNTk2Mzg0OTQ4OTA2Yjg3YmFkOTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.-3koskK41JVQOwMxVDrylJJ1mbqtbubeNTLA5cI9Tec)

Those results looks very similar with the ones done on our side. I believe that this setup is a good starting point for the demo that you have to do

uLipe, 2024-12-06:
@cosmindanielradu19 , that is great, thank you for the validation!

ABADY1000, 2025-05-27:
Hello, nice to see VGLite is already working with lvgl 9.x

I am trying to run lvgl along with vglite but the compiler complains about missing file `vg_lite_options.h` and I can't find it any where.

Where could I get this file from?

onecoolx, 2025-07-10:
> [@cosmindanielradu19](https://github.com/cosmindanielradu19) , that is great, thank you for the validation!

Hi uLipe,  I would like to know if VGLite NXP can support vector API well? Because at least the SVG function depends on it now, I would like to know the plan in this regard.

uLipe, 2025-07-10:
@onecoolx yes all the vector code from the vg_lite implemenration remains untouched it was only moved to a self contained folder inside of the vglite unified API directory.

Finomnis, 2025-08-29:
How do we deal with this issue now that #8642 is merged?

uLipe, 2025-08-29:
@Finomnis for now you can use any VGLite implementation, in the next weeks I will be going back to the unification and and use the vg_lite implementation of thé drawing primitives on top of the NXP vglite dispatcher.

Finomnis, 2025-10-30:
@AndreCostaaa You got some benchmark data for imxrt11xx, just out of curiousity?