lvgl/lvgl #7349: Transparent background display issue

wu781521512, 2024-11-21:
### LVGL version

9.0

### Platform

linux

### What happened?

The effect I want to achieve is similar to Figure 1 to Figure 2
For example, in Figure 2, the alarm page is displayed during a call, with the upper part transparent
But the current effect is that Figure 1 has not been cleared, and Figure 1 and Figure 2 overlap and are displayed together
[image]
[image]

My code implementation is similar to the following
Please help me see how to solve this problem. Thank you very much

### How to reproduce?

```C
auto pageRoot = lv_obj_create (NULL);
lv_obj_set_style_bg_opa(pageRoot, LV_OPA_0, 0 );
lv_obj_t* rootContainer = lv_obj_create(pageRoot);
lv_obj_set_size(rootContainer, LV_HOR_RES, LV_VER_RES);
lv_obj_set_style_bg_color(rootContainer, lv_color_black(), 0);
lv_obj_set_style_border_width(rootContainer, 0, 0);
lv_obj_set_style_radius(rootContainer, 0, 0);

// Trigger a certain condition
lv_obj_clean(pageRoot);
lv_obj_t* rootContainer = lv_obj_create(pageRoot);
lv_obj_set_size(rootContainer, LV_HOR_RES, lv_pct(50));
lv_obj_set_style_bg_color(rootContainer, lv_color_white(), 0);
lv_obj_align(rootContainer, LV_ALIGN_BOTTOM_LEFT, 0, 0);
```

wu781521512, 2024-11-22:
lv_conf.h  Just set LV_COLOR_SCREEN_TRANSP

closed 2024-11-22 by wu781521512

kisvegabor, 2024-11-27:
In v9.0 there is no `LV_COLOR_SCREEN_TRANSP`.