lvgl/lvgl #7353: How to change the width of the Y label in the chart

vishvanathdoijad, 2024-11-22:
### Introduce the problem

My Y label ranges from 0 to 20000 in the chart. As you can see in the attached snap, the last digit is cropped. How can I
[image]
 overcome that issue?

lizhaoming634, 2024-11-22:
lv_chart_set_axis_tick(chart, axis, major_len, minor_len, major_cnt, minor_cnt, label_en, draw_size), maybe you can try to change the draw_size?

vishvanathdoijad, 2024-11-22:
lv_chart_set_axis_tick(ui_Chart3, LV_CHART_AXIS_SECONDARY_Y, 3, 5, 11, 2, true, 35);
   THis is generated by "squareline stdio" in ui_screen.c
   
   1. How can I change the draw_size?
   2. How to use #define LV_CHART_LABEL_MAX_TEXT_LENGTH 16, Means in which function I need to call this define?

lizhaoming634, 2024-11-22:
[image]

vishvanathdoijad, 2024-11-22:
Thanx,
Now I have a clear idea, and it works. 
I confused because, In "squareline" this field marked as "Font size on X/Y axis".

closed 2024-11-22 by vishvanathdoijad