lvgl/lvgl #7358: When a specified display is refreshed, is it possible to update the animation of only the display being refreshed?

M1NGS, 2024-11-23:
### Introduce the problem

When I was working on four screens, I found that as long as I refreshed one of the screens (lv_refr_now), the animations of the other three screens would also be updated. This means that when I refresh the four screens separately at different times, I will pay 4 times the CPU cost. Is this as expected?

### Proposal

If this isn't expected behavior, I may do some hacking on the anim_timer() to avoid updating all onscreen animations at once.

liamHowatt, 2024-12-02:
Animations are bound to object (usually), and objects are bound to displays. If an animation updates an object, it marks it as "invalidated" and the right parts will be redrawn on the next refresh of that display.

So I believe the animation callbacks will run and may invalidate some objects, but their displays will not refresh early.

https://github.com/lvgl/lvgl/blob/a87a28bcd4e8b10a6e3e1eb3fb81b92a5c2d617a/src/core/lv_refr.c#L83-L98

I might be wrong.

kisvegabor, 2024-12-03:
I can confirm what Liam said. LVGL will mark the areas as to-be-redrawn but will redraw them only when the display is re-rendered.

M1NGS, 2024-12-04:
This is a test results from gperf I did earlier, unfortunately, I deleted this test code, it written in LuaJIT+FFI, so you will see lua functions in the gperf data.

when creating a display with a lv_spinner, it can reach 8115 FPS, in SDL2.

[image]

 creating 999 displays with a lv_spinner each, FPS was cut in half

[image]

createing 1500 displays  with a  lv_spinner eachFPS is only 1/5 of that on one display

[image]

The pseudo code is as follows

```c
for (i=0; i < disp_num;i++)
{
disp[i] = lv_display_create(64, 64);
//some code for setup display
create_spinner(disp[i]);
}

while (1)
{

lv_refr_now(disp[0]); //only refersh one of them

SDL_UpdateTexture(lvgl, nil, payload.disp_buf, line_pixels); //update lvgl display buffer to texture
SDL.SDL_RenderCopy(renderer, lvgl, nil, rect);
SDL.SDL_RenderPresent(renderer); //show
count_fps();
}
``` 

I'll rewrite this test in c later.

M1NGS, 2024-12-20:
@kisvegabor @liamHowatt 

Here is my test case
[https://gist.github.com/M1NGS/a3cb7fd2f570996d8c57178f9ffaa12e](url)

Through my results, you can see that as the number of displays increases, the time to refresh a single display also increases.

```shell
# ./lvgl_bench 1 2 1000
Create 1 displays with 2 spinners each, run 1000 loops
56ms Used

# ./lvgl_bench 10 2 1000
Create 10 displays with 2 spinners each, run 1000 loops
84ms Used

# ./lvgl_bench 50 2 1000
Create 50 displays with 2 spinners each, run 1000 loops
314ms Used

# ./lvgl_bench 100 2 1000
Create 100 displays with 2 spinners each, run 1000 loops
772ms Used
```

kisvegabor, 2024-12-22:
Sorry for the late reply. 

`lv_refr_now` check all the 1500 animation if there is anything to do . Unfortunately I can't a good and simple optimization opportunity here. 🙁

What is the use case for 1500 displays?

M1NGS, 2024-12-22:
There is no need to apologize, I am grateful that you can reply to me despite your busy schedule. 
Don't worry, bro. it's just a not-so-smart attempt on my part. I originally planned to put a widget in each display, so that I could simply forward the mouse input when I mapped display to an SDL Texture, and refresh the corresponding display on demand to force the widgets to update.
 I have now realized that LVGL is not designed to work in this scenario. I have redesign logic to rendering multiple widgets in one display, mapping them to different SDL Texure coordinates, and refreshing a certain widget through a task queue. In this way, I expect to only need 4-5 displays to complete the work. But I still expect that in the future lv_refr_now(disp) can be able to do the work of only a single display in logically and practically, I don't have much time to read the code for the refresh part right now, but I will try to implement it and contribute to the project later if I have some free time.
Once again, Thank you for your reply, This gives me endless motivation to do my work.
Good health, good luck and much happiness throughout the year!

closed 2024-12-26 by M1NGS

kisvegabor, 2025-01-03:
Thank your for the kind words! I'm happy that you found a solution.  Unfortunately, I can't look into it now either, but I'll keep it in mind. Hopefully it will be connected to another issue/feature and we can hit two birds with one stone.