lvgl/lvgl #7364: Interrupt from the touch panel being triggered every 50ms

vishvanathdoijad, 2024-11-26:
### Introduce the problem

This results in unwanted repeated events being generated in the LVGL8.3 library, which can cause the button's release event to be registered multiple times.
[image]

liamHowatt, 2024-12-02:
You're seeing the `LV_EVENT_RELEASED` being sent rapidly? Could you share some code to reproduce the issue?

vishvanathdoijad, 2024-12-02:
void touch_interrupt_handler(void *CallbackRef) {
	XGpio *GpioPtr = (XGpio *)CallbackRef;
	touch_read();
	/* Clear the Interrupt */
	XGpio_InterruptClear(GpioPtr, 1);
}

void touch_read(void) {
	XIicPs_MasterSendPolled(&Iic, &reg_data, 1, devAddress);
	while (XIicPs_BusIsBusy(&Iic));
	XIicPs_MasterRecvPolled(&Iic,data,6, devAddress);
	while (XIicPs_BusIsBusy(&Iic));

	x=((data[1] << 8) | data[0]);
	y=((data[3] << 8) | data[2]);

	x= (int)((x -32769) / (float)15.75);
	y= (int)(y / (float)27.25);

// For Portrait
	rot_x= 600-y;
	rot_y= x;

//	if(x>=0 && x<1024){														// For landscape
	if(rot_y>=0 && rot_y<1024){													// For Portrait
		touchpad_pressed = 1;
//		xil_printf("Landscape Touch X :%d \t Y: %d\r\n",x,y);
//		xil_printf("Rotated Touch X :%d \t Y: %d\r\n",rot_x,rot_y);
	}
}

/*Will be called by the library to read the touchpad*/
static void lvgl_touch_cb(lv_indev_drv_t * indev_drv, lv_indev_data_t * data)
{
    static lv_coord_t last_x = 0;
    static lv_coord_t last_y = 0;

    /*Save the pressed coordinates and the state*/
    if(touchpad_pressed == 1) {
    	touchpad_pressed = 0;
    	last_x = rot_x;
    	last_y = rot_y;
        data->state = LV_INDEV_STATE_PR;
    }
    else {
        data->state = LV_INDEV_STATE_REL;
    }

    /*Set the last pressed coordinates*/
    data->point.x = last_x;
    data->point.y = last_y;
}

vishvanathdoijad, 2024-12-02:
"Touch INT occurs because the panel itself sends an INT every 50ms when touch is detected and then continues until the release."

kisvegabor, 2024-12-03:
You shouldn't set `touchpad_pressed = 0;` in the `lvgl_touch_cb`. Instead set/clear it only in `touch_read()`. 

Otherwise if the LVGL's touch read period is faster than the interrupt, you will read  `touchpad_pressed == 0`

closed 2024-12-20 by vishvanathdoijad