lvgl/lvgl #7369: Add polar chart

escherstair, 2024-11-28:
### Introduce the problem

LVGL has [chart](https://docs.lvgl.io/master/details/widgets/chart.html) widget that are useful to represet XY cartesian plots.
Sometimes a polar chart (i.e., radius+angle instead of XY) is needed.
As an example [this is](https://plotly.com/javascript/polar-chart/) polar chart for javascript.

Do you think that this kind of widget has a general purpose interest?

escherstair, 2024-12-02:
Hi @liamHowatt 
from your answe I understand that you don't see any chance of having this kind of control available in LVGL (soon or later).
The usage is pretty mathematics and so, maybe, not the main target for LVGL.

Am I right?

kisvegabor, 2024-12-03:
Hey,

I also feel like it'd be a niche feature, however now we are fully booked with partner/customer support and developing our UI editor.

We can get back to this and similar nice to have feature a little bit later. 

So I'm not saying it's not interesting and useful, but we needed to prioritize a few other things now.  :slightly_frowning_face:

escherstair, 2024-12-20:
Should this issue be marked with label Idea?

escherstair, 2024-12-23:
I add here some ideas I have, thinking to a future development of such a kind of widget:

- I think that [lv_chart](https://github.com/lvgl/lvgl/blob/cdaaeb81dc3e1f9023749d1a79c5c7970cad10c1/src/widgets/chart/lv_chart.h) should be used as a starting point for the development
- the interface (i.e., the public functions in lv_chart.h) should be kept, as much as possible
- the axes are not `x` and `y`, but radius `r` and `angle`
- an enumeration of the polar_chart axes could be
```C
  /**
   * Enumeration of the axis'
   */
  typedef enum {
      LV_CHART_AXIS_PRIMARY_R     = 0x00,
      LV_CHART_AXIS_SECONDARY_R   = 0x01,
      LV_CHART_AXIS_ANGLE     = 0x02,
      LV_CHART_AXIS_LAST
  } lv_polar_chart_axis_t;
```
  since `angle` axis could not have a secondary one.
  And even for the `radius` axis, I have some doubts.
- `lv_polar_chart_set_range()` would only accept `radius` axis
- a new function is needed to set where is the point with `angle` = 0° in the cartesian `x, y` plane
- a new function is needed to set the direction of increasing angles (i.e., clockwise vs counterclockwise)
- `lv_chart_set_div_line_count()` sets the number of division lines, that are circles and radiuses. circles can be drawn either with `lv_draw_arc` or [in this way](https://forum.lvgl.io/t/how-to-draw-a-perfect-circle/17151)
- at the first stage, only scattered points or line could be supported

Feel free to share your thoughts on the above ideas.

kisvegabor, 2025-01-03:
I've just added the "Investigating" label. 

I agree with your list, just one thing: On a polar chart we could have any number of radial axis with different ranges. It sounds a little overkill to me. What about having only `LV_CHART_AXIS_RADIAL` and `LV_CHART_AXIS_ANGLE`?
If multiple radial ranges are needed, transparent charts can be created on top of the current.

---

If you have implemented  it (even partially) feel free to send a Pull request.

escherstair, 2025-01-04:
> What about having only `LV_CHART_AXIS_RADIAL` and `LV_CHART_AXIS_ANGLE`?
> If multiple radial ranges are needed, transparent charts can be created on top of the current.

This is ok for me.

> If you have implemented it (even partially) feel free to send a Pull request.

Nothing ready yet @kisvegabor . I let you know.

escherstair, 2025-01-21:
I need some time to provide an idea of PR.
Is it possible not marking the issue as stale, please?

closed 2025-03-11 by lvgl-bot