lvgl/lvgl #7371: CMake generated visual studio project fails to build

zylalx1, 2024-11-29:
### LVGL version

v9.3.0

### Platform

visual studio 2022 on windows 11

### What happened?

an error occurs at pre-link event when building project lvgl:
...
2>Auto build dll exports
2>Couldn't open file '(something)/build/windows-base/lvgl.dir/Debug/lv_blend_helium.obj' with CreateFile()
...

seems like the obj files of those ARM .s code are included in the prerequisites of the dll library (check /build/windows-base/lvgl.dir/Debug/objects.txt), but MSVC doesn't know how to compile those files and just ignores them. 

i tried this patch on /env_support/cmake/custom.cmake and the issue solved, but not sure if i am doing right:

```
--- custom_old.cmake    2024-11-29 09:47:46.000000000 +0800
+++ custom.cmake        2024-11-29 15:05:11.943129600 +0800
@@ -18,7 +18,10 @@
 option(BUILD_SHARED_LIBS "Build shared libraries" OFF)

 # Set sources used for LVGL components
-file(GLOB_RECURSE SOURCES ${LVGL_ROOT_DIR}/src/*.c ${LVGL_ROOT_DIR}/src/*.S)
+file(GLOB_RECURSE SOURCES ${LVGL_ROOT_DIR}/src/*.c)
+if(NOT (CMAKE_C_COMPILER_ID STREQUAL "MSVC"))
+    file(GLOB_RECURSE SOURCES ${LVGL_ROOT_DIR}/src/*.S)
+endif()
 file(GLOB_RECURSE EXAMPLE_SOURCES ${LVGL_ROOT_DIR}/examples/*.c)
 file(GLOB_RECURSE DEMO_SOURCES ${LVGL_ROOT_DIR}/demos/*.c)
 file(GLOB_RECURSE THORVG_SOURCES ${LVGL_ROOT_DIR}/src/libs/thorvg/*.cpp ${LVGL_ROOT_DIR}/src/others/vg_lite_tvg/*.cpp)
```

### How to reproduce?

use cmake to generate visual studio solution with default config
build project "lvgl" in visual studio

liamHowatt, 2024-12-02:
This is a recurring issue. Please open a PR and we can test and review it. Thanks!

liamHowatt, 2024-12-04:
Fixed by #7401