lvgl/lvgl #7372: Font Converter Error

DarkLigther, 2024-11-29:
### LVGL version

v8.2.0

### Platform

Coter-M4F  with freertos.

### What happened?

1. Use Font Converter to generate a library containing eight Chinese characters in bin format. (计订认讥讨让讪讫)
2. After loading the font library, use lv_label_set_text(label, "讧"); Set the lable.
Ideally, lvgl reports an error indicating that the font library cannot be found.
Current situation: lvgl has drawn "讧" using the glyph of "计".
[image]
[font.zip](https://github.com/user-attachments/files/17956874/font.zip)

Allen-azh, 2024-12-04:
@kisvegabor

Allen-azh, 2024-12-20:
https://github.com/lvgl/lvgl/pull/7404#issuecomment-2552226223

have figured out

closed 2024-12-20 by Allen-azh