lvgl/lvgl #7383: Add support for LV_COLOR_FORMAT_ARGB1555 to lv_draw_dma2d.c

jhnlmn, 2024-12-01:
### Problem to solve

Need to show icons with transparency (1555 format) on top of background in 565 format.

### Success criteria

Bitmap with transparency is shown

### Solution outline

```
src/draw/dma2d/lv_draw_dma2d.c
133a134,135
>         case LV_COLOR_FORMAT_ARGB1555: // test: support transparent images
>             return LV_DRAW_DMA2D_OUTPUT_CF_ARGB1555;// test: support transparent images
311c313,315
<                          || dsc->header.cf == LV_COLOR_FORMAT_RGB565)
---
>                          || dsc->header.cf == LV_COLOR_FORMAT_RGB565
>                          || dsc->header.cf == LV_COLOR_FORMAT_ARGB1555 // test: support transparent images
>                          )

```
Note: I only added LV_COLOR_FORMAT_ARGB1555 for images, not for "base.layer" because it is not clear what a use case may be for having transparent background and how to test it.

### Rabbit holes

N/A

### Testing

.

### Teaching

N/A

liamHowatt, 2024-12-02:
Oh, yes, this should work. The `LV_DRAW_DMA2D_OUTPUT_CF_...` enum values correspond to DMA2D register value enumerations, so there's no reason this shouldn't work.

Can you open a PR for it? Thanks.

jhnlmn, 2024-12-04:
I do not have permission to make a branch/PR to lvgl. github offered me to make a full fork, but that looks like an overkill for these 4 lines of changes.

uLipe, 2024-12-06:
I can take over this, self assigned it to me.

uLipe, 2024-12-23:
Not stale, this is on my backlog

uLipe, 2025-01-06:
Follow the PR: https://github.com/lvgl/lvgl/pull/7555

PR #7555 (2025-01-10): feat(dma2d): add support for ARGB1555 color on top of RGB565 format
Fixes #7383 

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/code-format [... cut]
closed 2025-01-10 by liamHowatt