lvgl/lvgl #7385: theme_default size-dependent styles are decided once for all displays

liamHowatt, 2024-12-01:
### LVGL version

v9.3.0-dev

### Platform

NuttX simulator.

### What happened?

The default theme padding styles depend on the display size. It is decided one time globally for all displays. The first display created initializes the default theme upon creation and is not updated after resizing.

This is most noticeable with `lv_nuttx_fbdev.c` where the size is set arbitrarily

https://github.com/lvgl/lvgl/blob/1a0b6d313e5b2e8dc3bc943c36badc590f03fe92/src/drivers/nuttx/lv_nuttx_fbdev.c#L81

and then set to the real size later.

https://github.com/lvgl/lvgl/blob/1a0b6d313e5b2e8dc3bc943c36badc590f03fe92/src/drivers/nuttx/lv_nuttx_fbdev.c#L171

This means that for all NuttX users, if they set the display width to 240 the normal way through menuconfig, keyboards will look like this:

[image]

uLipe, 2024-12-02:
In that case since all files that are included to nuttx build system can see the `nuttx/config.h` why not create a Kconfig symbol set at the application level that can be fetched by the port layer and then perform this configuration correctly at first time?

```
 lv_display_t * disp = lv_display_create(CONFIG_FBDEV_DISPLAY_HEIGHT, CONFIG_FBDEV_DISPLAY_WIDTH); 
```

kisvegabor, 2024-12-02:
I can't comment on the NuttX part (cc @XuNeo), but in LVGL I'm thinking for long how to manages themes in better way. 

Now, e.g. when a button is created a lot of styles are assigned to it. However it doesn't make sense to add all the styles to all the buttons. We can just say, if it's a button, read its styles, and check the the theme styles of `lv_button_class` defined only once in the theme. It's probably for a later minor or major version.

liamHowatt, 2024-12-03:
> In that case since all files that are included to nuttx build system can see the `nuttx/config.h` why not create a Kconfig symbol set at the application level that can be fetched by the port layer [...]

That would be a good solution but the specific config depends on which driver implements the fbdev. It could be `CONFIG_SIM_FBWIDTH`/`HEIGHT`, `CONFIG_VNCSERVER_SCREENWIDTH`/`HEIGHT`, `CONFIG_ESP32S3_LCD_VRES`/`HRES`.

XuNeo, 2024-12-03:
The issue with NuttX fbdev can be resolved relatively easily. The primary challenge lies in managing the theme after a resize event.

Would it be appropriate to subscribe to display events in lv_theme_default.c and update the padding style accordingly?

kisvegabor, 2024-12-03:
> The issue with NuttX fbdev can be resolved relatively easily. The primary challenge lies in managing the theme after a resize event.

Great! 

> Would it be appropriate to subscribe to display events in lv_theme_default.c and update the padding style accordingly?

We can do something like that, but now there is only a single theme which gets initialized. If multiple displays are used with different resolution the theme will be incorrectly updated affecting both displays. Now it's set once, which is not ideal either, but at least easier to see what's happening.

liamHowatt, 2024-12-19:
> The issue with NuttX fbdev can be resolved relatively easily.

Any objections to me opening a PR which addresses it in NuttX? Fyi `lv_linux_drm` uses this pattern too.

closed 2025-01-17 by lvgl-bot